olikraus/u8g2 #2065: SSD1322 256x64 with RPI Pico: only left half of the screen used

XFer012, 2023-01-02:
Hello,
using u8g2 v2.33.15 with Raspberry Pico (RP2040).
IDE is Arduino 1.8.19, board package is Earle Philover's Arduino-Pico v2.6.5
Trying to run "Graphic Test" example (from u8g2 lib)

Display: Yellow OLED with SSD1322 controller, SPI interface
Connections (pinout https://images.theengineeringprojects.com/image/webp/2021/03/what-is-raspberry-pi-pico.png.webp?ssl=1): 
SCK on Pico GP18
MOSI on GP19
D/C on GP15
CS on GP17
Reset on GP14

Constructor:

```
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 17, /* dc=*/ 15, /* reset=*/ 14);	// Enable U8G2_16BIT in u8g2.h
```

I have enabled U8G2_16BIT in 
libraries\U8g2\src\clib\u8g2.h :

```
/*
  The following macro enables 16 Bit mode. 
  Without defining this macro all calulations are done with 8 Bit (1 Byte) variables.
  Especially on AVR architecture, this will save some space. 
  If this macro is defined, then U8g2 will switch to 16 Bit mode.
  Use 16 Bit mode for any display with more than 240 pixel in one 
  direction.
*/
#define U8G2_16BIT

```

Problem: 
Only the left-half part of the display is drawed upon; the right-half part stays black

[image]

olikraus, 2023-01-02:
Did you try the FlipMode example? It will use the full display size.

XFer012, 2023-01-02:
It works! Nice to know!

closed 2023-01-02 by XFer012