olikraus/u8g2 #2066: Signals for the LCD are not the same on 2 different MCUs

defavltt, 2023-01-03:
Trying to talk to a st7565(ks0713 type) LCD with AVR128DB48, nothing shows up on it. Talking to it with 328P, it works. Same code, both software SPI. Signals between the 2 MCUs look completely different despite the speed.
Was able to write successfully to a couple of different LCDs with both MCUs no problem: st7565(erc12864), st7565(lx12864), st7567(os12864).
Seems like this constructor in particular st7565(ks0713) doesnt behave how it should.
Tried both HW and SW SPI, doesnt help.
Here are the signals going to the LCD from 2 MCUs.
[image]

olikraus, 2023-01-03:
Not sure how I can help here. To me it looks more like the uC runs into reset condition very soon. Is your Watchdog Timer enabled?

defavltt, 2023-01-03:
I have just remembered, I modified u8x8_d_st7565.c file and added my own constructor for ks0713, but never tried it on this mcu. Think I just copied it from the st7565_lx12864 constructor and modified the height and width. Now its kinda working with the st7565_64128n constructor part of the image is broken, the display I use is 132x32. The thing is when I change the tile width from 16(for 128px) to 17(for 132px) in the constructor info the image disappears, this is the problem I think.

olikraus, 2023-01-03:
> when I change the tile width from 16(for 128px) to 17(for 132px) in the constructor info 

If you change this value, then you also need to change the static memory allocation used by u8g2. If you don't do this, then this will cause a memory leak / issue, which may cause a reset.

defavltt, 2023-01-04:
This time actually the MCU didn't freeze, it continued to run the loop, just no image. This could be that the AVR128DB48 arduino core handles static memory differently maybe, normal arduino uno works fine in this case. Also isn't this data in the constructor info used for addressing the size for the buffer? You said I have to do it manually, searched for how the buffer is allocated in the library files, but couldnt find it.

olikraus, 2023-01-04:
Lets take the st7565 nhd c12832 as example with the tile dimension of 16x4, which are 16*4*8 = 512 bytes.

The info structure is here:
https://github.com/olikraus/u8g2/blob/b42953b87a60c10d0e0887f9259537a93f122aaf/csrc/u8x8_d_st7565.c#L505-L529

Indeed (for memory allocation), the tile size has to be applied a second time here:
https://github.com/olikraus/u8g2/blob/b42953b87a60c10d0e0887f9259537a93f122aaf/tools/codebuild/codebuild.c#L1255-L1261

codebuild is a code generator, which will generate all the constructors and also ensures that the memory is assigned correctly.

For the "full" buffer constructor / init, the corresponding setup procedure is created here:
https://github.com/olikraus/u8g2/blob/b42953b87a60c10d0e0887f9259537a93f122aaf/csrc/u8g2_d_setup.c#L4818-L4825

This setup calls the "u8g2_m" function, which actually does the  memory allocation  here:
https://github.com/olikraus/u8g2/blob/b42953b87a60c10d0e0887f9259537a93f122aaf/csrc/u8g2_d_memory.c#L28-L38

If you increase the size in the display info struct and you do not want to call codebuild, then the most simple way would be to find the corresponding "u8g2_m" function and increase the static buffer size there.

defavltt, 2023-01-04:
Made it work finally!

u8x8_d_st7565.c constructor info:
```
static const u8x8_display_info_t u8x8_st7565_64128n_display_info =
{ //changed
  /* chip_enable_level = */ 0,
  /* chip_disable_level = */ 1,
  
  /* post_chip_enable_wait_ns = */ 150,	/* st7565 datasheet, table 26, tcsh */
  /* pre_chip_disable_wait_ns = */ 50,	/* st7565 datasheet, table 26, tcss */
  /* reset_pulse_width_ms = */ 1, 
  /* post_reset_wait_ms = */ 1, 
  /* sda_setup_time_ns = */ 50,		/* st7565 datasheet, table 26, tsds */
  /* sck_pulse_width_ns = */ 120,	/* half of cycle time (100ns according to datasheet), AVR: below 70: 8 MHz, >= 70 --> 4MHz clock */
  /* sck_clock_hz = */ 4000000UL,	/* since Arduino 1.6.0, the SPI bus speed in Hz. Should be  1000000000/sck_pulse_width_ns */
  /* spi_mode = */ 0,		/* active high, rising edge */
  /* i2c_bus_clock_100kHz = */ 4,
  /* data_setup_time_ns = */ 40,	/* st7565 datasheet, table 24, tds8 */
  /* write_pulse_width_ns = */ 80,	/* st7565 datasheet, table 24, tcclw */
  /* tile_width = */ 17,		/* width of 16*8=128 pixel */ //original 16
  /* tile_hight = */ 4, //original 8
  /* default_x_offset = */ 0, //original 4
  /* flipmode_x_offset = */ 0,
  /* pixel_width = */ 132, //original 128
  /* pixel_height = */ 32 //original 64
};
```

u8g2_d_setup.c:
```
void u8g2_Setup_st7565_64128n_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb)
{
  uint8_t tile_buf_height;
  uint8_t *buf;
  u8g2_SetupDisplay(u8g2, u8x8_d_st7565_64128n, u8x8_cad_001, byte_cb, gpio_and_delay_cb);
  buf = u8g2_m_17_4_f(&tile_buf_height);
  u8g2_SetupBuffer(u8g2, buf, tile_buf_height, u8g2_ll_hvline_vertical_top_lsb, rotation);
}
```

All that needed to be done is to put the appropriate memory allocation function in the setup, in this case `u8g2_m_17_4_f`

Really appreciated your help!

closed 2023-01-04 by defavltt

olikraus, 2023-01-04:
> All that needed to be done is to put the appropriate memory allocation function in the setup, in this case u8g2_m_17_4_f

of course, this is also possible...