olikraus/u8g2 #2069: ST7567 with JHD12864

spi5498, 2023-01-06:
Hello,

Just got a "GFX Pack" for Raspberry Pi Pico, featuring a LCD with the ST7567 controller.
The LCD itself is a COG JHD12864-G176B-G.

I do have a functioning port of U8G2 library to the Pico platform, so I wanted to try this one out also.
(As soon as I have tested more LCDs and done some cleaning I will be happy to post the code of this port, limited time on my side, so the port has taken some time!)

Unfortunately I was only find one constructor that works with the LCD, and that is:
u8g2_Setup_st7567_pi_132x64_f (Had to use the SetContrast function)
This is unfortunately not the correct size, however the other constructors with the correct size for ST7567 does not work from what I have found.

Do you have any tips on modifications needed for this? I will be happy to assist to do any testing to get this to work correctly!
Just let me know what info or tests are needed.

olikraus, 2023-01-07:
I could create a copy of this constructor with the correct dimensions.
Any datasheet for your display?

spi5498, 2023-01-07:
> I could create a copy of this constructor with the correct dimensions.
> 
> Any datasheet for your display?
> 
> 
> 
> 

That would be fantastic!

The closest one i could find seems to be this: https://biakom.com/pdf/JHD12864-G343BSW.pdf

olikraus, 2023-01-08:
I just was about to update the repo and saw, that there are already some 128x64 constructors.
Did you test them all?

U8G2_ST7567_JLX12864, U8G2_ST7567_ENH_DG128064, U8G2_ST7567_ENH_DG128064I, U8G2_ST7567_OS12864, U8G2_ST7567_JLX12864

spi5498, 2023-01-20:
Hey,

I just did another test with all of the above.
The only constructor besides the "pi" one that works is "os12864".
I must have missed that, sorry!
If you need anything other tested, just let me know. Otherwise i guess we can close this one.
Thanks a lot for checking in!

olikraus, 2023-01-22:
Ok, great... :-)

closed 2023-01-22 by olikraus