olikraus/u8g2 #2071: GU128x64-800B VFD-Display not working properly.

Jan1503, 2023-01-07:
Hi!

Just saw that your wonderful library supports this retro vfd-display thanks to a request about a year ago by @Bankst .

However I cant get it to work in SPI-Mode whether I use hw or sw-spi it stays dark.
The Adafruit-GFX compatible library mentioned in the initial issue works without a problem so wiring should be fine.

I then desoldered the jumper to put it back in parallel mode and then it works but its very dark.

[image]

If I reset the display and connect the test-pin it is very bright so your library seems to dim the display.
Setting the contrast has no effect.

Would you mind take a quick look?
Thanks!

olikraus, 2023-01-07:
I don't own such a display, so I can't do any tests. I am open to any suggestions.

Jan1503, 2023-01-07:
I bought them from [ebay](https://www.ebay.de/itm/294089544835?mkcid=16&mkevt=1&mkrid=707-127634-2357-0&ssspo=_BvE2mnkSxO&sssrc=2349624&ssuid=cHVi3Zk6R2G&var=&widget_ver=artemis&media=COPY) currently...

Where should I take a look about the brightness in your code? It would be ok to use the parallel mode but not if I cant increase the brightness.

olikraus, 2023-01-07:
The default brightness is assigned here: https://github.com/olikraus/u8g2/blob/b42953b87a60c10d0e0887f9259537a93f122aaf/csrc/u8x8_d_gu800.c#L161

Hmm, I don't remember whether I had any information on the commands...

Jan1503, 2023-01-07:
I think this is the full datasheet?
[GU128x64-800B](https://media.digikey.com/pdf/Data%20Sheets/Noritake%20PDFs/GU128X64-800B.pdf)

It seems 4F should be 40 instead according to page 7 and 9 but I might be wrong, took just a quick look while traveling.
But the display seems to initialize at 100% according to page 8, chapter 9.3.

olikraus, 2023-01-07:
Ok, so it should be 0x40 for 100% brightness, right?

Jan1503, 2023-01-08:
Yes, I can confirm after setting 0x40 display is full bright. You may even omit the sequence as the display initializes to full brightness after start.

Your setContrast method for this display must be altered, too.

olikraus, 2023-01-08:
Yes, i have updated the default init and also the set contrast function: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_gu800.c

You can check the changes here: https://github.com/olikraus/u8g2/commit/cd44b0538806f41402431ae0306362707011fbb7

Shell I create a release? Instead of a full release you could just copy this file to your local u8g2 installation.

Jan1503, 2023-01-16:
Sorry for my late reply.
No extra release necessary.
Regards!

closed 2023-01-16 by Jan1503