olikraus/u8g2 #2072: Previous versions of u8g2 gone?

amaschas, 2023-01-08:
Up until today I was using v2.33.15 as a project dependency, installing from github, but it appears that all of the older versions of the library have vanished other than 2.29.11. Was this an accident? Is there a better way to access older versions?

v1993, 2023-01-08:
This is git, so you definitely can access older versions by their commit hashes. Check out [commit history](https://github.com/olikraus/u8g2/commits/master) to find them.

amaschas, 2023-01-08:
Sure, but I used to be able to reference `olikraus/U8g2@2.33.15` as a dependency, and that no longer seems possibe. I can go through the git history and see that you made two commits named `2.33.15` on 9/4/2022, and then I can go and figure out which one corresponds to the version I was using before, but that seems a little... suboptimal? Did something change recently with this?

olikraus, 2023-01-08:
I am not aware that I changed something here. The main u8g2 repo only receives tags which are named according to the release number.

amaschas, 2023-01-08:
Currently, if I look at the set of tags on this repo I see only one tag for v2.29.11.

olikraus, 2023-01-08:
hmm... I think there should be more.
At least the derived Arduino project contains all the tags: https://github.com/olikraus/U8g2_Arduino/tags

olikraus, 2023-01-09:
hmm sure how I can support here. At least I have not intentionally removed and tags :-/

amaschas, 2023-01-09:
I think the only solution here is probably manually re-adding them, at least for the versions documented in the changelog.

amaschas, 2023-01-11:
Ah interesting. Should I be using that repo for non-Arduino implementations as well? Would v2.33.15 on the arduino repo be identical to the version here? I'm actually using u8g2 with a library that adapts it for use with ESP32.

olikraus, 2023-01-11:
Yes, versions between the repos are always identical. And yes, please also use U8g2_Arduino also for none Arduino code.

closed 2023-02-22 by olikraus