olikraus/u8g2 #2073: Arduino for STM32 can't use software I2C OLED

ericzhang-debug, 2023-01-08:
Hello, guys
I use latest Stm32duino and my chip is STM32F103ZET6
I tried to use the 128X64 OLED by using U8g2Lib's software I2C , but it doesn't work.
However, I used the same code on Arduino UNO,and it worked well (except pins definations).Also, I tried the hardware I2C on this STM32 board and it worked well.

SDA => PF8
SCL => PF10

Bellow is my code:
`#include <U8g2lib.h>
#include <Wire.h>
#define SCL_OLED PF10
#define SDA_OLED PF8
U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /*clock=*/SCL_OLED, /*data=*/SDA_OLED, /*reset=*/U8X8_PIN_NONE);
void setup() {
  pinMode(SDA_OLED, OUTPUT);
  pinMode(SCL_OLED, OUTPUT);
  u8g2.begin();
  u8g2.enableUTF8Print(); // enable UTF8 support for the Arduino print() function
}

void loop() {
  u8g2.setFont(u8g2_font_unifont_t_symbols);
  u8g2.firstPage();
  do
  {
    u8g2.setCursor(0, 15);
    u8g2.print("Hello World!");
  } while (u8g2.nextPage());
  delay(1000);
}`

olikraus, 2023-01-08:
Maybe an output+ pull-up is required

amolide, 2023-01-15:
Try to change "NONAME_F_SW_I2C" to  "NONAME_1_SW_I2C". Because do-while function can only be used in page-buffer mode.

pangflash, 2023-02-22:
I also encountered the same situation. The hardware I2C is OK, but the software I2C is unrealistic. Have you solved this problem?

ericzhang-debug, 2023-02-22:
No, I tried all the solutions mentioned,but no one could work. I am waiting for the official solution and i changed to the spi these days

| |
***@***.***
|
|
***@***.***
|

---- Replied Message ----
| From | ***@***.***> |
| Date | 02/22/2023 09:05 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [olikraus/u8g2] Arduino for STM32 can't use software I2C OLED (Issue #2073) |

I also encountered the same situation. The hardware I2C is OK, but the software I2C is unrealistic. Have you solved this problem?

ChitoKim, 2023-05-13:
Actually I bumped into the same issue, but on RP2040. In my case, the issue persists only in a specific retailer's SSD1306 OLED. I bought OLEDs from two retailers, and only one of them has this issue. The faulty one's behaviour is really weird. It displays fine on AVR both via HW I2C, and SW I2C. However, in case of RP2040, it runs normal via HW I2C, but it shows nothing via SW I2C.

olikraus, 2023-05-13:
Difficult to say. Do you use the original Arduino IDE and code?

ChitoKim, 2023-05-14:
Yes, I used arduino ide(legacy version with vscode extension) , and example "HelloWorld". I uploaded the code with RP2040 core arduino-pico by earlephilhower. My board is just a plain Raspberry pi pico.

olikraus, 2023-05-14:
Maybe there is an issue with arduino-pico core lib in your case.