olikraus/u8g2 #2075: UC1638C_240X128

olikraus, 2023-01-09:
Created from https://github.com/olikraus/u8g2/issues/371

Datasheet: https://github.com/olikraus/u8g2/files/10375280/JHD240128-G02PFWD-B1.pdf

> jchifos pick 2017 u8x8_d_uc1608.c and modified to match UC1638 (I don't know how much modified)
Me modified jchifos's file to adapt to 240x128 JHD240128 screen (Line 187,191 and 199, just 3 values)

I have created something complete new out of the orignal code. There had been a lot of misunderstandings. I personally would prefer to add the code from scratch. For example the modified code you have attached uses U8X8_D1 macro for the init code. This is wrong and was fixed by me in the repo. Moreover jchifos never confirmed whether my modifications are correct or not.

Anyhow, let's start from the beginning. 

Your request is a UC1638 for 240x128 display size.

olikraus, 2023-01-09:
I have started the new display here: https://github.com/olikraus/u8g2/blob/c2c83694fe8a9c439ac8370527fe26d845a23c3b/csrc/u8x8_d_uc1638.c#L311

If I understood correctly, then you took this init code, correct?
``` C++
static const uint8_t u8x8_d_uc1608_erc24064_init_seq[] = {
    
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */

  U8X8_C(0x024),            		/*	 set temp comp*/
  U8X8_C(0x0c2),            		/*	mirror y and mirror x */  /* WAS: c2 */
  U8X8_C(0x0a2),            		/*	line rate */
  U8X8_C(0x0d6),            		/*	gray scale 2 */
  U8X8_C(0x0eb),            		/*	 set bias*/
  U8X8_C(0x095),            		/*	 set 1 bit per pixel, pattern 0*/
  U8X8_C(0x089),            		/*	 set auto increment, low bits are AC2 AC1 AC0 */  

  /* U8X8_C(0x0a5),            		   all on   */
  U8X8_C(0x081),		/* set contrast, EA default: 0x054 */    /* UC1638*/
  U8X8_D1(0x0a0),		/* set contrast, EA default: 0x054 */    /* UC1638*/

  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

```

srcastor, 2023-01-10:
thanks for you effort,  seems to me allright , letme know when ready to test

unfortunately I can't answer you about cs active, just I copy the file from jchifos and worked

olikraus, 2023-01-10:
ok, I have created version 2.34.12. New constructor is: U8G2_UC1638_192X96...

Are you able to test this somehow?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

srcastor, 2023-01-10:
I can test it right now but I don't understand the constructor , shouldn't it be U8G2_UC1638_240X128... ?
U8G2_UC1638_240X128_F_4W_HW_SPI u8g2(...

srcastor, 2023-01-11:
> Great. I have one more request:
> 
> Please execute FlipMode.ino example (See example menu in Arduino IDE or [here](https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/FlipMode/FlipMode.ino)). The flip mode test example will cycle through eight modes. In all modes there must be a solid frame next to the border of the display.
> 
> If it deviates, can you send me a picture?
As I don't understand what I see, I made a video, hopes 
https://user-images.githubusercontent.com/12575586/211906319-573740d2-f477-4ae7-82bd-31abdacb5ddd.MOV

srcastor, 2023-01-11:
As this workarround has soo many time, I just realized two weird things (that also happend with the jchifos original file)

One is that I must set the rotation to U8G2_R2 to get the right orientation, this picture it's made with U8G2_R0
[image]

Another its that I must set contrast to ~120 after begin, this picture it's made without contrast
[image]

I'm not sure if this it's normal behavior, just commenting

olikraus, 2023-01-11:
Thanks for the video. I have tried to fix the F1 mode, so hopefully the new release (2.34.13) will fix flip mode.
Flip mode is indeed identical to R2 but implements this inside the display hardware (using flip mode is faster than using R2). I also updated the default value for the contrast to 120.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

srcastor, 2023-01-12:
Contrast it's now ok by default,  for orientation you mean to use flipmode instead of R2? can be possible to fix it in the configuration (line343)?
About the flipmode.ino, still wrong in F1 modes
[image]

I figured it out what did you changed and tried random values until I get (I think) the right one: 4*16+0 , all 8 screens has a perfect frame across borders with correct words and no artifacts, hope its help 

[image]

olikraus, 2023-01-12:
> I figured it out what did you changed and tried random values until I get (I think) the right one: 4*16+0 , all 8 screens has a perfect frame across borders with correct words and no artifacts, hope its help

Great, I will fix this.

olikraus, 2023-01-12:
> By other hand, this is the ghosting I'm having, I wonder if this can be improved / bad init configuration or it's just this type of lcd limitation

There are multiple reasons possible, but there is also a limitation of the LCDs.
Let us start to ensure that there is no software issue: If you use u8g2 first/next page loop, then there must be no value change inside the loop itself.

```
  /* this is wrong ! */
  u8g2.setFont(u8g2_font_ncenB14_tr);
  u8g2.firstPage();
  do {
    value = some_sensor_output();
    u8g2.setCursor(0, 15);
    u8g2.print(value);
  } while ( u8g2.nextPage() );
```
Instead it should be:
```
  u8g2.setFont(u8g2_font_ncenB14_tr);
  value = some_sensor_output();
  u8g2.firstPage();
  do {
    u8g2.setCursor(0, 15);
    u8g2.print(value);
  } while ( u8g2.nextPage() );
```

If the ghosting artefacts are still there, then I assume there is LCD issue. Usually it can't be avoided fully, but maybe some improvements are possible: Changing the pixel values is faster if there is sufficient current available. This is usually ensured by capacitors very next to the display: There should be 1uF ceramic cap between VLCD and VSS and also between VDD and VSS. 2.2uF might be better. And maybe a parallel 100nF could further improve the situation.
The same applies to all the caps between VA1-/+, VA2-/+, VB1-/+ and VB2-/+. 1uF should be minimum.  2.2uF might be better. And of course: Every cap should be as close as possible to the flex cable connector of the display.

Thanks for testing and fixing the flip mode example!

closed 2023-01-19 by olikraus