olikraus/u8g2 #2078: SSD1351 as a monochrome constructor?

cranefist, 2023-01-17:
Would there be any issues in adding SSD1351 display as a monochrome constructor to this library?
By this i mean, as either R/G/B? Utilizing just one channel?

Im just asking this before i try it out if there are some obvious reason why this would be incompatible approach.

This way i could use it as a drop in replacement, as i just want a green oled.

olikraus, 2023-01-17:
Of course it would be possible, but I never tried this. Might require some programming effort.

The biggest issue, what I see, is performance. The SSD1351 requires 24 bits for the three color channels. This means the amount of data would be 24 times higher compared to an 1-Bit SSD1306 monochrome OLED. I would assume that you might reach up to 1 second to refresh the screen.

olikraus, 2023-01-17:
> To my understanding its 4bits per channel.. one channel being 5bit, can't remember which one of them. So if im updating only one channel, there is no way to get by with only 4 bits?

16 bit might be possible for this OLED, but then its 5/6/5 bit for the RGB data, so we would need 5 bit.
And no: To my knowledge we need to transfer the complete 16 bit.

> A lot of these monochrome oleds are also 4bit, like the SSD1327. And i got that upto 70fps.

Possible with a ESP32.

> Because my main issue with the color libraries was the speed, there was no fast way to clean the screen and it always looks like the screen is blinking.

Agree... well if you have 70fps you still might get 18 fps or so...

olikraus, 2023-01-17:
According to section 8.3.2 of the SSD1351 manual, 18 Bits are expected.

cranefist, 2023-01-18:
Ok, then this approach does not really make sense. I was hoping there was a way to only address one channel at a time.

If it always needs 18 bits, then it cant work. And yes, it was 5/6/5bits. Not 4/5/4.

I did get the adafruit library to work ok now, i just prefer this. At least i can use the fonts. Just makes it much more complicated to get a flicker free output.

Using monochrome oleds, even if the framerate is low.. i dont get the flickering that happens with this. I guess its because this updates every change individually as storing it all would take so much memory. So if i erase something, for a moment i have a blank screen before i redraw the content on it.

While your library updates the whole page on command from memory, so a blank screen is never displayed.

Thanks for the input!