olikraus/u8g2 #2079: Missing Unicode glyphs? Trying to print subscript 2 (Unicode 2082) for a CO₂ Monitor display.

ken830, 2023-01-18:
For a C0₂ monitor, I'm trying to print the subscript 2, but I can't seem to find any fonts that have that glyph. Is there another way to work around this?

ken830, 2023-01-18:
Subscript 2 is U+2082 in Unicode. 

[https://www.compart.com/en/unicode/U+2082](https://www.compart.com/en/unicode/U+2082)

The shifting workaround would be cumbersome because the subscript character is in the middle of a string and the string (and its length) is dynamic. I guess my first attempt would be in creating a custom font.

olikraus, 2023-01-19:
ok, this should be 8322 in decimal.
There are already a lot of fonts, which seem to include subscript chars.
The mentioned fonts below are located here: https://github.com/olikraus/u8g2/tree/master/tools/font/bdf
So you don't need to draw something, we just need to rerun the u8g2 font converter with the extended unicode range. 

10x20.bdf:ENCODING 8322
5x7.bdf:ENCODING 8322
5x8.bdf:ENCODING 8322
6x10.bdf:ENCODING 8322
6x12.bdf:ENCODING 8322
6x13B.bdf:ENCODING 8322
6x13.bdf:ENCODING 8322
6x9.bdf:ENCODING 8322
7x13.bdf:ENCODING 8322
7x14.bdf:ENCODING 8322
8x13.bdf:ENCODING 8322
9x15.bdf:ENCODING 8322
9x18.bdf:ENCODING 8322
b10_b.bdf:ENCODING 8322
b10.bdf:ENCODING 8322
b12_b.bdf:ENCODING 8322
b12.bdf:ENCODING 8322
b14_b.bdf:ENCODING 8322
b14.bdf:ENCODING 8322
b16_b.bdf:ENCODING 8322
b16.bdf:ENCODING 8322
courB08.bdf:ENCODING 8322
courB10.bdf:ENCODING 8322
courB12.bdf:ENCODING 8322
courB14.bdf:ENCODING 8322
courB18.bdf:ENCODING 8322
courB24.bdf:ENCODING 8322
courR08.bdf:ENCODING 8322
courR10.bdf:ENCODING 8322
courR12.bdf:ENCODING 8322
courR14.bdf:ENCODING 8322
courR18.bdf:ENCODING 8322
courR24.bdf:ENCODING 8322
cu12.bdf:ENCODING 8322
f10_b.bdf:ENCODING 8322
f10.bdf:ENCODING 8322
f12_b.bdf:ENCODING 8322
f12.bdf:ENCODING 8322
f14_b.bdf:ENCODING 8322
f14.bdf:ENCODING 8322
f16_b.bdf:ENCODING 8322
f16.bdf:ENCODING 8322
helvB08.bdf:ENCODING 8322
helvB10.bdf:ENCODING 8322
helvB12.bdf:ENCODING 8322
helvB14.bdf:ENCODING 8322
helvB18.bdf:ENCODING 8322
helvB24.bdf:ENCODING 8322
helvR08.bdf:ENCODING 8322
helvR10.bdf:ENCODING 8322
helvR12.bdf:ENCODING 8322
helvR14.bdf:ENCODING 8322
helvR18.bdf:ENCODING 8322
helvR24.bdf:ENCODING 8322
lubB08.bdf:ENCODING 8322
lubB10.bdf:ENCODING 8322
lubB12.bdf:ENCODING 8322
lubB14.bdf:ENCODING 8322
lubB18.bdf:ENCODING 8322
lubB19.bdf:ENCODING 8322
lubB24.bdf:ENCODING 8322
lubBI08.bdf:ENCODING 8322
lubBI10.bdf:ENCODING 8322
lubBI12.bdf:ENCODING 8322
lubBI14.bdf:ENCODING 8322
lubBI18.bdf:ENCODING 8322
lubBI19.bdf:ENCODING 8322
lubBI24.bdf:ENCODING 8322
lubI08.bdf:ENCODING 8322
lubI10.bdf:ENCODING 8322
lubI12.bdf:ENCODING 8322
lubI14.bdf:ENCODING 8322
lubI18.bdf:ENCODING 8322
lubI19.bdf:ENCODING 8322
lubI24.bdf:ENCODING 8322
luBIS08.bdf:ENCODING 8322
luBIS10.bdf:ENCODING 8322
luBIS12.bdf:ENCODING 8322
luBIS14.bdf:ENCODING 8322
luBIS18.bdf:ENCODING 8322
luBIS19.bdf:ENCODING 8322
luBIS24.bdf:ENCODING 8322
lubR08.bdf:ENCODING 8322
lubR10.bdf:ENCODING 8322
lubR12.bdf:ENCODING 8322
lubR14.bdf:ENCODING 8322
lubR18.bdf:ENCODING 8322
lubR19.bdf:ENCODING 8322
lubR24.bdf:ENCODING 8322
luBS08.bdf:ENCODING 8322
luBS10.bdf:ENCODING 8322
luBS12.bdf:ENCODING 8322
luBS14.bdf:ENCODING 8322
luBS18.bdf:ENCODING 8322
luBS19.bdf:ENCODING 8322
luBS24.bdf:ENCODING 8322
luIS08.bdf:ENCODING 8322
luIS10.bdf:ENCODING 8322
luIS12.bdf:ENCODING 8322
luIS14.bdf:ENCODING 8322
luIS18.bdf:ENCODING 8322
luIS19.bdf:ENCODING 8322
luIS24.bdf:ENCODING 8322
luRS08.bdf:ENCODING 8322
luRS10.bdf:ENCODING 8322
luRS12.bdf:ENCODING 8322
luRS14.bdf:ENCODING 8322
luRS18.bdf:ENCODING 8322
luRS19.bdf:ENCODING 8322
luRS24.bdf:ENCODING 8322
ncenB08.bdf:ENCODING 8322
ncenB10.bdf:ENCODING 8322
ncenB12.bdf:ENCODING 8322
ncenB14.bdf:ENCODING 8322
ncenB18.bdf:ENCODING 8322
ncenB24.bdf:ENCODING 8322
ncenR08.bdf:ENCODING 8322
ncenR10.bdf:ENCODING 8322
ncenR12.bdf:ENCODING 8322
ncenR14.bdf:ENCODING 8322
ncenR18.bdf:ENCODING 8322
ncenR24.bdf:ENCODING 8322
t0-11b-uni.bdf:ENCODING 8322
t0-11-uni.bdf:ENCODING 8322
t0-12b-uni.bdf:ENCODING 8322
t0-12-uni.bdf:ENCODING 8322
t0-13b-uni.bdf:ENCODING 8322
t0-13-uni.bdf:ENCODING 8322
t0-14b-uni.bdf:ENCODING 8322
t0-14-uni.bdf:ENCODING 8322
t0-15b-uni.bdf:ENCODING 8322
t0-15-uni.bdf:ENCODING 8322
t0-16b-uni.bdf:ENCODING 8322
t0-16-uni.bdf:ENCODING 8322
t0-17b-uni.bdf:ENCODING 8322
t0-17-uni.bdf:ENCODING 8322
t0-18b-uni.bdf:ENCODING 8322
t0-18-uni.bdf:ENCODING 8322
t0-22b-uni.bdf:ENCODING 8322
t0-22-uni.bdf:ENCODING 8322
timB08.bdf:ENCODING 8322
timB10.bdf:ENCODING 8322
timB12.bdf:ENCODING 8322
timB14.bdf:ENCODING 8322
timB18.bdf:ENCODING 8322
timB24.bdf:ENCODING 8322
timR08.bdf:ENCODING 8322
timR10.bdf:ENCODING 8322
timR12.bdf:ENCODING 8322
timR14.bdf:ENCODING 8322
timR18.bdf:ENCODING 8322
timR24.bdf:ENCODING 8322
unifont.bdf:ENCODING 8322

olikraus, 2023-01-19:
I will extend the "E" fonts with the subscript characters.

ken830, 2023-01-19:
That'll be great, thank you! Yesterday, I already created a custom font for my own application and it's working out great! Love u8g2!

yiyueSir, 2023-01-26:
> That'll be great, thank you! Yesterday, I already created a custom font for my own application and it's working out great! Love u8g2!

Brother, have you solved this problem? How can C0 ₂ be displayed

ken830, 2023-01-27:
> > That'll be great, thank you! Yesterday, I already created a custom font for my own application and it's working out great! Love u8g2!
> 
> Brother, have you solved this problem? How can C0 ₂ be displayed

Yeah! Just have to create a new font though and include it in your code. Creating a simple font is super-easy because I found someone created a web-based font generator [http://www.kidsgo.net/u8g2/](url). It was intended for adding missing CJK characters, but you can just paste any unicode in there. I created a font with the subscript 2 and the degree symbol (so I can use just one font for everything). The website will spit out the font. Save it in a .h file and #include it before you use it in your u8g2 routines.

olikraus, 2023-01-27:
I think include is not required. If the font array is small enough, you can just copy & paste the same into your .ino file.

olikraus, 2023-06-06:
I think, this could be closed, once the font rerun had happend (maybe with https://github.com/olikraus/u8g2/issues/2185 )

olikraus, 2023-06-13:
font pictures updated for all *te.png fonts

closed 2023-06-13 by olikraus

ken830, 2023-06-13:
Hmm.. I thought I commented here last week but I don't see it.

I don't see 0x2082 (8322 dec) in the png or the ttf. Is it something I'm missing?

olikraus, 2023-06-13:
where did you look? Which font?

ken830, 2023-06-13:
I think I mis-understood. I thought you are specifically talking about this being resolved with font in #2185, but re-reading this, I think I now understand that #2186 is just triggering the "font rerun?"

olikraus, 2023-06-14:
Yes, I triggered the rerun yesterday and the subscript chars should be now available in the beta release of u8g2 (of course only in cases where the subscript chars are part of the font itself).

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

ken830, 2023-06-14:
Thank you and sorry about my initial confusion!

olikraus, 2023-06-14:
no problem