olikraus/u8g2 #2081: ST7539-G4 - Display 192x64

furcipus, 2023-01-19:
Hello Oliver, 

i need to use a display with a ST7539-G4 Displaycontroller.
Please help me to get the display running.

I got following code from the supplier:

```
void init_lcd()
{
	wr_cmd(0xE2);	// RESET
	wr_cmd(0xA0);	// 76HZ
	wr_cmd(0xEB);	//BIAS SET 1/9
	wr_cmd(0xC2);	//COM0~COM63,SEG191~SEG0
	wr_cmd(0x81);	//VOP SET
        wr_cmd(145);	//VOP
        wr_cmd(0xB0);
        wr_cmd(0x10);
        wr_cmd(0x00);
	 wr_cmd(0x40);
	wr_cmd(0xaf);	//display on
	delay_ms(100);
}

```

In the attachment is the datasheet of the Displaycontroller.
My Display has a resolution of 192x64 pixel. (datasheet in the attachment)

best regards
Harald
[ST7539i2-G4_V1.0.pdf](https://github.com/olikraus/u8g2/files/10456775/ST7539i2-G4_V1.0.pdf)
[SG19264FRB-FPMNDW.pdf](https://github.com/olikraus/u8g2/files/10456777/SG19264FRB-FPMNDW.pdf)

olikraus, 2023-01-22:
I am a little bit busy at the moment with a lot of other topics. I can try to add support, but I would need your help to test this, includig some pictures of the output. Would this be ok?

furcipus, 2023-01-23:
Thank you, of course i will help you and do my best to support you.

olikraus, 2023-06-12:
Is there still a need for supporting this display?

olikraus, 2023-06-12:
Seems to be command compatible to the UC1601 controller. cmd table updated.

olikraus, 2023-06-12:
Looks like this is already implemented with a different issue: https://github.com/olikraus/u8g2/issues/2146

closed 2024-08-24 by olikraus