olikraus/u8g2 #2085: ESP8266 resetting/hangs with 128x128 OLEDs with 1327 controller over I2C

JayWayAr, 2023-01-28:
I have tried Wemos D1 Mini te get to run Hello World and GraphicsTest on 2 different 128x128 1.5 inch OLEDs with the 1327 controller over I2C. One OLED is the generic one as sold by TinyTronics.nl. The other one is the Waveshare 1.5 16 gray scale converted to I2C. 

Hello World keeps resetting the ESP8266 (blue led flashing) after displaying the "Hello World" text (OLED goes black and then displays the text again). GraphicsTest only displays first page indefinitely, does not run any further. Vendor tech rep is able to reproduce this for the generic OLED on NodeMCU. I have gotten several Full buffer and Page buffer constructors to "work", both with HW and SW I2C implementation. Other than the speed of displaying, same issue: reset/hangs. 

I have done the same with the 128x64 0.96 inch OLEDs with the 1306 controllers, works flawlessly. 

Nothing else is connected to the D1 mini. It is not the hardware. Multiple constructors "worked", i.e., the I2C address was correct and something was displayed.

It must be something of the ESP8266 (I have read it has no HW I2C) that is causing this. 

Thanks for the hard work and I'm looking forward to any advice (pins for SPI are already taken).

JayWayAr, 2023-01-29:
I got the Adafruit SSD1327 library to work with the Waveshare OLED over I2C (using I2C address 0x3C). In the [Adafruit_SSD1327.cpp](https://github.com/adafruit/Adafruit_SSD1327/blob/master/Adafruit_SSD1327.cpp) file I found this remark in the .display() method:

  // ESP8266 needs a periodic yield() call to avoid watchdog reset.
  // With the limited size of SSD1327 displays, and the fast bitrate
  // being used (1 MHz or more), I think one yield() immediately before
  // a screen write and one immediately after should cover it.  But if
  // not, if this becomes a problem, yields() might be added in the
  // 32-byte transfer condition below.

I think this might be the issue. Thank you to Limor Fried/Ladyada for commenting. Hope I do not infringe on any copyrights. 
I hope this can lead to the library being updated to work with the ESP8266. Being a cut-and-paster myself, I do not trust me to contribute this, sorry.

JayWayAr, 2023-02-02:
Vendor tech support tried a workaround and added a yield(); statement in u8g2_buffer.c in line 86. 

do
  {
    yield(); // added yield() statement for ESP8266
    u8g2_send_tile_row(u8g2, src_row, dest_row);
    src_row++;
    dest_row++;
  } while( src_row < src_max && dest_row < dest_max );

This prevents the ESP8266 from crashing with the 1327 controller, but the refresh rate is now over 1 sec for the ESP at 80 MHz, which is not the case for the Adafruit library. So the statement is probably in the wrong plasce. Anyhow, it is useful as a proof that this is a software/timing issue. 

@olikraus , do you see any chance for a quick fix? Thank you again for the hard work. Regards.

olikraus, 2023-02-03:
I assume, that the problem can be solved if you call yield() inside your own picture loop, maybe before the next page statement.
``` C++
void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB10_tr);
    u8g2.drawStr(0,24,"Hello World!");
    yield();
  } while ( u8g2.nextPage() );
}
```

I have not looked much into the problem, but I assume this is a watchdog issue. Maybe you disable the watchdog or increase watchdog timeout. Some details might be here: https://sigmdel.ca/michel/program/esp8266/arduino/watchdogs_en.html