olikraus/u8g2 #2086: ST7567 (jlx12864) STM32f303 connection problem

bondiik, 2023-01-30:
I am trying to connect ST7567 to STM32F3-Discovery (stm32f303). The screen lights up and blinks several times. The image appears and then the screen goes blank.
Please advise what could be causing the problem.

The setting function is correct. I tested it on the Arduino board and it works.

**Сallback functions**
```
uint8_t u8g2_gpio_and_delay_stm32(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{

	switch(msg)
	{
	case U8X8_MSG_GPIO_AND_DELAY_INIT:
		break;

	case U8X8_MSG_DELAY_MILLI:
		HAL_Delay(arg_int);
		break;

	case U8X8_MSG_GPIO_CS:
		HAL_GPIO_WritePin(CS_LCD_PORT, CS_LCD_PIN, arg_int);
		break;

	case U8X8_MSG_GPIO_DC:
		HAL_GPIO_WritePin(OLED_DC_GPIO_Port, OLED_DC_Pin, arg_int);
		break;

	case U8X8_MSG_GPIO_RESET:
		HAL_GPIO_WritePin(OLED_RES_GPIO_Port, OLED_RES_Pin, arg_int);
		break;
	}
	return 1;
}

uint8_t u8x8_byte_stm32_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	switch(msg) {
	case U8X8_MSG_BYTE_SEND:
		/* Insert codes to transmit data */
		if(HAL_SPI_Transmit(&hspi2, arg_ptr, arg_int, 1000) != HAL_OK) return 0;
		break;
	case U8X8_MSG_BYTE_INIT:
		/* Insert codes to begin SPI transmission */

		break;
	case U8X8_MSG_BYTE_SET_DC:
		/* Control DC pin, U8X8_MSG_GPIO_DC will be called */
		u8x8_gpio_SetDC(u8x8, arg_int);
		break;
	case U8X8_MSG_BYTE_START_TRANSFER:
		/* Select slave, U8X8_MSG_GPIO_CS will be called */
		u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);

		HAL_Delay(1);
		break;
	case U8X8_MSG_BYTE_END_TRANSFER:
		HAL_Delay(1);
		/* Insert codes to end SPI transmission */

		u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
		break;
	default:
		return 0;
	}
	return 1;
}

```
**Main function**
```
  u8g2_Setup_st7567_jlx12864_1(&u8g2, U8G2_R0, u8x8_byte_stm32_hw_spi, u8g2_gpio_and_delay_stm32);
  u8g2_InitDisplay(&u8g2);
  u8g2_SetPowerSave(&u8g2, 0);
  u8g2_ClearBuffer(&u8g2);

  while (1)
  {
	  u8g2_FirstPage(&u8g2);
	     do
	     {
	       u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
	       u8g2_DrawStr(&u8g2, 0, 15, "Hello World!");
	       u8g2_DrawCircle(&u8g2, 64, 40, 10, U8G2_DRAW_ALL);
	     } while (u8g2_NextPage(&u8g2));
  }
```

**SPI config**

```
hspi2.Instance = SPI2;
  hspi2.Init.Mode = SPI_MODE_MASTER;
  hspi2.Init.Direction = SPI_DIRECTION_2LINES;
  hspi2.Init.DataSize = SPI_DATASIZE_8BIT;
  hspi2.Init.CLKPolarity = SPI_POLARITY_LOW;
  hspi2.Init.CLKPhase = SPI_PHASE_1EDGE;
  hspi2.Init.NSS = SPI_NSS_SOFT;
  hspi2.Init.BaudRatePrescaler = SPI_BAUDRATEPRESCALER_16;
  hspi2.Init.FirstBit = SPI_FIRSTBIT_MSB;
  hspi2.Init.TIMode = SPI_TIMODE_DISABLE;
  hspi2.Init.CRCCalculation = SPI_CRCCALCULATION_DISABLE;
  hspi2.Init.CRCPolynomial = 7;
  hspi2.Init.CRCLength = SPI_CRC_LENGTH_DATASIZE;
  hspi2.Init.NSSPMode = SPI_NSS_PULSE_ENABLE;
```

bondiik, 2023-01-31:
I continued my research. Very strange thing.
CPOL = 0; CPHA = 0;
When sending InitDisplay() - packets are broken.
But the DrawStr() packages are good.

STM32 (the image appears and then the screen goes blank)
[image]

ESP32 ( ST7567 works)
[image]

bondiik, 2023-02-01:
Solved the previous problem (an error in the logic analyzer). But the display never worked. Maybe someday someone will be able to solve it :)

https://user-images.githubusercontent.com/13630305/216107011-4c798c29-4357-42d8-9a7f-0f9d17d9b287.MOV

closed 2023-02-01 by bondiik

olikraus, 2023-02-03:
oh, still nice you found the issue...

inows, 2023-06-09:
how does you make it work? i have the same prolbem