olikraus/u8g2 #2087: Auto detect display

alextrical, 2023-01-30:
I know this question has been asked before but hoping there may be some progress with U8G2.

Using a I2C display, It is possible to read the registers of an SSD1306,SH1106 and SH1107 to be able to detect presence, driver type, and address. See https://github.com/bitbank2/ss_oled

Is there any way that it could be possible to setup, or create a new display type that follows this kind of functionality with U8G2?

I believe this feature would be of great use to the community, as a group I'm collaborating is having issues with sellers mixing SSD1306 and SH1107 drivers within listings.

alextrical, 2023-01-30:
Further information:
The 3 displays in question can be set to either 0x3C or 0x3D

```
  // find the device address if requested
  if (iAddr == -1 || iAddr == 0 || iAddr == 0xff) // find it
  {
    I2CTest(&pOLED->bbi2c, 0x3c);
    if (I2CTest(&pOLED->bbi2c, 0x3c))
       pOLED->oled_addr = 0x3c;
    else if (I2CTest(&pOLED->bbi2c, 0x3d))
       pOLED->oled_addr = 0x3d;
    else
       return rc; // no display found!
  }
  else
  {
    pOLED->oled_addr = iAddr;
    I2CTest(&pOLED->bbi2c, iAddr);
    if (!I2CTest(&pOLED->bbi2c, iAddr))
       return rc; // no display found
  }
```

and the registers can be read to identify the display in question

```
  // Detect the display controller (SSD1306, SH1107 or SH1106)
  uint8_t u = 0;
  I2CReadRegister(&pOLED->bbi2c, pOLED->oled_addr, 0x00, &u, 1); // read the status register
  u &= 0x0f; // mask off power on/off bit
  if (u == 0x7 || u == 0xf) // SH1107
  {
    pOLED->oled_type = OLED_128x128;
    rc = OLED_SH1107_3C;
    bFlip = !bFlip; // SH1107 seems to have this reversed from the usual direction
  }
  else if (u == 0x8) // SH1106
  {
    rc = OLED_SH1106_3C;
    pOLED->oled_type = OLED_132x64; // needs to be treated a little differently
  }
  else if (u == 3 || u == 6) // 6=128x64 display, 3=smaller
  {
    rc = OLED_SSD1306_3C;
  }
  if (pOLED->oled_addr == 0x3d)
     rc++; // return the '3D' version of the type
```

https://github.com/bitbank2/ss_oled/blob/01fb9a53388002bbb653c7c05d8e80ca413aa306/src/ss_oled.cpp#L810-L830

closed 2023-01-30 by alextrical

olikraus, 2023-02-03:
Nice...