olikraus/u8g2 #2091: I have a 144 x 32 LCD driven by a ST7920 chip. (Tight physical-height requirement, displays is 36mm tall, all the room I have). I've used u8g before, but never had this particular problem: how to specify/set display size in pixels.

olikraus, 2023-02-03:
I have a 144 x 32 LCD driven by a ST7920 chip. (Tight physical-height requirement, displays is 36mm tall, all the room I have). I've used u8g before, but never had this particular problem: how to specify/set display size in pixels.

There's `getDisplayHeight()` and `getDisplayWidth()` methods; are there corresponding `set` methods?

Or should I just choose a constructor for a differently geometry, eg. 160 x 32, and restrict my horizontal addresses to 144? Eg. `U8G2_ST7920_160X32_F_6800(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [full framebuffer, size = 640 bytes]`?

SORRY newbie error: somehow clicked POST with incomplete subject line. No way to edit subject nor delete post.

_Originally posted by @tomjennings in https://github.com/olikraus/u8g2/discussions/2084_

craigmoates, 2023-04-17:
We also have this 144 x 32 on ST7920 via SW SPI. I've tried to patch in values but have not succeeded yet. I'm looking now on how to rebuild via codebuild, and saw that others also have a need for this new constructor. Thank you!

olikraus, 2023-06-12:
Is this topic still open? Can you do the testing, if I would do the code build changes?

tomjennings, 2023-06-12:
Unfortunately I managed to use a more conventional LCD module and I don't have a good system to check this with. Sorry.

olikraus, 2023-06-14:
I have created beta 2.34.20 which includes the constructor U8G2_ST7920_144X32...

It would be great if someone could test this. 

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

craigmoates, 2023-06-17:
Tested with several examples but still not working correctly. Let me know what feedback may be helpful. Compiled OK but has same behavior as 160x32 construct it seems. The 144x32 has 18 instead of 20 columns, but other than that they should respond similarly. Here is a code example (tested good): https://www.crystalfontz.com/products/document/2999/CFAG14432_SPI_example.zip

olikraus, 2023-06-18:
Which u8g2 code did you use for testing? How will it look on the display? Any picture of the current u8g2 output?

craigmoates, 2023-06-18:
Arduino IDE 2.1.0 Mega2560 U8g2_2.34.20 results shown. Ready for other testing as needed.

[image]
[image]
[image]

https://github.com/olikraus/u8g2/assets/18092204/7d997bd0-f2a1-4260-9baf-7ddd9524cca3

olikraus, 2023-06-18:
Thanks for the pics and video.
To me it looks like that there is some noise on the signal lines or some other wiring issue.

craigmoates, 2023-06-18:
The other example which I linked above for your reference works fine, also SW SPI. Here is a video, same setup, just uploaded.

https://github.com/olikraus/u8g2/assets/18092204/8917af37-a31d-483c-8206-1f1dfabab5d6

olikraus, 2023-06-18:
So HW SPI works, but SW SPI doesn't ?

craigmoates, 2023-06-18:
These examples shown, including the working Crystalfontz one above, are SW SPI. No HW SPI here. Here is a section of their code, including bus timing comments:

#define FAST_SPI 1

//FAST_SPI = 0: The SS pin is high for ~429uS (24-bits)
//              The clock frequency is ~57KHz
//              Full graphic update takes ~285mS
//FAST_SPI = 1: With no delays, the SS pin is high for ~12uS (24-bits)
//              The clock frequency is ~2MHz
//              But that is too fast for the ST7920, so slow it down
//              with the additional _delay_us(1) calls, which makes:
//              The SS pin is high for ~39uS (24-bits)
//              The clock frequency is ~615KHz
//              Full graphic update takes ~22.5mS

//Software SPI (10-bit transfers, difficult to do using the hardware SPI)
#define SPIPORT (PORTB)
#define SPITOGGLE (PINB)

olikraus, 2024-08-24:
looks very complicated...