olikraus/u8g2 #2092: The 128*240 resolution of ssd1322 is not supported in u8g2 library. How to modify it to make u8g2 support it?

olikraus, 2023-02-03:
The 128*240 resolution of ssd1322 is not supported in u8g2 library. How to modify it to make u8g2 support it?

_Originally posted by @1921096212 in https://github.com/olikraus/u8g2/discussions/2089_

1921096212, 2023-02-06:
> 我会调查这个问题。 http://www.hpinfotech.ro/SSD1322.pdf

ok

1921096212, 2023-02-06:
> I will look into this. http://www.hpinfotech.ro/SSD1322.pdf
ok

1921096212, 2023-02-06:
> I will look into this. http://www.hpinfotech.ro/SSD1322.pdf
[image]

By the way, my screen is 4.7 inches, 240 x 128

olikraus, 2023-02-19:
Excuse for the delay with this topic. I simply didn't found time to work on this. It least it is there on the todo list for the next release.

olikraus, 2023-02-22:
I have add a new constructor: U8G2_SSD1322_240X128...
There is also a new beta release 2.34.14 which includes the new code.
Can you download and test this code and constructor?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

1921096212, 2023-03-01:
These days, I transplanted to m4 platform and tested.  I drew two lines and found that x=96~240 and y=0~63,
x needs to be added 96, y should be 0~127, but it becomes 0~63.  The following is my code and effect drawing.

My English is poor, the above is the software translation...

1921096212, 2023-03-01:
These days, I transplanted it to m4 platform and conducted a test.
I drew two lines and found that x=96-240, y=0-63,x needed to be added 96, y should be 0-127, but it became 0-63.
The following is my code and effect drawing.

1921096212, 2023-03-01:
> These days, I transplanted it to m4 platform and conducted a test. I drew two lines and found that x=96-240, y=0-63,x needed to be added 96, y should be 0-127, but it became 0-63. The following is my code and effect drawing.

My English is poor, the above is the software translation...

1921096212, 2023-03-01:
`    u8g2_DrawLine(&u8g2_Handler, 96, 0, 240, 63);
    u8g2_DrawLine(&u8g2_Handler, 96, 63, 240, 0);`

1921096212, 2023-03-01:
> ` u8g2_DrawLine(&u8g2_Handler, 96, 0, 240, 63); u8g2_DrawLine(&u8g2_Handler, 96, 63, 240, 0);`

[image]

olikraus, 2023-03-05:
To change the x offset please locate file u8x8_d_ssd1322.c on your local drive.
Then change 0x000 to 96 in the following line: 

https://github.com/olikraus/u8g2/blob/0319996fef65f6f4dc7bbc76fab1da9f8ce0a6df/csrc/u8x8_d_ssd1322.c#L357

Will this help?

Can you also run the flipmode example and send a picture of the first flipmode picture which is generated?

1921096212, 2023-03-06:
I modified the offset and can display full screen, but the display is abnormal.
`default_x_offset = */ 24`

Below I show two built-in 16 by 16 ICONS, but they look like 16 by 32.

[image]

olikraus, 2023-03-07:
Maybe we have to disable the dual com mode here:
Instead of
https://github.com/olikraus/u8g2/blob/4d9f63394f070a9f6e079ae8bccb97bad5967481/csrc/u8x8_d_ssd1322.c#L377
it might be
U8X8_CAA(0xa0, 0x06, 0x001),	/* Set Re-Map / Dual COM Line Mode */  

How will it look with this setup?
BTW: Is there any datasheet with a default init sequence for this display?

1921096212, 2023-03-10:
Thank you very much for this open source library, I modified the following parameters, now can display normally. I'm continuing to test to see if there are any other problems.
`U8X8_CAA(0xa0, 0x36, 0x001), /* Set Re-Map / Dual COM Line Mode */`

[image]

olikraus, 2023-03-10:
Looks good already. I will apply the change to the repo. 

A good test would be the execution of the FlipMode.ino example (See example menu in Arduino IDE or [here](https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/FlipMode/FlipMode.ino)). The flip mode test example will cycle through eight modes. In all modes there must be a solid frame next to the border of the display.

closed 2023-06-12 by olikraus