olikraus/u8g2 #2098: Multiple SPI Interfaces with u8g2

martinmmi, 2023-02-11:
Hello, iam martin. I working with lora, old display and ethernet IoT and i have a question to that lib.

The lora shield and the display working **both with spi** and i want to try to work with two spi slaves.

For this i shared the hw spi interface mosi, misom sclk. For cs i used two different and actived cs when i using one of the two spi moduls.

The spi for the lora shield is actived with:
SPI.begin(LORA_SCLK, LORA_MISO, LORA_MOSI, LORA_CS);

When i used the oled constructor:
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2
before, the wireing is not okay for the lora shield.

**How can i work with multiple spi interfaces?**

For lora, i used the sandeepmistry lib.
[sandeepmistry](https://github.com/sandeepmistry/arduino-LoRa)

Many thanks!

svdrummer, 2023-02-15:
Same pins, different CS pin for each

olikraus, 2023-02-15:
It should work if the SPI Configuration is the same for both devices. You might also need to reduce the SPI bus clock.

martinmmi, 2023-02-19:
Hey thank you @svdrummer and @olikraus. I seted the CS for each pin, but it didnt work. 

@olikraus do you have an recommand for a good slower spi clock? I have no idea for it.

I found this command for reduce the spi clock: SPI.setFrequency(15000000); For example, here my setup code to try to initialized a sd-card reader and the lora modul. The OLED is excluded, because i wait for an monochrome spi oled.

`void setup() {

    Serial.begin(115200);
    while (!Serial);

    pinMode(DISPLAY_CS, OUTPUT);                                //CS Display
    pinMode(SD_CS, OUTPUT);                                     //CS SD-Reader
    pinMode(LORA_MISO, INPUT_PULLUP);

    SPI.begin(LORA_SCLK, LORA_MISO, LORA_MOSI, LORA_CS);        //Begin and CS LORA
    SPI.setFrequency(15000000);

    Serial.println("");
    Serial.println(name);

    digitalWrite(SD_CS, LOW);              //Select SDCard SPI Device

    while (true) {
            if (SD.begin(SD_CS)) {
                Serial.println("SDCard init succeeded.");
                delay(300);
                break;
            }
            Serial.println("SDCard init failed. Check your connections.");
            delay(300);
            break;
        }

    digitalWrite(SD_CS, HIGH);              
    digitalWrite(DISPLAY_CS, LOW);          //Select Display SPI Device

    //u8g2.begin();
    //u8g2.clearBuffer();
    //u8g2.setFont(u8g2_font_6x10_tf);
    //u8g2.setContrast(defaultBrightnessDisplay);                  
    //u8g2.setFlipMode(1);

    //        Color, Delay, Runs
    printLogo(0, 50);
    delay(1000);

    printLoad(1, 60, 2);

    printLogo(0, 25);
    delay(500);

    printLoad(1, 60, 4);

    Serial.println("Version: " + version);
    sprintf(buf_version, "%s", version);
    //u8g2.drawStr(99,60,buf_version);
    //u8g2.sendBuffer();

    Serial.println("OLED init succeeded.");
    oledInit = "OLED init";
    sprintf(buf_oledInit, "%s", oledInit);
    //u8g2.drawStr(0,15,buf_oledInit);
    //u8g2.sendBuffer();
    delay(300);

    digitalWrite(DISPLAY_CS, HIGH);    
    digitalWrite(LORA_CS, LOW);          //Select LORA SPI Device      

    // override the default CS, reset, and IRQ pins (optional)
    LoRa.setPins(LORA_CS, LORA_RST, LORA_IRQ); // set CS, reset, IRQ pin
    LoRa.setTxPower(17);  //2-20 default 17
    LoRa.setSpreadingFactor(7);    //6-12 default 7
    LoRa.setSignalBandwidth(125E3);   //7.8E3, 10.4E3, 15.6E3, 20.8E3, 31.25E3, 41.7E3, 62.5E3, 125E3, 250E3, and 500E3 default 125E3
    LoRa.setCodingRate4(5);   //5-8 default 5
    LoRa.setPreambleLength(8);    //6-65535 default 8
    LoRa.begin(868E6);  //set Frequenz 915E6 or 868E6

    if (!LoRa.begin(868E6)) {             // initialize ratio at 868 MHz
        Serial.println("LoRa init failed. Check your connections.");
        loraInit = "LoRa failed";
        digitalWrite(LORA_CS, HIGH);       
        digitalWrite(DISPLAY_CS, LOW);        //Select Display SPI Device   
        sprintf(buf_loraInit, "%s", loraInit);   
        //u8g2.drawStr(0,35,buf_loraInit);
        //u8g2.sendBuffer();
        digitalWrite(DISPLAY_CS, HIGH);     
        while (true);                       // if failed, do nothing
    }

    digitalWrite(LORA_CS, HIGH);       
    digitalWrite(DISPLAY_CS, LOW);      //Select Display SPI Device   

    Serial.println("LoRa init succeeded.");
    loraInit = "LoRa init";
    sprintf(buf_loraInit, "%s", loraInit);   
    //u8g2.drawStr(0,35,buf_loraInit);
    //u8g2.sendBuffer();
    delay(300);

    pinMode(gpioP1, INPUT_PULLDOWN);
    pinMode(gpioP2, INPUT_PULLDOWN);
    pinMode(gpioP3, INPUT_PULLDOWN);
    pinMode(gpioP4, INPUT_PULLDOWN);

    Serial.println("Outputs init succeeded.");
    outputInit = "Outputs init";
    sprintf(buf_outputInit, "%s", outputInit);   
    //u8g2.drawStr(0,45,buf_outputInit);
    //u8g2.sendBuffer();
    delay(500);

    printLora(1);
    delay(2500);

    emptyDisplay();
    printDisplay();

    digitalWrite(DISPLAY_CS, HIGH);

    if(!SPIFFS.begin(true)){
        Serial.println("An Error has occurred while mounting SPIFFS");
        return;
    }
    Serial.println("SPIFFS init succeeded.");

    WiFi.onEvent(WiFiEvent);

    pinMode(NRST, OUTPUT);

    digitalWrite(NRST, 0);
    delay(200);
    digitalWrite(NRST, 1);
    delay(200);
    digitalWrite(NRST, 0);
    delay(200);
    digitalWrite(NRST, 1);

    ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);

    // Use static ip address config
    IPAddress local_ip(192, 168, 1, 128);
    IPAddress gateway(192, 168, 1, 1);
    IPAddress subnet(255, 255, 0, 0);

    ETH.config( local_ip,
                gateway,
                subnet
                // IPAddress dns1 = (uint32_t)0x00000000,
                // IPAddress dns2 = (uint32_t)0x00000000
                );

    if (MDNS.begin("redtally")) {
        Serial.println("MDNS responder started.");
    }

    ArduinoOTA.begin();
 

    server.on ("/", []() {
        if (!server.authenticate(username, password)) {
            return server.requestAuthentication();
        }
        handleRoot();
    });

    server.on("/inline", []() {
        if (!server.authenticate(username, password)) {
            return server.requestAuthentication();
        }
        server.send(200, "text/plain", "this works as well");
    });

    server.onNotFound(handleNotFound);

    server.begin();
    Serial.println("HTTP server started.");

    }
`

olikraus, 2023-02-19:
Try https://github.com/olikraus/u8g2/wiki/u8g2reference#setbusclock

Apply 1000000 or lower.

martinmmi, 2023-02-25:
hey oli,

unfortunately my LoRa Lib just doesn't work with the 0.96 inch OLED. Not until I totally exclude the u8g2 lib. I don't understand why I have such problems with Multiple SPI. In the setup everything seems to be initialized properly, but later my SPI-LORA shield just doesn't work anymore.

Is there a way to turn off u8g2 completely after u8g2.begin() and later print? Maybe with u8g2.end()?

olikraus, 2023-02-25:
U8g2 will just call the Arduino SPI procedures. It might be required to re-init Lora instead.

martinmmi, 2023-02-25:
I have found that if I do the following then my lora communication will work.

    SPI.begin(LORA_SCLK, LORA_MISO, LORA_MOSI);                 
    digitalWrite(LORA_CS, LOW);

    sendMessage(outgoing);

    digitalWrite(LORA_CS, HIGH);
    SPI.end();

    delay(2000);

    counter++;

    SPI.begin(DISPLAY_SCLK, DISPLAY_MISO, DISPLAY_MOSI);           
    digitalWrite(DISPLAY_CS, LOW);

    sprintf(buf_tx, "%d", counter);   
    u8g2.clear();
    u8g2.drawStr(0,45,buf_tx);
    u8g2.sendBuffer();

    digitalWrite(DISPLAY_CS, HIGH);
    SPI.end();

Is it possible to separate the two spi buses without using sw spi or i2c?

martinmmi, 2023-02-25:
Which values i have to set for the oled display?

SPISettingsSPI.beginTransaction();
SPI.endTransaction();
SPI.beginTransaction();
SPI.beginTransaction();
SPISettingsSPI.setBitOrder();
SPI.setClockDivider();
SPI.setDataMode()

olikraus, 2023-02-26:
It depends on the display itself. The code is here:
https://github.com/olikraus/u8g2/blob/3d41860b4308d60a920f56b65e1b01c155fdfe81/cppsrc/U8x8lib.cpp#L811-L835