olikraus/u8g2 #2100: U8X8_SH1106_128X64 DRAW LARGE BITMAP?

Beppi4U, 2023-02-16:
Hi! 
Is it possible to draw large Bitmap image with U8x8?
Have already included U8x8 for Text but want to show a big Bitmap at start as a welcome screen!
Use an Arduino Nano as Board
There should be 4.5 Kbit left for the Bitmap after Sketch compiling.
Isnt enough space for another Lib.
Thanx for Helping!
Andreas

olikraus, 2023-02-19:
U8g2 is there for bitmap graphics. Of course you could also use u8x8 for graphics, but then you need to use the draw tile function and break down you picture into 8x8 tiles.

Beppi4U, 2023-02-22:
Hi!
Na... not an option!
I tried to use both lib's and now i am afraid all i made is a bunch of Spaghetti code!
Do i have to init the U8g2 to show it? Because it compiles and waits 3sec. but does'nt show the Bitmap!
Would you min d to throw an eye on it?
Rest works fine. 
Greetings Andi 
Sorry for ugly coding....my bad!

`
#include <Wire.h>
#include <Arduino.h>
#include <U8x8lib.h>
#include "U8glib.h"
#include "AVL.h"
#ifdef U8x8_HAVE_HW_I2C
 
#endif
U8GLIB_SH1106_128X64 u8g(U8G_I2C_OPT_NONE);  // I2C / TWI 
U8X8_SH1106_128X64_NONAME_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE);  // DEFINITION DES DISPLAYS UND DER SCHNITTSTELLE
 
#define button1    3                       // Button B1 is connected to Arduino pin 9 CHANGE BUTTON 
#define button2    2                       // Button B2 is connected to Arduino pin 8 SET BUTTON 
// DER DRITTE BUTTON (RESET BUTTON) IST DIREKT AUSGEFÜHRT RST TO GND 
 
void setup(void) {
  pinMode(button1, INPUT_PULLUP);
  pinMode(button2, INPUT_PULLUP);
  delay(500);
  
  draw(); // SHOW AVL LOGO ON SCREEN
  delay(2999);
  
  u8x8.begin(); 
  u8x8.clear();
   
  // VORSCHAU
  u8x8.clearDisplay();
  u8x8.setFont(u8x8_font_inr21_2x4_r);
  fadeinout(0, 2, "FTIR i60", 255);
  u8x8.setFont(u8x8_font_pcsenior_r);
  u8x8.drawString(0, 6,"___");
  delay(250);
  u8x8.setFont(u8x8_font_inr21_2x4_r);
  fadeinout(0, 2, "FTIR i60", 255);
  delay(10);
  u8x8.setFont(u8x8_font_pcsenior_r);
  u8x8.drawString(0, 6,"______");
  delay(250);
  u8x8.drawString(0, 6,"_________");
  delay(250);
  u8x8.setFont(u8x8_font_inr21_2x4_r);
  fadeinout(0, 2, "FTIR i60", 255);
  u8x8.setFont(u8x8_font_pcsenior_r);
  u8x8.drawString(0, 6,"________________");
  delay(2500);
 
  
  u8x8.clear();
  u8x8.setContrast(255);
  u8x8.setFont(u8x8_font_pcsenior_r); //  (u8x8_font_5x7_f)
  u8x8.drawString(0,1," ZUR ERINNERUNG");
  u8x8.drawString(0,2,"   AN DEINE ");
  u8x8.drawString(0,3," KOLLEGEN BEI");
  u8x8.drawString(0,4,"  DER ******* ");
  u8x8.drawString(0,5,"  MESSTECHNIK! ");
  u8x8.drawString(0,6," 23.FEBRUAR 2023");
  delay(5000);
  u8x8.clear();
  u8x8.setFont(u8x8_font_pcsenior_r); //  (u8x8_font_5x7_f)
[... 280 lines cut ...]
  0x80, 0x00, 0x3f, 0xff, 0x03, 0xfc, 0x01, 0xe0, 0x00, 0x00, 0x7f, 0xbe, 0xa8, 0x0e, 0x00, 0x01, 
  0x80, 0x00, 0x3f, 0xff, 0x81, 0xfc, 0x01, 0xe0, 0x00, 0x00, 0x67, 0xd5, 0xf8, 0x00, 0x00, 0x01, 
  0x80, 0x00, 0x7f, 0xff, 0x81, 0xf8, 0x01, 0xe0, 0x00, 0x00, 0x61, 0xd0, 0xfe, 0x00, 0x00, 0x01, 
  0x80, 0x00, 0x78, 0x07, 0x80, 0xf8, 0x01, 0xe0, 0x00, 0x00, 0x21, 0xd0, 0xc0, 0x00, 0x00, 0x01, 
  0x80, 0x00, 0xf8, 0x03, 0xc0, 0xf8, 0x01, 0xff, 0xfc, 0x00, 0x31, 0xc0, 0xbf, 0x00, 0x00, 0x01, 
  0x80, 0x00, 0xf0, 0x03, 0xc0, 0xf0, 0x01, 0xff, 0xfc, 0x00, 0x19, 0xc1, 0x7f, 0x80, 0x00, 0x01, 
  0x80, 0x01, 0xf0, 0x03, 0xe0, 0x70, 0x01, 0xff, 0xfc, 0x00, 0x0f, 0x80, 0xf3, 0xc0, 0x00, 0x01, 
  0x80, 0x01, 0xf0, 0x01, 0xe0, 0x60, 0x01, 0xff, 0xfc, 0x00, 0x07, 0x01, 0xe0, 0xc0, 0x00, 0x01, 
  0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xc1, 0x80, 0x00, 0x01, 
  0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xc3, 0x00, 0x00, 0x01, 
  0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x7c, 0x00, 0x00, 0x01, 
  0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
  0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
  0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
  0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
  0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
  0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
  0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
  0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
  0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
  0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
  0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
  0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff
};
`

olikraus, 2023-02-22:
Ok, I stopped reading after the first view lines: do no use old u8g lib and u8x8 together. Use u8g2 only. U8g2 includes u8x8 (Use u8g2.getU8x8)

Beppi4U, 2023-02-23:
Mixing up libs is never a good idea.
Thanx for help! 
Gonna try u8g2 out.