olikraus/u8g2 #2107: Display still dark on ESP8266 with 128x64 display

lorenzoPrimi, 2023-02-20:
Hi, 
I'm trying to run some code on my ESP8266 with 128x64 display but after many tests, it is always dark.
I successfully ran some code with [another library](https://github.com/ThingPulse/esp8266-weather-station/blob/master/examples/WorldClockDemo/WorldClockDemo.ino), and managed to see something on the screen:
```
const int I2C_DISPLAY_ADDRESS = 0x3c;
const int SDA_PIN = D3;
const int SDC_PIN = D4;

// Initialize the oled display for address 0x3c
// sda-pin=14 and sdc-pin=12
SSD1306Wire  display(I2C_DISPLAY_ADDRESS, SDA_PIN, SDC_PIN);
OLEDDisplayUi ui     ( &display );
```

So now I'm trying to adapt that code to this library.
I checked [this issue](https://github.com/olikraus/u8g2/issues/2045) and added this code, but the screen is still dark 

```
#define SDA 0 // Pin 0 = D3
#define SCL 2 // // Pin 2 = D4
U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, SCL, SDA, U8X8_PIN_NONE);
```
What am I doing wrong? Conceptually it seems correct.

olikraus, 2023-02-20:
Indeed, looks all correct

lorenzoPrimi, 2023-02-20:
Now works, I was doing something wrong in another part of the code. Thanks

closed 2023-02-20 by lorenzoPrimi

olikraus, 2023-02-20:
👍