olikraus/u8g2 #2115: SSD1317 64x48 I2C OLED

dejavu1987, 2023-02-25:
Hi,
I just bought[ this tiny screen from Aliexpress](https://de.aliexpress.com/item/1005002760899272.html?spm=a2g0o.order_list.order_list_main.123.21ef5c5fIuFqsW&gatewayAdapt=glo2deu) to realize that there is not much mention of this driver anywhere.
[image]

I am ready to help in anyways to add it to the library but I am not very experienced with it yet.
I have it connected to clock= 9, data=8, reset=NONE, *the module does have a reset pin which I haven't used.*
I have tried the hello world sketch with several constructors
```
U8G2_SSD1317_96X96_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ 9, /* data=*/ 8, /* reset=*/ U8X8_PIN_NONE);
U8G2_SSD1312_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ 9, /* data=*/ 8, /* reset=*/ U8X8_PIN_NONE);
U8G2_SSD1312_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
```
None of which worked.
Just to see, I swapped the display with a 128x32 SSD1306 it showed a hello world but mirrored and some stray pixels.
[image]

dejavu1987, 2023-02-25:
So I just ate an apple 🍎 and thought I would connect EN to Reset of the oled and it seem to bring it into life.
currently using
```U8G2_SSD1306_64X48_ER_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);```
[image]

closed 2023-02-26 by dejavu1987

olikraus, 2023-02-26:
looks good :-)

olikraus, 2023-04-10:
Use any regular GPIO pin for reset (RST) pin. See also here: https://github.com/olikraus/u8g2/wiki/setup_tutorial

dejavu1987, 2023-04-16:
@fabrice1236 yes that is exactly what I did

> Hey @dejavu1987, complete newbie here. I want to buy the same screen from aliexpress to use with my arduino pro micro, but I don't completely understand how you got it to work, did it just start working when you connected the screen's RST pin to your S2 mini's EN pin, or am I mistaken ? if you could explain a little more clearly, I would be very grateful.

fabrice1236, 2023-05-08:
Hey, so I've bought the screen, but I can't get it to work with my Arduino Due, I'm using `U8G2_SSD1306_64X48_ER_1_HW_I2C u8g2(U8G2_R0, 8)`.. When I run a basic sketch that looks for I2C devices, I can only see the display for 5-10 secs after uploading the code/powering the arduino, then nothing.. I've tried connecting the RST pin of the display to pin 8, but it didn't change anything.. Do you have any idea of how I could fix this ? Also, I am using SDA1 and SCL1, which means I'm using Wire1, if that changes something.