olikraus/u8g2 #2119: How to use SPI2 or SPI3 in Bluepill?

PadmaIoT, 2023-02-27:
Hello,

1st of all I am describing my setup:

**MCU Board:** Stm32 Bluepill
**Environment:** Arduino IDE
**Connection type:** SPI (HW_SPI)

Bluepill or STM32F103C8T6 has multiple physical SPI pins. If I use  
`U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0, /* CS=*/ 10, /* reset=*/ 8); ` 
this one and configure CS & reset for SPI2 then the program does not work. I also tried SW it didn't work either for **SPI2.** **How to run my code to display something by **SPI 2**?**

Note: for the older version (U8glib) it was working just by only stating the SPI2 pins in the below line 
`U8GLIB_ST7920_128X64_1X u8g(PB13, PB15, PB12);  // SPI Com: SCK = en = 18, MOSI = rw = 16, CS = di = 17. `

olikraus, 2023-02-27:
You could use the u8g2 SW SPI constructor . This should be identical to the mentioned u8glib constructor.

PadmaIoT, 2023-03-02:
It works. Thanks

closed 2023-03-02 by PadmaIoT