olikraus/u8g2 #2121: ST7529 240x160 Display Support?

WaldobertDE, 2023-02-27:
Hey,
I was able to salvage some "Topway LM240160DCW" Displays. 

Can I use the `U8G2_UC1611_EW50850_F_8080 u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7,  /*enable=*/ 18, /*cs=*/ 3, /*dc=*/ 16, /*reset=*/ 17); // 240x160, Connect RD/WR1 pin with 3.3V, CS is aktive high` Constructor? (It is the only one with this size)
How do I connect the Pins to my ESP32? 
Is (Pin 8 -> DB0), (Pin 9 -> DB1) etc. the right order?

Thanks

Datasheet: https://www.topwaydisplay.com/sites/default/files/2020-02/LM240160DCW.pdf

Pins are as following:
| Pin No. 	| Pin Name 	| I/O   	| Descriptions                                                                                            	|
|---------	|----------	|-------	|---------------------------------------------------------------------------------------------------------	|
| 1       	| VDD      	| Power 	| Positive Power Supply                                                                                   	|
| 2       	| VSS      	| Power 	| 0V Supply, Ground (0V)                                                                                  	|
| 3       	| /CS      	| Input 	| Chip Select /CS=LOW : Data IO is enabled                                                                	|
| 4       	| RS       	| Input 	| Register Select RS=HIGH: data on DB0 to DB7 is display data RS=LOW:  data 0n DB0 to DB7 is control data 	|
| 5       	| /WR      	| Input 	| Write enable input, active LOW                                                                          	|
| 6       	| /RD      	| INPUT 	| Read enable input, active LOW                                                                           	|
| 7       	| DB0      	| IO    	| Bi-directional data bus                                                                                 	|
| 8       	| DB1      	| IO    	| Bi-directional data bus                                                                                 	|
| 9       	| DB2      	| IO    	| Bi-directional data bus                                                                                 	|
| 10      	| DB3      	| IO    	| Bi-directional data bus                                                                                 	|
| 11      	| DB4      	| IO    	| Bi-directional data bus                                                                                 	|
| 12      	| DB5      	| IO    	| Bi-directional data bus                                                                                 	|
| 13      	| DB6      	| IO    	| Bi-directional data bus                                                                                 	|
| 14      	| DB7      	| IO    	| Bi-directional data bus                                                                                 	|
| 15      	| /RST     	| Input 	| Reset: /RST=LOW: Initialization is executed /RST=HIGH: Normal                                           	|
| 16      	| BLA      	| Power 	| Backlight Positive Power Supply                                                                         	|

Image:
[image]

olikraus, 2023-02-28:
> How do I connect the Pins to my ESP32?
> Is (Pin 8 -> DB0), (Pin 9 -> DB1) etc. the right order?

Anything is fine, you just need to put the pin numbers in the correct order for the U8g2 Constructor.
My suggestion is to add one more column into your table above and add the ESP32 pin number to which the display port is connected.

Notes:
 -  DC = RS
 - /RD is not used and must be pulled high
 - /WR is the enable signal as mentioned in the constructor

WaldobertDE, 2023-02-28:
> Anything is fine, you just need to put the pin numbers in the correct order for the U8g2 Constructor.
My suggestion is to add one more column into your table above and add the ESP32 pin number to which the display port is connected.

Updated table:

| ESP32 DevBoardC GPIO 	| Display Pin No. 	| Pin Name 	| I/O   	| Descriptions                                                                                            	|
|----------------------	|-----------------	|----------	|-------	|---------------------------------------------------------------------------------------------------------	|
| 3V3                  	| 1               	| VDD      	| Power 	| Positive Power Supply                                                                                   	|
| GND                  	| 2               	| VSS      	| Power 	| 0V Supply, Ground (0V)                                                                                  	|
| 23                   	| 3               	| /CS      	| Input 	| Chip Select /CS=LOW : Data IO is enabled                                                                	|
| 22                   	| 4               	| RS       	| Input 	| Register Select RS=HIGH: data on DB0 to DB7 is display data RS=LOW:  data 0n DB0 to DB7 is control data 	|
| 15                   	| 5               	| /WR      	| Input 	| Write enable input, active LOW                                                                          	|
| 3V3                  	| 6               	| /RD      	| INPUT 	| Read enable input, active LOW                                                                           	|
| 21                   	| 7               	| DB0      	| IO    	| Bi-directional data bus                                                                                 	|
| 19                   	| 8               	| DB1      	| IO    	| Bi-directional data bus                                                                                 	|
| 18                   	| 9               	| DB2      	| IO    	| Bi-directional data bus                                                                                 	|
| 5                    	| 10              	| DB3      	| IO    	| Bi-directional data bus                                                                                 	|
| 17                   	| 11              	| DB4      	| IO    	| Bi-directional data bus                                                                                 	|
| 16                   	| 12              	| DB5      	| IO    	| Bi-directional data bus                                                                                 	|
| 4                    	| 13              	| DB6      	| IO    	| Bi-directional data bus                                                                                 	|
| 0                    	| 14              	| DB7      	| IO    	| Bi-directional data bus                                                                                 	|
| 2                    	| 15              	| /RST     	| Input 	| Reset: /RST=LOW: Initialization is executed /RST=HIGH: Normal                                           	|
| 3V3                  	| 16              	| BLA      	| Power 	| Backlight Positive Power Supply                                                                         	|

> One more point: Is the ST7529 compatible to the UC1611? I mean, I am not sure whether the UC1611 constructor can be used to drive a ST7529 display

My fault. Didn't see the github link to the constructors in the .ino example.
As far as I can see, the display (ST7529?) is not supported by the library. Is there any chance that you will add support for the display?

olikraus, 2023-03-05:
I had a look into the ST7529 datasheet from here: https://www.crystalfontz.com/controllers/Sitronix/ST7529/
It looks very complicated.  At least what I can say: It will be slow. And: U8g2 in general is a monochrome lib, so you have a 32 level grayscale display, but only two levels of all 32 levels will be supported. Moreover, u8g2 still needs to transfer 5 bit for each pixel. 

To add support, I need a similar display with the same controller. There is on option here: https://aliexpress.com/item/1751722020.html but this display has a flex connector only.

All in all it will be difficult and I am not sure whether it will make sense...
Displays with the ST75256 controller hare more common, are available with the same dimensions and are supported by u8g2.