olikraus/u8g2 #2122: No keypress/button event for C language

ZakiyaMiller, 2023-02-28:
am i missing something or there is no key press detect for C language, there is one for C++ in "u8g2.begin( button 1 , 2 ,3, so on ) , is there something similar for C language too ?

olikraus, 2023-03-01:
It is not a question of the language, but instead a question of the environment.
C++ / C do not provide any key detection method. Instead Arduino Environment provides this.
But because Arduino is C++ only, it looks like this is there for C++ only.

I am working on somehting what is called MUI, it will work with C and C++ and will fully rely on external user events.

MUI: https://github.com/olikraus/u8g2/issues/1747

ZakiyaMiller, 2023-03-02:
okay , thanks !