olikraus/u8g2 #2125: for the gallery: GM12864-03A LCD display

JoeLoginIsAlreadyTaken, 2023-03-04:
Here some new display for the gallery.
It's a 128x64 LED display using SPI.
Nice design with mounting holes and metal frame. 
Sells on Aliexpress for ~3€. 

Used with Arduino Micro (32u4) with this Pin assignment & constructor:

/*
 * (PRO) Micro 32u4
 * CS - 10
 * RST - 8
 * A0 - 9
 * SCK - 15 (SCKL)
 * SDA - 16 (MOSI)
 * GND - GND
 * VDD - +5V
 * BA - 0-3.3V (backlight) (use resistor!)
 * BK - GND
 */

U8G2_ST7567_ENH_DG128064I_F_4W_HW_SPI u8g2(U8G2_MIRROR, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8); 

[image]

olikraus, 2023-03-04:
Nice, looks like this display: https://de.aliexpress.com/item/33026003753.html , right?

olikraus, 2023-03-04:
... added to gallery

olikraus, 2023-03-04:
Maybe you can mention the constructor name at aliexpress...

JoeLoginIsAlreadyTaken, 2023-03-04:
Yes, that's the one. There are multiple sellers.
Got mine from here https://www.aliexpress.com/item/1005004574644035.html
I have posted the constructor in my review.

zuoyi001, 2024-02-16:
very nice