olikraus/u8g2 #2129: Support for 240x128 UCI6963C

Rotebay, 2023-03-08:
Hi, I've recently bought a 240x128 lcd display, it was labeled as T6963C IC but later it turned out it's a UCI6963C. Would it be possible to use this display with u8g2 and what constructor should I use? Sorry for my ignorance I'm new to the topic. Thank you for help.

olikraus, 2023-03-08:
Probably you can still use the T6963 constructor.

aselstyne, 2024-07-26:
> Hi, I've recently bought a 240x128 lcd display, it was labeled as T6963C IC but later it turned out it's a UCI6963C. Would it be possible to use this display with u8g2 and what constructor should I use? Sorry for my ignorance I'm new to the topic. Thank you for help.

Hi there, did you get this working?

olikraus, 2024-08-18:
I am quite sure it will work with the T6963 constructor

aselstyne, 2024-10-23:
Confirming for anyone else who may find this in the future, it does indeed work.