olikraus/u8g2 #2133: Using font with different height cause issues with the line height

dhysuiej, 2023-03-14:
I'm using U8x8lib with the Arduino Menu library. Everything was being printed properly on the screen but then I decided to use a bigger font but it seems the line height is not being adjusted properly.

I switched from `u8x8_font_amstrad_cpc_extended_f` to `u8x8_font_8x13_1x2_r` like so:

	display.setFont(u8x8_font_8x13_1x2_r);

And I'm using `setCursor` and `print` so show text on the screen:

	o.setCursor(0, 0);
	o.print("Temp: ");
	o.print(input, 0);
	o.print("/");
	o.print(setPoint, 0);
	o.print("\xb0" "C");

	o.setCursor(0, 1);
	o.print("Heat: ");
	o.print(heat == HIGH ? "On" : "Off");

	o.setCursor(0, 2);
	o.print("Speed: ");
	o.print(speed);
	o.print("%");

	o.setCursor(0, 3);
	o.print("Motor: ");
	o.print(motor == HIGH ? "On" : "Off");

[image]

dhysuiej, 2023-03-14:
Thanks for the quick reply @olikraus.

It looks like someone was having a [similar issue](https://github.com/neu-rah/ArduinoMenu/issues/193) and they solved it by subclassing the display driver and using a [column multiplier from setCursor](https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuIO/SSD1306AsciiOut.h#L50).

I was thinking perhaps it was the display driver, in my case U8x8lib, the one that should figure out the font height and adjust the line height accordingly.

olikraus, 2023-03-14:
👍