olikraus/u8g2 #2135: Please add ELW0901AA to LD7032

iggymayer, 2023-03-19:
https://mega.nz/file/AF0QSAYR#xLxlIQQKIUClCdAZPUkWpPQ64zNi-doQkAz_ZGgTioM
Please add support ELW0901AA,Driver IC:LD7032.
If you can,please support both I2C & SPI interfaces,Thank you!

olikraus, 2023-03-19:
The init sequence in the above datasheet implies, that you will drive the display logic power with 1.8V and you will provide an external 18V power supply for the OLED itself, correct?

olikraus, 2023-03-19:
I have created a new beta release with the new constructor "U8G2_LD7032_128X36..."

Can you check whether the new constructor works?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

iggymayer, 2023-03-20:
I will test it after receiving the PCB next month，Thank you!

olikraus, 2023-06-12:
Any feedback on this?

iggymayer, 2023-06-14:
Set Address 0x6E
Can be lit but cannot be displayed normally.

olikraus, 2023-06-14:
which constructor did you use? which code/example did you run?

iggymayer, 2023-06-14:
> which constructor did you use? which code/example did you run?

> Hello World
#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>

U8G2_LD7032_128X36_F_HW_I2C u8g2(U8G2_R0, 9, 8, 1);

void setup(void)
{
  u8g2.setI2CAddress(0x6E);
  u8g2.begin();
}

void loop(void)
{
  u8g2.clearBuffer();                   // clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);   // choose a suitable font
  u8g2.drawStr(10, 10, "Hello World!"); // write something to the internal memory
  u8g2.sendBuffer();                    // transfer internal memory to the display
  delay(1000);
}

iggymayer, 2023-06-14:
> hello world
#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>

U8G2_LD7032_128X36_F_HW_I2C u8g2(U8G2_R0, 9, 8, 1);

void setup(void)
{
  u8g2.setI2CAddress(0x6E);
  u8g2.begin();
}

void loop(void)
{
  u8g2.clearBuffer();                   // clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);   // choose a suitable font
  u8g2.drawStr(10, 10, "Hello World!"); // write something to the internal memory
  u8g2.sendBuffer();                    // transfer internal memory to the display
  delay(1000);
}

olikraus, 2023-06-14:
I have created a new beta with some updates: 2.34.19
Will this work better?
You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

iggymayer, 2023-06-15:
The version number of the file downloaded by the link is 2.34.20.
The problem remains.

olikraus, 2023-06-15:
Strange. Are you able to check the same with SPI interface instead?

iggymayer, 2023-06-15:
Only use I2C

olikraus, 2023-06-15:
I am not sure how to continue from here. I have exactly implemented the information from your data sheet and the last commit fixed some values which might have been wrong.

iggymayer, 2023-06-15:
Thank you for your help!
Let's close this issue for now, and I'll figure it out when I have time.

olikraus, 2023-06-18:
Sure, no problem. Closing for now.

closed 2023-06-18 by olikraus