olikraus/u8g2 #2136: Adding library to PlatformIo / Arduino project fails to compile (missing SPI.h file)

shan-mcarthur, 2023-03-19:
Using Visual Studio Code, and the PlatformIO extension:  After following the installation instructions of adding the olikraus/U8g2@^2.34.15 lib to the project, then calling "pio run -t clean" then "pio run" to compile the project results in a compilation error: 
.pio/libdeps/heltec_wifi_kit_32_V3/U8g2/src/U8x8lib.cpp:45:10: fatal error: SPI.h: No such file or directory

The project compiled fine before I added the library.  I haven't even added any code to use the library yet - just added the dependent library. This is a Heltec wifi kit 32 v3 board with the SSD1306 OLED display.

My platformio.ini file contains:
[env:heltec_wifi_kit_32_V3]
platform = espressif32
board = heltec_wifi_kit_32_V3
framework = arduino
lib_deps = olikraus/U8g2@^2.34.15

olikraus, 2023-03-19:
Please contact the PlatformIO people. This project only supports ArduinoIDE

shan-mcarthur, 2023-03-27:
I found the solution.  I can either put an include to SPI.H in my main.cpp, or I can add SPI as an additional dependency.  It seems like I only need to do this once to get the compiler to pick up the right Arduino framework dependencies.  I have noticed this with a few different libraries and does not seem to be related to your product.  Closing the issue.

olikraus, 2023-03-27:
Thanks for the feedback