olikraus/u8g2 #2137: getStrWidth behaviour change between V2.28 and 2.34

vdesmedt, 2023-03-19:
This code here below does not produces the same result in V2.28 and V2.34.

`
  mainLcd.begin();
  mainLcd.setFont(MMDigit08);
  Serial.println(mainLcd.getStrWidth("010"));
  Serial.println(mainLcd.getStrWidth("0")+1+mainLcd.getStrWidth("0")+1+mainLcd.getStrWidth("1"));
`

here below is the font but I tried it also with standard fonts : 
`
const uint8_t MMDigit08[1694] U8G2_FONT_SECTION("MMDigit08") = 
  "\0\0\3\2\3\4\4\4\4\5\11\0\376\7\376\0\0\1\233\2\326\6\201\0\5\0D\7\1\11-T"
  "W\266\64\205\3\2\12<\304\216\62H\322\360\20\3\6\13D\306\0\4\6\33D\306\66\5\6+D\306"
  "v\6\5\0D\7\7\5\0D\7\10\5\0D\7\11\5\0D\7\12\5\0D\7\13\5\0D\7\14"
  "\5\0D\7\15\5\0D\7\16\5\0D\7\17\5\0D\7\20\5\0D\7\21\5\0D\7\22\5\0"
  "D\7\23\5\0D\7\24\5\0D\7\25\5\0D\7\26\5\0D\7\27\5\0D\7\30\5\0D\7"
  "\31\5\0D\7\32\5\0D\7\33\5\0D\7\34\5\0D\7\35\5\0D\7\36\5\0D\7\37\5"
  "\0D\7 \5\0D\7!\7\271\304\305\240\4\42\5\0D\7#\5\0D\7$\5\0D\7%\5"
  "\0D\7&\5\0D\7'\5\0D\7(\5\0D\7)\5\0D\7*\5\0D\7+\5\0D"
  "\7,\5\0D\7-\5\0D\7.\6\11DE\0/\5\0D\7\60\14=D\317\222ii\246%"
  "\13\0\61\6)N\207\42\62\13=D\317\32&K\230.\0\63\13\274D\307\230\15a\66$\0\64\11"
  "-LG\246%k\30\65\13=D\317\22\246k\230,\0\66\14=D\317\22\246K\246%\13\0\67\11"
  "\264L\307\230\345@\26\70\14=D\317\222i\311\222i\311\2\71\14=D\317\222i\311\32&\13\0:"
  "\6!LE\24;\6\231\324F\22<\10<D_\324\330\0=\10\35T\307\240\16\2>\10\274DG"
  "\330\324\6\77\12=D\317\222\65\346P\4@\5\0D\7A\13=DW\226\324\206!\263\5B\15="
  "D\307\20U\242d\252D\203\2C\12=D\317\222\211m\311\2D\17=D\307\20U\242$J\242$"
  "\32\24\0E\13=D\307\61\34\222\60\34\4F\12=D\307\61\34\222\260\10G\13=D\317\222\211\245"
  "-Y\0H\12=DGf\33\206\314\26I\10\273D\307\22u\31J\12=D\327\26\266D\221\4K"
  "\15=DG&%%-\211*Y\0L\10=DG\330\343 M\13=DG\246-K\242\331\2N"
  "\14=DG\246MJ\42mZ\0O\12=D\317\222yK\26\0P\14=D\307\220d\332\240\204E"
  "\0Q\12E<\317\222y\251\254\1R\15=D\307\220d\332\240\224*Y\0S\13=D\317\222\251\253"
  "\226,\0T\11=D\307 \205=\1U\11=DG\346[\262\0V\15=DGfKJI\224d"
  "\21\0W\12=DG\346\222(\311-X\13=DG\246%\265JM\13Y\12=DG\246%\265\260"
  "\11Z\12=D\307 f\35\7\1[\7:\304\305\322E\134\5\0D\7]\7:\304\205\322e^\5"
  "\0D\7_\5\0D\7`\5\0D\7a\12-D\317\232\14Z\62\4b\13=DGX\61i\223"
  "\242\0c\12-D\317\222\211Y\262\0d\12=Dg\305\264I\212\22e\12-D\317\222\15C\272\0"
  "f\13=D\227T\311\206(\254\1g\14=\64\317\240i\311\20j\311\2h\12=DGX\61i\266"
  "\0i\10\273DO(\265\14j\13\314\64_\254\325$)Q\0k\13=DGX\223\222\251\222\5l"
  "\10\273D\207\324\313\0m\13-D\207\322\242$J\242\5n\11-DGb\322l\1o\11-D\317"
  "\222\331\222\5p\14=\64Gb\322&E\11C\0q\12=\64\217i\223\24%,r\11-DGb"
  "\22\213\0s\11-D\317\222\36\24\0t\14=DO\230\15QX\212\24\0u\11-DG\346\244("
  "\1v\13-DG\246%\245$\213\0w\12-DGfI\224.\0x\12-DG\226\324*\265\0"
  "y\14=\64GfR\224PK\26\0z\11-D\307\240\265\15\2{\5\0D\7|\5\0D\7}"
  "\5\0D\7~\5\0D\7\177\5\0D\7\200\5\0D\7\201\5\0D\7\202\5\0D\7\203\5\0"
  "D\7\204\5\0D\7\205\5\0D\7\206\5\0D\7\207\5\0D\7\210\5\0D\7\211\5\0D\7"
  "\212\5\0D\7\213\5\0D\7\214\5\0D\7\215\5\0D\7\216\5\0D\7\217\5\0D\7\220\5"
  "\0D\7\221\5\0D\7\222\5\0D\7\223\5\0D\7\224\5\0D\7\225\5\0D\7\226\5\0D"
  "\7\227\5\0D\7\230\5\0D\7\231\5\0D\7\232\5\0D\7\233\5\0D\7\234\5\0D\7\235"
  "\5\0D\7\236\5\0D\7\237\5\0D\7\240\5\0D\7\241\5\0D\7\242\5\0D\7\243\5\0"
  "D\7\244\5\0D\7\245\5\0D\7\246\5\0D\7\247\5\0D\7\250\5\0D\7\251\5\0D\7"
  "\252\5\0D\7\253\5\0D\7\254\5\0D\7\255\5\0D\7\256\5\0D\7\257\5\0D\7\260\10"
  "\33d\306\220\14\1\261\5\0D\7\262\5\0D\7\263\5\0D\7\264\5\0D\7\265\5\0D\7\266"
  "\5\0D\7\267\5\0D\7\270\5\0D\7\271\5\0D\7\272\5\0D\7\273\5\0D\7\274\5\0"
  "D\7\275\5\0D\7\276\5\0D\7\277\5\0D\7\300\5\0D\7\301\5\0D\7\302\5\0D\7"
  "\303\5\0D\7\304\5\0D\7\305\5\0D\7\306\5\0D\7\307\5\0D\7\310\5\0D\7\311\5"
  "\0D\7\312\5\0D\7\313\5\0D\7\314\5\0D\7\315\5\0D\7\316\5\0D\7\317\5\0D"
  "\7\320\5\0D\7\321\5\0D\7\322\5\0D\7\323\5\0D\7\324\5\0D\7\325\5\0D\7\326"
  "\5\0D\7\327\5\0D\7\330\5\0D\7\331\5\0D\7\332\5\0D\7\333\5\0D\7\334\5\0"
  "D\7\335\5\0D\7\336\5\0D\7\337\5\0D\7\340\5\0D\7\341\5\0D\7\342\5\0D\7"
  "\343\5\0D\7\344\5\0D\7\345\5\0D\7\346\5\0D\7\347\5\0D\7\350\5\0D\7\351\5"
  "\0D\7\352\5\0D\7\353\5\0D\7\354\5\0D\7\355\5\0D\7\356\5\0D\7\357\5\0D"
  "\7\360\5\0D\7\361\5\0D\7\362\5\0D\7\363\5\0D\7\364\12\65D\317\222\331\222\212\1\365"
  "\5\0D\7\366\5\0D\7\367\5\0D\7\370\5\0D\7\371\5\0D\7\372\5\0D\7\373\5\0"
  "D\7\374\5\0D\7\375\5\0D\7\376\5\0D\7\377\5\0D\7\0\0\0\4\377\377\0";
`

olikraus, 2023-03-19:
see the change log:
https://github.com/olikraus/u8g2/blob/5a58330f1defb65248673f6c47382c0ac5009fe8/ChangeLog#L260

More details are here:
https://github.com/olikraus/u8g2/issues/1561

vdesmedt, 2023-03-21:
Thank you @olikraus, sorry for not having done my homework before bothering you ;)

closed 2023-03-21 by vdesmedt