olikraus/u8g2 #2141: warning warning: unused variable 'data' [-Wunused-variable] on ESP platform due to uint8_t *data pointer definition triggered in u8x8_byte_arduino_hw_spi()

stefan123t, 2023-03-23:
The following warning is triggered on ESP platform

```code
.pio/libdeps/lolin32_lite/U8g2/src/U8x8lib.cpp: In function 'uint8_t u8x8_byte_arduino_hw_spi(u8x8_t*, uint8_t, uint8_t, void*)':
.pio/libdeps/lolin32_lite/U8g2/src/U8x8lib.cpp:885:12: warning: unused variable 'data' [-Wunused-variable]
   uint8_t *data;
            ^~~~
```

The warning on ESP platform is due to uint8_t *data pointer definition triggered in u8x8_byte_arduino_hw_spi() while the data variable remains unused for ESP_PLATFORM in this method. 

https://github.com/olikraus/u8g2/blob/5a58330f1defb65248673f6c47382c0ac5009fe8/cppsrc/U8x8lib.cpp#L884

I would suggest to fix it:

```cpp
#if !defined(ESP_PLATFORM)
  uint8_t *data;
#endif
```

olikraus, 2023-03-24:
Thanks for providing this issue.
However I think the data definition should be moved more close to the while loop below

https://github.com/olikraus/u8g2/blob/5a58330f1defb65248673f6c47382c0ac5009fe8/cppsrc/U8x8lib.cpp#L883-L909

stefan123t, 2023-03-25:
Hi @olikraus that is a good option. I was not sure whether the data pointer would be required outside of the loop. Thanks for considering the change.

olikraus, 2023-06-18:
This is fixed in "head". I guess i missed to link the commit with this issue.

closed 2023-06-18 by olikraus