olikraus/u8g2 #2144: SSD1309 Problem

ettare, 2023-03-25:
This problem looks similar to issue #1916, but with an i2c interface;  unfortunately, using full frame buffer ‘F’ doesn’t seem to fix the problem.  

This is with ‘East Rising’ software which works ok; 
[image]

This is U8g2 which has broken characters
[image]

U8g2 constructor is:  U8G2_SSD1309_128X64_NONAME0_F_HW_I2C u8g2(U8G2_R0);
Arduino IDE 2.04, Teensy3.2, 2.4” 128x64 oled: 

https://www.aliexpress.com/item/1005005241315177.html?spm=a2g0o.order_detail.order_detail_item.3.184df19cijnRR7

Any thoughts?

(Incidentally, the software constructor:  U8G2_SSD1309_128X64_NONAME0_F_SW_I2C u8g2(U8G2_R0); won’t compile and throws a ‘no matching function’ error)

olikraus, 2023-03-25:
Are you able to provide some example code which reproduces the problem?

ettare, 2023-03-25:
With this code you only see the top half of the words 'Hello World '

`#include <U8g2lib.h>

U8G2_SSD1309_128X64_NONAME0_F_HW_I2C u8g2(U8G2_R0);

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_ncenB08_tr);
  u8g2.drawStr(0, 10,"Hello World!");
  u8g2.sendBuffer();	
  delay(1000);  
}`

olikraus, 2023-03-25:
Can you draw the text multiple times, maybe at 0,25 and 0, 40. How will it look? Picture would be great

ettare, 2023-03-25:
Here it is with the text at: 0,0  0,24  0,40

[image]

olikraus, 2023-03-26:
very strange

Maybe a timing problem. Would U8G2_SSD1309_128X64_NONAME0_F_SW_I2C u8g2(U8G2_R0, ....); work better?

ettare, 2023-03-26:
Thanks for helping out with this problem.  Unfortunately the software constructor throws a compiler error:

Compilation error: no matching function for call to 'U8G2_SSD1309_128X64_NONAME0_F_SW_I2C::U8G2_SSD1309_128X64_NONAME0_F_SW_I2C(const u8g2_cb_t*)'

I agree, it looks like a timing issue so I tried setting the bus clock to 100k (default was 200k), but that made no difference.  I then tried some external 4k7 pullups on SCL and SDA (the schematic showed that it should be fitted with internal pullups), but that stopped the display working.

Weirdly, it half works with this waveform on the SDA line (without external pullups):
[image]

But it doesn’t work with this waveform (with external pullups):
[image]

I’m a bit stumped and not sure what to try next…

olikraus, 2023-03-26:
The software I2C constructor requires the clock and data Arduino pin numbers as additional arguments. Once provided, it should work. Maybe you could test again with the SW I2C constructor.

ettare, 2023-03-27:
Oops, sorry about that; rtfm...   That works, the display shows the text properly with the software constructor.  Is there any disadvantage in using software mode?  Perhaps uses more processing power etc.?  Thanks again for your help.

olikraus, 2023-03-28:
> Is there any disadvantage in using software mode? 

It is much slower...

ettare, 2023-03-29:
Ok, thanks; the refresh rate (about 8fps) in software i2c mode is fine for my application.  I guess we can put it down to the nature of the display and close the issue.

closed 2023-03-29 by ettare