olikraus/u8g2 #2149: stm32f103cbt6 :0.96ssd1306oled,hardware SPI with DMA can't work .build done but don't show anything

AagsAags, 2023-04-01:
[image]
[image]
[image]
[image]
The above is the u8g2 callback function（written by myself） and main function part .
The following is hardware SPI and dma initialization.
[image]
[image]
Download after successful compilation, the u8g2 part（"u8g2Init(&u8g2); " in the main function） does not work.

olikraus, 2023-04-02:
CS signal handling is missing in the byte transfer procedure
``` C++
    case U8X8_MSG_BYTE_START_TRANSFER:
      u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);  
      u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->post_chip_enable_wait_ns, NULL);
      break;
    case U8X8_MSG_BYTE_END_TRANSFER:
      u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->pre_chip_disable_wait_ns, NULL);
      u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
      break;
```

AagsAags, 2023-04-02:
thank you so much ,it work well!!