olikraus/u8g2 #2152: Modify screen width

olikraus, 2023-04-02:
### Discussed in https://github.com/olikraus/u8g2/discussions/2140

<div type='discussions-op-text'>

<sup>Originally posted by **Ultramxn** March 23, 2023</sup>
Hello,
I have a ST7657 12232 LCD cannot display anything with 12232 constructor in u8g2 but display well with OS12864 constructor except skipping six pixels on the left of the screen, that is, missing first character in the test. May I ask can we override screen width of a constructor by passing parameter?

</div>

olikraus, 2023-04-02:
Summary:
Based on the U8G2_ST7567_OS12864_F_4W_SW_SPI there should be a new constructor for a 122x32 display with x offset 10.

@ultramxn is there any link to sales location or a datasheet for the display?

Ultramxn, 2023-04-11:
@olikraus It was bought from ali before covid so I'm sorry I haven't got data sheet and the link's no longer exist.

olikraus, 2023-06-18:
There is already a 122x32 constructor available in u8g2 (issue #1759). Will this work for you? How will the output look like with this existing `U8G2_ST7567_122X32_xxx` constructor? 
You might need to adjust contrast with `u8g2.setContrast`.

Ultramxn, 2023-06-23:
> There is already a 122x32 constructor available in u8g2 (issue #1759). Will this work for you? How will the output look like with this existing `U8G2_ST7567_122X32_xxx` constructor? You might need to adjust contrast with `u8g2.setContrast`.

By changing contrast as you said, I got something displayed but freezed at graphics test example when the same example running well with OS12864  constructor except there were six pixels missing on the left. The following was taken when running graphics test with u8g2 122X32 constructor:
[image]

olikraus, 2023-06-23:
Please execute FlipMode.ino example (See example menu in Arduino IDE or [here](https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/FlipMode/FlipMode.ino)). The flip mode test example will cycle through eight modes. In all modes there must be a solid frame next to the border of the display.

So maybe can you send a picture of that border frame?

Ultramxn, 2023-07-08:
> Please execute FlipMode.ino example (See example menu in Arduino IDE or [here](https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/FlipMode/FlipMode.ino)). The flip mode test [...]

There was no display using 12232 constructor :
[image]
and these are some photos taken when using OS12864 constructor :
[image]
[image]
[image]

olikraus, 2023-07-09:
ok, so you have used the OS12864 constructor for the ST7567 to generate the above pictures with a custom set contrast value. What had been your costom value?

Ultramxn, 2023-07-10:
Yes, I've used OS12864 constructor but its default value to generate the above pictures.

olikraus, 2023-08-14:
Excause, probably I missed the email replies on this topic.
Actually I don't know how to proceed here.  Initially you mentioned, that the `U8G2_ST7567_122X32` constructor work. Later it didn't work any more. 

Is there any datasheet for this display? Where did you buy this display?

closed 2024-08-24 by olikraus