olikraus/u8g2 #2154: Someone port the lcd display 128*64 st7567 i2c

JamToJum, 2023-04-02:
Congratulations. Could you please add support for display with extension 128*64 , type: lcd, connection type: i2c.I am attaching a photo of the display and its characteristics 
[image]
[image]
[image]

olikraus, 2023-04-03:
Can you test one of the existing st7567 constrictor? What will you see , which one works best?

JamToJum, 2023-04-05:
> Чи можете ви перевірити один із існуючих констрикторів st7567? Що ви побачите, який з них найкраще працює?

I've checked all possible options from the library and only one works, but it's designed for a 128*32 display and I'm having trouble scaling to 128*64

JamToJum, 2023-04-05:
https://github.com/mworkfun/ST7567A_128X32DOT_LCD This code works but not full screen.

JamToJum, 2023-04-05:
I uncommented each one in turn and none produced the expected result, only the one I wrote above. found a library from another user, it works correctly, but there are no instructions for how to output data through it. I drop the link to it:https://github.com/luetee/ST7567S_128X64_I2C

olikraus, 2023-04-05:
> only the one I wrote above

Which constructor did work then? I don't find any note from you on any working u8g2 constructor...

JamToJum, 2023-04-07:
> Отже, ви кажете, що жоден із наступних конструкторів нічого не покаже:
> 
> U8G2_ST7567_JLX12864_1_SW_I2C U8G2_ST7567_ENH_DG128064_1_SW_I2C U8G2_ST7567_ENH_DG128064I_1_SW_I2C U8G2_ST7567_OS12864_1_SW_I2C
> 
> Як називається конструктор, який частково працює?

and where are these constructors? Can I file? I downloaded the whole archive and can't find them anywhere...

JamToJum, 2023-04-07:
opened u8g2logo from the folder page_buffer

olikraus, 2023-04-07:
> and where are these constructors?

You can just use them. 

The constructors are all listed here: https://github.com/olikraus/u8g2/blob/master/cppsrc/U8g2lib.h

And also here in the wiki: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp

Maybe just paste the hello world example here.

JamToJum, 2023-04-07:
The constructor works from the display:
U8G2_ST7567_HEM6432_F_HW_I2C
[image]
This is what the test sketch looks like

JamToJum, 2023-04-07:
i'm not too strong in programming but give me a hint in the fileu8g2lib.h there are classes, you need them copy as constructor to try running full screen?

olikraus, 2023-04-07:
ok, and the other constructors?
U8G2_ST7567_JLX12864_F_HW_I2C
U8G2_ST7567_ENH_DG128064_F_HW_I2C
U8G2_ST7567_ENH_DG128064I_F_HW_I2C
U8G2_ST7567_OS12864_F_HW_I2C
Will you see something with them?
You might need to adjust contrast (u8g2.setContrast())

JamToJum, 2023-04-08:
and how to add these constructors? I can't understand what I'm doing wrong... They are not in the file firmware.

olikraus, 2023-04-08:
The constructors are part of the u8g2 library for Arduino IDE

JamToJum, 2023-04-08:
that is, the library that is in the Arduino IDE is not the one that is here on the site?

olikraus, 2023-04-08:
> that is, the library that is in the Arduino IDE is not the one that is here on the site?

It is. If you use Arduino IDE and if you have installed u8g2 via the Arduino library manager, then the constructors should be there. Ensure that you have the latest official u8g2 release. Note that not all constructors are listed in the examples. Check the official list.

If that still doesn't work then post some code here

JamToJum, 2023-04-08:
OK, I'll do it today

JamToJum, 2023-04-13:
Вибачте, але що б я не робив, як би не намагався вивести інформацію на дисплей, мені не получається цього зробити. Моїх знань не достатньо щоб вирішити це питання, тому напевно буду закидати всі способи запустити цей дисплей, і простіше реально купити інший, ніж зламати собі голову об цей дисплей. 

Думаю потрібно написати , що в бібліотеці відсутні потрібні налаштування для роботи ST7567S 128*64 I2C Ver. 2

В прикладах реалізовано дуже багато конструкторів для 1602 та подібних дисплеїв, та ви повинні зрозуміти що на дворі вже 2023 рік, і дуже скоро ці дисплеї стануть просто не потрібними, так як різниця в 50 центів або 1 доллар не велика різниця, щоб створити більш красивий пристрій на кращому дисплеї. Тому думаю варто зосередити увагу також і на тому, чи потрібно зосереджуатися на старих дисплеях.

JamToJum, 2023-04-13:
I'm sorry, but no matter what I do, no matter how I try to get the information on the display, I can't get it to work. My knowledge is not enough to solve this question, so I will probably throw out all the ways to get this display to work, and it is easier to actually buy another one than to break my head over this display.

I think it is necessary to write that the library lacks the necessary settings for ST7567S 128*64 I2C Ver. 2

The examples implement a lot of constructors for 1602 and similar displays, but you have to understand that it's 2023 and very soon these displays will simply be useless, because a difference of 50 cents or 1 dollar is not a big difference to make a nicer device on a better display. Therefore, I think it is worth focusing attention also on whether it is necessary to focus on old displays.

JamToJum, 2023-04-13:
`#include <MUIU8g2.h>
#include <U8g2lib.h>
#include <U8x8lib.h>
#include <Wire.h>

U8G2_ST7567_JLX12864_F_SW_I2C 
u8g2(U8G2_R0, /*clock*/ 5, /*data*/ 4,/*reset*/ U8X8_PIN_NONE);

void setup() {
  u8g2.begin();
}

void loop() {
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_ncenB08_tr); // Вибрати шрифт
  u8g2.drawStr(0, 20, "hello Word"); // Вивести напис
  u8g2.sendBuffer();
  delay(1000);
}`

olikraus, 2023-04-13:
So you say, that
``` C++
#include <U8g2lib.h>
#include <U8x8lib.h>
#include <Wire.h>
U8G2_ST7567_HEM6432_F_HW_I2C u8g2(U8G2_R0, /*clock*/ 5, /*data*/ 4,/*reset*/ U8X8_PIN_NONE);
void setup() {
  u8g2.begin();
}

void loop() {
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_ncenB08_tr); // Вибрати шрифт
  u8g2.drawStr(0, 20, "hello Word"); // Вивести напис
  u8g2.sendBuffer();
  delay(1000);
}
```
works, but
``` C++
#include <U8g2lib.h>
#include <U8x8lib.h>
#include <Wire.h>
U8G2_ST7567_JLX12864_F_SW_I2C 
u8g2(U8G2_R0, /*clock*/ 5, /*data*/ 4,/*reset*/ U8X8_PIN_NONE);
void setup() {
  u8g2.begin();
}

void loop() {
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_ncenB08_tr); // Вибрати шрифт
  u8g2.drawStr(0, 20, "hello Word"); // Вивести напис
  u8g2.sendBuffer();
  delay(1000);
}
```
does not work, correct?

JamToJum, 2023-04-14:
[image]

JamToJum, 2023-04-14:
inscriptions are shown on the display what could be the problem with this? U8G2_ST7567_HEM6432_F_HW_I2C this is the only constructor that outputs at least some information to the display. Can you fix some constructor or do you need to create a new class?

olikraus, 2023-04-14:
It is very unusual, that the other constructors don't work.
What is your microcontroller? Does it have enough RAM?

Please run the ContrastTest Example with the other constructors:
U8G2_ST7567_JLX12864_1_SW_I2C
U8G2_ST7567_ENH_DG128064_1_SW_I2C
U8G2_ST7567_ENH_DG128064I_1_SW_I2C
U8G2_ST7567_OS12864_1_SW_I2C
I am very sure that something should be visible.

JamToJum, 2023-04-15:
I useArduino nano 328P about 35% of RAM remains when loading test sketches.

JamToJum, 2023-04-15:
I checked, no changes, the constructors both did not work and do not work. With the constructor that worked, I got the following result:

https://user-images.githubusercontent.com/129545653/232238286-2419596a-5a14-4168-9d2b-b0d5ff440ad8.mp4

JamToJum, 2023-04-15:
I think the sequence of turning on the display is not properly organized in these constructors

JamToJum, 2023-04-15:
After I loaded the working builder, I tried to load the others but the display stuck and wouldn't change...

olikraus, 2023-04-15:
ok, let me look into this more in detail.

Here is the code for the HEM6432 display:
https://github.com/olikraus/u8g2/blob/fbeba5d1c384872caef828f762af51a28fbfc6f8/csrc/u8x8_d_st7567.c#L906-L937

This is the init code for the U8G2_ST7567_OS12864_1_SW_I2C constructor
https://github.com/olikraus/u8g2/blob/fbeba5d1c384872caef828f762af51a28fbfc6f8/csrc/u8x8_d_st7567.c#L1050-L1082

You can see that it is almost the same. 
The differences are:
 1. the LCD bias 0xa2 vs 0xa3
 2. the v0 voltage 0x24 vs 0x26
 3. the contrast 225/4 vs 50/4

We can use the same init sequence with the U8G2_ST7567_OS12864_1_SW_I2C constructor:
By using the OS12864 constructor and after u8g2.begin() sent the following  command:

``` C++
u8g2.sendF("ccca", 0xa2, 0x24, 0x81, 225/4);
```
This will configure the display in the same way like the HEM6432 display.

> I useArduino nano 328P about 35% of RAM remains when loading test sketches.

I hope you have used the picture loop examples (constructor with `_1_` ). Due to the limited RAM the full buffer mode will probably not work (constructor with `_F_`). This means, you can not use the u8g2.sendBuffer() command, instead you MUST use the picture loop:

https://github.com/olikraus/u8g2/blob/fbeba5d1c384872caef828f762af51a28fbfc6f8/sys/arduino/u8g2_page_buffer/HelloWorld/HelloWorld.ino#L407-L414

JamToJum, 2023-04-17:
Well, I tried the code you suggested, pasted it in and downloaded it to the Arduino, but it didn't make any sense

JamToJum, 2023-04-17:
Sketch uses 8,862 bytes (28%) of application storage. The limit is 30720 bytes.
Global variables use 568 bytes (27%) of dynamic memory, leaving 1480 bytes for local variables. The limit is 2048 bytes.

olikraus, 2023-04-17:
As I said before: Do NOT use .sendBuffer(), instead use
``` C++
   u8g2.firstPage(); 
   do { 
     u8g2.setFont(u8g2_font_ncenB10_tr); 
     u8g2.drawStr(0,24,"Hello World!"); 
   } while ( u8g2.nextPage() ); 
```

You can also just use an "page buffer"  example. 

`u8g2.sendF("ccca", 0xa2, 0x24, 0x81, 225/4);` should be correct.

JamToJum, 2023-04-25:
I finally found time for the project

unfortunately, this code also does not give any results

olikraus, 2023-04-25:
> unfortunately, this code also does not give any results

Can you sent your complete code?

JamToJum, 2023-05-16:
No longer relevant.Sorry for not answering for a long time. I left the project for another time

closed 2023-05-16 by JamToJum

FrankHovis, 2023-10-13:
Has anyone done anymore with this? I'm trying one of these displays on a ESP32-C3 and using the constructor
U8G2_ST7567_ENH_DG128064I_F_SW_I2C u8g2(U8G2_R0, 4, 5, U8X8_PIN_NONE);
seems to work but it slow. Using the constructor of
U8G2_ST7567_ENH_DG128064I_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE, 4, 5); 
just locks the CPU up.

Any suggestions?

olikraus, 2023-10-13:
Maybe a watchdog issue. This was discussed more often here. Also look here: https://stackoverflow.com/questions/51750377/how-to-disable-interrupt-watchdog-in-esp32-or-increase-isr-time-limit