olikraus/u8g2 #2158: SH1107 i2c display low frame rate?

zalexzperez, 2023-04-06:
Hi,

I'm using the u8g2 library on an i2c 1.29" SH1107 display with 64x128 resolution. This is the constructor I chose: 

`U8G2_SH1107_64X128_F_HW_I2C u8g2(U8G2_R1, U8X8_PIN_NONE);`

I'm simply displaying the word "Hello!" on the void loop :
```
void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0,10,"Hello!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(1000);  
}
```

If the screen is fixed in place, the text is readable. But if you start moving it up and down, the text becomes hard to read because the screen is not updating fast enough. As a matter of fact, if you try to record a video of it with a phone, you can see black diagonal stripes across the screen.
Also, the screen brightness is quite low by default. 

This does not occur with the Adafruit SH110x library. It's readable even when moving the display around and the screen seems to be at full brightness, while being quieter. With the u8g2 library, the display makes a louder buzzing sound. 

Any ideas?
Thanks in advance.

zalexzperez, 2023-04-06:
Hi, @olikraus . Thanks for your quick reply.

Adding the .sendF instructions only increased the brightness. But the display is still showing a low frame rate.
Here's a very short video comparing the two libraries:
https://youtu.be/dyXKyjGkMrQ 

About the adafruit init sequence, I don't know for sure. All I can tell you is that I'm using the latest library version and all I have in my sketch is: 

```
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SH110X.h>

Adafruit_SH1107 display = Adafruit_SH1107(64, 128, &Wire);

void setup() {
  Serial.begin(115200);

  display.begin(0x3C, true); // Address 0x3C default
  display.setRotation(1);

  display.clearDisplay();
  display.display();
  display.setCursor(0,0);
  display.setTextSize(2);
  display.setTextColor(SH110X_WHITE);
  display.print("Adafruit");
  display.display();

}
void loop() {
}
```

olikraus, 2023-04-06:
You could try the `u8g2.setBusClock(400000);` or even `u8g2.setBusClock(800000);` to improve the refresh speed.

And of course you need to remove the delay command in u8g2:
``` C++
void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0,10,"Hello!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  // delay(1000);    // REMOVE THIS COMMAND
}
```

olikraus, 2023-04-06:
Moreover, in your adafruit example the code is put into setup() procedure. For comparison you should do the same with the u8g2 code, otherwise the comparison is not fair.

``` C++
void setup(void) {
  u8g2.begin();
  u8g2.sendF("ca", 0x81, 0x4f);  
  u8g2.sendF("ca", 0xa8, 0x7f);
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0,10,"Hello!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
}
void loop(void) {
}
```
If you otherwise would put the adafruit code into loop() then you would also see similar flicker. I mean your title and inital post is not fair at all, because the adafruit example does not do any refresh due to the empty loop() and the low frame rate was probably caused by the `delay()`.

zalexzperez, 2023-04-06:
Here's a longer video where I demonstrate the used code (running from the setup, without delays) and the flashing process with both libraries.
Setting the bus clock to those speeds didn't make a difference.

https://youtu.be/1MWUT57q9Kk

olikraus, 2023-04-06:
I think the flickering now is caused by the display and your video cam.

zalexzperez, 2023-04-06:
> I think the flickering now is caused by the display and your video cam.

The flickering you see in the video is not visible to the human eye,  it's only visible when you use a camera to record it, as I stated in my first post. The video just proves there's a difference in the way the display is operating depending on the used library. Not only that, but also the sound coming from the display is different, but that's impossible for me to capture, as you have to put your ear very close to it to hear it.

Anyway, it's something you can tell VERY clearly with your own eyes when either moving the display or yourself. It's a similar feeling to like going from a 60Hz display to a 165Hz one while trying to read a scrolling text.

olikraus, 2023-04-07:
Sure, but it is the display  itself which operates at a certain speed.

zalexzperez, 2023-04-07:
Can we keep trying to replicate what the Adafruit implementation has until we see the same behavior? Or is it too complicated?
Thanks

olikraus, 2023-04-07:
In my first reply I have mentioned the adafruit setup (provided that I captured the right init code). You can use this init code and send this to your display with sendF command after u8g2.begin(). The first number is the command (like 0x81), the second number is the argument for the command (0x4f), so you usually need to use "ca" for command+argumnet in the sendF command.
Once all commands are identical to the adafruit init sequence, then it will behave in the same way.

Note, however, that the correnct init sequence is mentioned in the datasheet of the display vendor.

zalexzperez, 2023-04-07:
@olikraus , I found the culprit! Here's a video showing the result:
https://youtube.com/shorts/wTPfWhRh4YE

You gave me the right answer at your first reply. You pointed out the difference in U8X8_CA(0x0a8, 0x7f), 3f 
However, there was a little mistake in your answer: your suggested code included the old 0x7f value instead of 0x3f.

u8g2.sendF("ca", 0xa8, 0x7f);

It was my fault for simply reusing your code without paying full attention to what your words meant. 
Thanks to your last reply I was able to send the commands, one by one, to have Adafruit's init sequence,  until the low refresh rate was gone. 

Thanks Oliver!
Does this mean all SH1107 displays need this init sequence change, or is it just mine/some?

TL;DR For anyone reading this topic, including this line fixed the low-frame rate problem in my SH1107 i2c display:

`u8g2.sendF("ca", 0xa8, 0x3f); `

olikraus, 2023-04-07:
Ah, yes, my mistake, in fact I wanted to write "u8g2.sendF("ca", 0xa8, 0x3f); " :-(

> Does this mean all SH1107 displays need this init sequence change, or is it just mine/some?

The SH1107 is just the display controller. Each display however differs and requires different setup.
A 128x64 green OLED might require different setup compared to a 128x64 yellow OLED. And also if there is a progress in the technology of the OLEDs itself, the controller might stay, but the power consumption might change, again requiring a different init sequence. Thats why I said: Only the display vender knows the correct sequence.

closed 2023-05-21 by zalexzperez