olikraus/u8g2 #2159: Using u8g2 C library with Pico-SDK

nordseele, 2023-04-10:
Hi, 
I'm trying to use the u8g2 library in a RPI pico W project for driving a SSD1305 display via SPI. I'm not using the Arduino core but the Pico-SDK and I have a VSCODE building environment configured, etc.

I've added two functions `u8x8_byte_pico_hw_spi` and `u8x8_gpio_and_delay_template` but I'm having troubles compiling the u8g2 library. The compiler throws a lot of 'undefined references' errors: ```
[build] /usr/local/Cellar/arm-none-eabi-gcc/10.3-2021.07/gcc/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: u8g2/libu8g2.a(u8g2_d_setup.c.obj): in function `u8g2_Setup_gp1287ai_256x50_2':
[build] u8g2_d_setup.c:(.text+0xcdbc): undefined reference to `u8x8_d_gp1287ai_256x50'
```

I'll complete this post with the full details when I'm back at my computer but before that: has anybody successfully used u8g2 with a Raspberry Pi Pico ? If so, do you have a copy of your cmakelist or something please ? Do I need to compile the u8g2 library separately ?
Thanks!

nordseele, 2023-04-12:
Thanks Olikraus, 
I'm still investigating this issue. I managed to compile once or twice after removing references to other types of displays in `u8g2_d_setup.c` but obviously, this is not the way to go, the linker is probably faulty in my case, and after all the display is still not working.

I'm gonna try to compile the library separately and include it as a static library in my project instead of including the "cmakelist". I'll report here.

nordseele, 2023-04-14:
I can now compile u8g2 along with my Rpi pico project, without problem. 

I edited my Pico project `cmakelists.txt` this way:

``
target_include_directories(${target_proj} PRIVATE u8g2/csrc)
file(GLOB U8G2_SRC u8g2/csrc/*.c)
add_library(u8g2 ${U8G2_SRC})
target_link_libraries(${target_proj} u8g2)
``

The only issue now is displaying something on the OLED [NHD-2.23-12832UCB3](https://fr.newhavendisplay.com/content/specs/NHD-2.23-12832UCB3.pdf) I think I'm getting close to the goal, but right now it only displays two or three dots on the right of the display.

I've checked the voltage, the wiring and the GPIOS and even tested with the Arduino IDE (I used a 1305 Adafruit software SPI constructor with success), but with the Pico-SDK, no luck so far; I tried both noname and adafruit 1 and F. _4 wire software SPI_ doesn't work for me, HW neither but at least it shows two dots on the screen :D, that's a start.

I think my `u8x8_gpio_and_delay...` and `u8x8_byte...` functions are correct, I tried to set CS to 0 and to 1, perhaps I'm missing something else. Do you know anything  else I could check please ? Thanks ! 

```
#include <stdio.h>
#include "pico/stdlib.h"
#include "hardware/gpio.h"
#include "hardware/spi.h"
#include <string.h>
#include <u8g2.h>

#define UART_ID uart1
#define BAUD_RATE 9600
#define UART_TX_PIN 8
#define UART_RX_PIN 9

u8g2_t u8g2; 

// ADD THE INIT DISPLAY 
#define SPI_PORT spi0
#define PIN_CS 5
#define PIN_SCK 2
#define PIN_MOSI 3
#define SPI_SPEED 4000 * 1000
#define PIN_DC 4
#define PIN_RST 11

uint8_t u8x8_byte_pico_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) {
  uint8_t *data;
  switch (msg) {
  case U8X8_MSG_BYTE_SEND:
    data = (uint8_t *)arg_ptr;
    spi_write_blocking(SPI_PORT, data, arg_int);
    break;
  case U8X8_MSG_BYTE_INIT:
    u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
    break;
  case U8X8_MSG_BYTE_SET_DC:
    u8x8_gpio_SetDC(u8x8, arg_int);
    break;
  case U8X8_MSG_BYTE_START_TRANSFER:
    u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);
    u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO,u8x8->display_info->post_chip_enable_wait_ns, NULL);
    break;
  case U8X8_MSG_BYTE_END_TRANSFER:
    u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO,
                            u8x8->display_info->pre_chip_disable_wait_ns, NULL);
    u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
    break;
  default:
    return 0;
  }
  return 1;
}

uint8_t u8x8_gpio_and_delay_pico(u8x8_t *u8x8, uint8_t msg,uint8_t arg_int, void *arg_ptr) {
  switch (msg) {
  case U8X8_MSG_GPIO_AND_DELAY_INIT: 
    spi_init(SPI_PORT, SPI_SPEED);
    gpio_set_function(PIN_CS, GPIO_FUNC_SIO);
    gpio_set_function(PIN_SCK, GPIO_FUNC_SPI);
    gpio_set_function(PIN_MOSI, GPIO_FUNC_SPI);
    gpio_init(PIN_RST);
    gpio_init(PIN_DC);
    gpio_init(PIN_CS);
    gpio_set_dir(PIN_RST, GPIO_OUT);
    gpio_set_dir(PIN_DC, GPIO_OUT);
    gpio_set_dir(PIN_CS, GPIO_OUT);
    gpio_put(PIN_RST, 1);
    gpio_put(PIN_CS, 1);
    gpio_put(PIN_DC, 0);
    break;                  
  case U8X8_MSG_DELAY_NANO: // delay arg_int * 1 nano second
    sleep_us(1000 * arg_int);
    break;
  case U8X8_MSG_DELAY_100NANO: // delay arg_int * 100 nano seconds
    sleep_us(1000 * 100 * arg_int);
    break;
  case U8X8_MSG_DELAY_10MICRO: // delay arg_int * 10 micro seconds
    sleep_us(arg_int * 10);
    break;
  case U8X8_MSG_DELAY_MILLI: // delay arg_int * 1 milli second
    sleep_ms(arg_int);
    break;
  case U8X8_MSG_GPIO_CS: // CS (chip select) pin: Output level in arg_int
    gpio_put(PIN_CS, arg_int);
    break;
  case U8X8_MSG_GPIO_DC: // DC (data/cmd, A0, register select) pin: Output level
    gpio_put(PIN_DC, arg_int);
    break;
  case U8X8_MSG_GPIO_RESET: // Reset pin: Output level in arg_int
    gpio_put(PIN_RST, arg_int);  // printf("U8X8_MSG_GPIO_RESET %d\n", arg_int);
    break;
  default:
    u8x8_SetGPIOResult(u8x8, 1); // default return value
    break;
  }
  return 1;
}

void draw_display() {
	char hey[10] = "Hello";
	u8g2_ClearBuffer(&u8g2);
  u8g2_ClearDisplay(&u8g2);
	u8g2_DrawStr(&u8g2, 10, 10, hey);
	u8g2_SetDrawColor(&u8g2, 1);
  u8g2_UpdateDisplay(&u8g2);
}

void display_sequence(){
	u8g2_Setup_ssd1305_128x32_noname_f(&u8g2, U8G2_R0, u8x8_byte_pico_hw_spi, u8x8_gpio_and_delay_pico);
  u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,
  u8g2_SetPowerSave(&u8g2, 0); // wake up display
	draw_display();
}

int main()
{	
    uart_init(UART_ID, BAUD_RATE);
    gpio_set_function(UART_TX_PIN, GPIO_FUNC_UART);
    gpio_set_function(UART_RX_PIN, GPIO_FUNC_UART);

    // U8G2
    display_sequence();
    
    stdio_init_all();

    while (true) {
        bool b = spi_is_busy(SPI_PORT);
        const char* s = b ? "true" : "false";
        uart_puts(UART_ID, s);
        sleep_ms(250);
    }

    return 0;
}
```

[image]

olikraus, 2023-04-14:
You have missed to assign a suitable font: https://github.com/olikraus/u8g2/wiki/u8g2reference#setfont
You could use a standard font like helvB14: https://github.com/olikraus/u8g2/wiki/fntgrpadobex11#helvb14

Then the command my look like this:
``` C++
        u8g2_SetFont(&u8g2, u8g2_font_helvB14_tr);
	u8g2_SetDrawColor(&u8g2, 1);  // use this before call to DrawStr
	u8g2_DrawStr(&u8g2, 10, 10, hey);
```

nordseele, 2023-04-15:
That's right, I completely forgot to set the font!
I also edited my Pico `cmakelists.txt` to prevent a _RAM overflow_ (as found in another issue in this repo) with `set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -ffunction-sections -fdata-sections -Wl,--gc-sections")`

Thank you very much for your help and for the u8g2 library @olikraus ! :-) 

[image]

Here's the edited **cmakelists.txt** if anyone's looking for that (and the **main.c** is below that:

```
cmake_minimum_required(VERSION 3.13)

set(PICO_BOARD pico_w)
include(pico_sdk_import.cmake)

set(target_proj demo_oled_spi_pico)
project(${target_proj} C CXX ASM)
set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -ffunction-sections -fdata-sections -Wl,--gc-sections")
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

pico_sdk_init()

add_executable(${target_proj}
    main.c
)

target_include_directories(${target_proj} PRIVATE u8g2/csrc)
file(GLOB U8G2_SRC u8g2/csrc/*.c)
add_library(u8g2 ${U8G2_SRC})
target_link_libraries(${target_proj} u8g2)

target_include_directories(${target_proj} PRIVATE ${CMAKE_CURRENT_LIST_DIR})
target_link_options(${target_proj} PRIVATE -Xlinker --print-memory-usage)
target_compile_options(${target_proj} PRIVATE -Wall -Wextra -DCFG_TUSB_DEBUG=1)
target_link_libraries(${target_proj}  pico_stdlib hardware_spi)
pico_add_extra_outputs(${target_proj})

pico_enable_stdio_usb(${PROJECT_NAME} 0)
pico_enable_stdio_uart(${PROJECT_NAME} 1)

```

And the `main.c` example code, I switched the constructor to "Adafruit" and the random dots are now gone from the display: 

```
#include <stdio.h>
#include "pico/stdlib.h"
#include "hardware/gpio.h"
#include "hardware/spi.h"
#include <string.h>
#include <u8g2.h>

#define UART_ID uart1
#define BAUD_RATE 9600
#define UART_TX_PIN 8
#define UART_RX_PIN 9

u8g2_t u8g2; 

// ADD THE INIT DISPLAY 
#define SPI_PORT spi0
#define PIN_CS 5
#define PIN_SCK 2
#define PIN_MOSI 3
#define SPI_SPEED 4000 * 1000
#define PIN_DC 4
#define PIN_RST 11

uint8_t u8x8_byte_pico_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) {
  uint8_t *data;
  switch (msg) {
  case U8X8_MSG_BYTE_SEND:
    data = (uint8_t *)arg_ptr;
    spi_write_blocking(SPI_PORT, data, arg_int);
    break;
  case U8X8_MSG_BYTE_INIT:
    u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
    break;
  case U8X8_MSG_BYTE_SET_DC:
    u8x8_gpio_SetDC(u8x8, arg_int);
    break;
  case U8X8_MSG_BYTE_START_TRANSFER:
    u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);
    u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO,u8x8->display_info->post_chip_enable_wait_ns, NULL);
    break;
  case U8X8_MSG_BYTE_END_TRANSFER:
    u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO,
                            u8x8->display_info->pre_chip_disable_wait_ns, NULL);
    u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
    break;
  default:
    return 0;
  }
  return 1;
}

uint8_t u8x8_gpio_and_delay_pico(u8x8_t *u8x8, uint8_t msg,uint8_t arg_int, void *arg_ptr) {
  switch (msg) {
  case U8X8_MSG_GPIO_AND_DELAY_INIT: 
    spi_init(SPI_PORT, SPI_SPEED);
    gpio_set_function(PIN_CS, GPIO_FUNC_SIO);
    gpio_set_function(PIN_SCK, GPIO_FUNC_SPI);
    gpio_set_function(PIN_MOSI, GPIO_FUNC_SPI);
    gpio_init(PIN_RST);
    gpio_init(PIN_DC);
    gpio_init(PIN_CS);
    gpio_set_dir(PIN_RST, GPIO_OUT);
    gpio_set_dir(PIN_DC, GPIO_OUT);
    gpio_set_dir(PIN_CS, GPIO_OUT);
    gpio_put(PIN_RST, 1);
    gpio_put(PIN_CS, 1);
    gpio_put(PIN_DC, 0);
    break;                  
  case U8X8_MSG_DELAY_NANO: // delay arg_int * 1 nano second
    sleep_us(1000 * arg_int);
    break;
  case U8X8_MSG_DELAY_100NANO: // delay arg_int * 100 nano seconds
    sleep_us(1000 * 100 * arg_int);
    break;
  case U8X8_MSG_DELAY_10MICRO: // delay arg_int * 10 micro seconds
    sleep_us(arg_int * 10);
    break;
  case U8X8_MSG_DELAY_MILLI: // delay arg_int * 1 milli second
    sleep_ms(arg_int);
    break;
  case U8X8_MSG_GPIO_CS: // CS (chip select) pin: Output level in arg_int
    gpio_put(PIN_CS, arg_int);
    break;
  case U8X8_MSG_GPIO_DC: // DC (data/cmd, A0, register select) pin: Output level
    gpio_put(PIN_DC, arg_int);
    break;
  case U8X8_MSG_GPIO_RESET: // Reset pin: Output level in arg_int
    gpio_put(PIN_RST, arg_int);  // printf("U8X8_MSG_GPIO_RESET %d\n", arg_int);
    break;
  default:
    u8x8_SetGPIOResult(u8x8, 1); // default return value
    break;
  }
  return 1;
}

void draw_display() {
	char hey[10] = "Hello world";
	u8g2_ClearBuffer(&u8g2);
        u8g2_ClearDisplay(&u8g2);
	u8g2_SetDrawColor(&u8g2, 1);
        u8g2_SetFont(&u8g2, u8g2_font_t0_11_te);
        u8g2_DrawStr(&u8g2, 10, 10, hey);
        u8g2_UpdateDisplay(&u8g2);
}

void display_sequence(){
  u8g2_Setup_ssd1305_128x32_adafruit_f(&u8g2, U8G2_R0, u8x8_byte_pico_hw_spi, u8x8_gpio_and_delay_pico);
  u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,
  u8g2_SetPowerSave(&u8g2, 0); 
  draw_display();
}

int main()
{	
    uart_init(UART_ID, BAUD_RATE);
    gpio_set_function(UART_TX_PIN, GPIO_FUNC_UART);
    gpio_set_function(UART_RX_PIN, GPIO_FUNC_UART);

    // U8G2
    display_sequence();
    
    stdio_init_all();

    return 0;
}
```

Szpila123, 2023-04-17:
Will this end up as another hal added to ./sys?

ChrisChrisLoLo, 2023-04-24:
This is great! Thanks for showing the source, got the rp2040 + u8g2 to work for me!

One small thing I noticed was `U8X8_MSG_DELAY_NANO` and `U8X8_MSG_DELAY_100NANO`. There's 1000 nanos in a micro (rather than 1000 micros in a nano), so changing this
```
  case U8X8_MSG_DELAY_NANO: // delay arg_int * 1 nano second
    sleep_us(1000 * arg_int);
    break;
  case U8X8_MSG_DELAY_100NANO: // delay arg_int * 100 nano seconds
    sleep_us(1000 * 100 * arg_int);
    break;
```
to
```
  case U8X8_MSG_DELAY_NANO: // delay arg_int * 1 nano second
    sleep_us(arg_int); // 1000 times slower, though generally fine in practice given rp2040 has no `sleep_ns()`
    break;
  case U8X8_MSG_DELAY_100NANO: // delay arg_int * 100 nano seconds
    sleep_us(arg_int);
    break;
```
_greatly_ improved performance for me, alteast when using a Sharp memory LCD. There might be hacks out there to get the sleep to be even smaller (e.g. maybe running assembly `nops`), but I'm generally happy with the current performance.

nordseele, 2023-04-25:
Thanks a lot for the suggestion @ChrisChrisLoLo ! 
I've modified the example code in consequence after running some tests on my side with the Newhaven display (this explains my late reply). I didn't do "stress tests" before but you're absolutely right, the performance is MUCH BETTER with this modification :-)

olikraus, 2024-04-14:
Example projects are always welcome under https://github.com/olikraus/u8g2/tree/master/sys
And yes, I would PR this...

infovore, 2025-01-27:
Just to say - I've got sample code for running the C library on Pico SDK with the I2C 1306 OLED display (as opposed to SPI above), will try to share it somewhere this week. This thread was very useful for wrapping my head around a few things, so thanks for your CMakeList files, @nordseele.

lifeisafractal, 2025-09-07:
I also ran into running out of flash on an RPi Pico board. You can use the following in your `CMakeLists.txt` to avoid needing to modify u8g2 directly:

```cmake
add_subdirectory(u8g2)
# Add function sections and garbage collection so unused fonts get trimmed to fit in flash
set_property(TARGET u8g2 PROPERTY COMPILE_OPTIONS -ffunction-sections -fdata-sections -Wl,--gc-sections)
```