olikraus/u8g2 #2161: Parallel usage of C and C++ in same projects

tusker-tools, 2023-04-20:
Hi,
I'm working on a project which is basically written in C, but it includes also the C++ lib [LCDMenuLib2](https://github.com/Jomelo/LCDMenuLib2), which uses the C++ API of u8g2.

Now I would like the C++ instance of U8G2 to use the same u8g2_t variable as the C parts of my software.

To make it more clear:
Somewhere in the C part, I have the following define and initialization: 
```
u8g2_t u8g2;
u8g2_Setup_st7920_s_128x64_f(&u8g2, U8G2_R0, u8x8_byte_stm32_hw_spi, u8x8_stm32_gpio_and_delay);
```

But in the C++ instance, it is redefined as protected variable:

```
class U8G2
#ifdef ARDUINO
: public Print
#endif
{
  protected:
    u8g2_t u8g2;
    u8x8_char_cb cpp_next_cb; /*  the cpp interface has its own decoding function for the Arduino print command */
  public:
```

Thes would lead to memory beeing allocated for the u8g2_t variable twice.

Is there a possibility to make the C++ API use an already existing instance like in my case?

Personally I was already able to solve that problem by removing the define in U8g2lib.h and adding it as extern like this:
```
extern u8g2_t u8g2;
class U8G2
#ifdef ARDUINO
: public Print
#endif
{
  protected:
   //u8g2_t u8g2;
    u8x8_char_cb cpp_next_cb; /*  the cpp interface has its own decoding function for the Arduino print command */
  public:
```
Do you see a clean way to implement that in the official project? My goal is to have a 100% reuse of the u8g2 lib code, without any manual modifications.

Thanks in advance for your Ideas!

MrTC1277, 2023-05-08:
u8g2 have C API,why not use it?

olikraus, 2023-05-09:
@tusker-tools 
If LCDMenu calls the u8g2 constructor, then you could get the C u8g2_t from the u8g2 object by using the C++ member function u8g2.GetU8g2(). It will return a pointer to u8g2_t. In your C function you could then always use that pointer to u8g2_t:

```
u8g2_t *u8g2_ptr;
...