olikraus/u8g2 #2164: Is it an UC1698 or other controller IC?

eggfly, 2023-04-24:
Hi~ U8g2 is a very nice project. I use it in many if my DIY project.

Recently I bought an Electronic Dictionary and I want to drive it using my esp32-s3 board. 
It's screen appears like these pictures:

https://edupoint.n.my/index.php?ws=showproducts&products_id=3585861ELECTRONIC-DICTIONARY

[image]

[image]

It has a 4.3 inch, 320 x 160 pixel display.

Fortunately, there are all test solder joints on his screen driver board and motherboard, and all symbols have been marked with silk screen printing, including WR0, WR1, CD, D0-D7, RST, CS

[image]

[image]

So it force BM set to 8-bits 8080 parallel mode, and it seems like the WR0, WR1, CD pins only appears in UC16XX series datasheet?
And maybe it uses UC1698 controller to use mono mode (instead of RGB mode) to drive this large resolution screen (320 x 160).

I try to use this https://github.com/Sinitax/uc1698u-8080 to drive it. But finally is only shows a line: 

[image]

I have two questions to ask for help:
1. How to determine which controller that screen uses ?
2. If it's controller is UC1698, can U8g2 lib support this controller?

Thank you very much!

olikraus, 2023-04-25:
> How to determine which controller that screen uses ?

Difficult. Some guess can be made according to the pins...

The UC1698 is not supported :-(

closed 2023-04-26 by eggfly

Jack-by, 2026-01-09:
Hi Mr olikraus.
I often use the u8g2 library. It's a great library. I ported it to the STM32. I'm using it with a 128x64 display on an ST7565P. I'm really looking forward to using a 160x160 display on the UC1698U. 
Is there any experience connecting the UC1698U to the u8g2 library? There's a lot of information about this display online. It's also readily available in many stores.

olikraus, 2026-01-15:
> This UC1698U is a tough nut to crack. 

yes :-D

What do you mean by "all-line"?

There are two native architectures supported by u8g2, One is the SSD1306 "page-like" architecture:
https://github.com/olikraus/u8g2/blob/dbd338af5d57b219e48ea1ffdb97c4153668676c/tools/codebuild/codebuild.c#L131
This is enabled by "u8g2_ll_hvline_vertical_top_lsb" and also described here: https://github.com/olikraus/u8g2/wiki/u8g2reference#memory-structure-for-controller-with-u8x8-support

Then there is the ST7920 architecture which indeed line based:
https://github.com/olikraus/u8g2/blob/dbd338af5d57b219e48ea1ffdb97c4153668676c/tools/codebuild/codebuild.c#L1015
This is enabled by "u8g2_ll_hvline_horizontal_right_lsb". Memory architecture should be close to the above pixel line.

These two architectures are the base architectures of u8g2. If the architecture deviates from this, then the data bits are indeed converted on the fly.

An example for the on-the-fly conversion is the SSD1325: https://github.com/olikraus/u8g2/blob/dbd338af5d57b219e48ea1ffdb97c4153668676c/csrc/u8x8_d_ssd1325.c#L195

olikraus, 2026-01-16:
nice

olikraus, 2026-01-16:
I have created an official Arduino driver for this display in beta 2.36.17:  U8G2_UC1698_160X160_1_4W_HW_SPI

Are you able to test the Arduino Version of U8g2 within Arduino Framework?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

Jack-by, 2026-01-17:
u8x8_byte_hw_fsmc
 is my function. It writes a byte of information directly to the parallel port. FSMC peripheral unit. It runs on an STM32 microcontroller.
### U8G2_UC1698_160X160_1_4W_HW_SPI
It's not suitable for this display. It only has a parallel 8080 data port. It should be called something like this.
### U8G2_UC1698_128X64_F_8080 u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7,  /*enable=*/ 18, /*cs=*/ 3, /*dc=*/ 16, /*reset=*/ 17); // UC1628 pin names: enable: /WR, dc: CD

Jack-by, 2026-01-17:
I can test how the display works on an Arduino. I have a board. I just uploaded a sketch to make the LED blink. 
[image]
Everything works. Unfortunately, I haven't worked with Arduino much and it's difficult for me to program it. If you send me the sketch, I'll compile and upload it. I'll connect the display to the board. No problem there.

olikraus, 2026-01-17:
All the constructors are included. In your case 
```
U8G2_UC1698_160X160_1_8080(const u8g2_cb_t *rotation, uint8_t d0, uint8_t d1, uint8_t d2, uint8_t d3, uint8_t d4, uint8_t d5, uint8_t d6, uint8_t d7, uint8_t enable, uint8_t cs, uint8_t dc, uint8_t reset = U8X8_PIN_NONE)
```
should be available already (however speed will be poor compared to your implementation i assume).

In the above examples you can go to the page_buffer examples, select one of the sketches (which you did already) but you need to uncomment your constructor.
If have added the SPI constructor in the examples, which then has to be replaced by the 8080 constructor.

I see, thats quite of an effort, because you need to reconnect your display to the Arduino board.

Please don't spent too much effort here.

Jack-by, 2026-01-18:
U8G2_UC1698_160X160_F_8080 
It worked. The sketch compiled and loaded. But the test image looks like this. I realized the display was only rendering the top strip of pixels. Nothing was rendering below that.

[image]

#include <Arduino.h>
#include <U8g2lib.h>
#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_UC1698_160X160_1_8080 u8g2(U8G2_R0, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11);

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(100);  
}
This is a sketch for outputting an image.

Jack-by, 2026-01-18:
U8G2_UC1698_160X160_F_8080 u8g2(U8G2_R0, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11);

void setup(void) {
  u8g2.begin();
}

void loop(void) {
 u8g2.clearBuffer();          // clear the internal memory
    u8g2.setFont(u8g2_font_logisoso20_tf);
    u8g2.drawStr(0,25,"ABCDEFGHIJKL");
    u8g2.drawStr(0,50,"MNOPQRSTUVWX");
    u8g2.drawStr(0,75,"YZ012345678");
    u8g2.drawStr(0,100,"~!@#$%^&*()");
    u8g2.drawStr(0,125,"abcdefghijkl");
    u8g2.drawStr(0,150,"mnopqrstuvwx");
  u8g2.sendBuffer();
}
This sketch doesn't draw anything. The display is blank.

olikraus, 2026-05-17:
There is only this 160x160 constructor available (see above in this thread). Moreover, i don't know the state of this constructor. The timing issue still might be included.
We could start to work on this, but probably we need to 
 * Update the dimensions and related configuration in the UC1698
 * Fix the timing issues