olikraus/u8g2 #2168: Requesting support for the SSD1680Z driver chip. There are some great e-ink screens that are using it.

piogrammer, 2023-05-03:
Thank you so much for sharing this project, my friend. I can't imagine how much time and effort you have put into it. This project has brought me and my children a lot of joy, and they can easily operate the computer to display what they want on the screen, which gives them a great sense of accomplishment. If we were in the same country, I would definitely buy you a beer.

To get to the point, I found a great value 2.9 inch e-ink display on Taobao Thank you so much for sharing this project, my friend. I can't imagine how much time and effort you have put into it. This project has brought me and my children a lot of joy, and they can easily operate the computer to display what they want on the screen, which gives them a great sense of accomplishment. If we were in the same country, I would definitely buy you a beer.

To get to the point, I found a great value 2.9 inch e-ink display on Taobao https://item.taobao.com/item.htm?id=673989348784

. The display area is 29*66mm, and the best thing about e-ink displays is that they can keep displaying content even after power off, it's like magic, isn't it? However, its driver chip is SSD1680Z, and at the moment, I don't have the ability to make the driver and add it to the u8g2 project. To be honest, I couldn't even read your code. So I was wondering if you have considered adding support for the SSD1680Z chip? If it's possible, I would like to gift you two screens to support your development. Looking forward to your reply.
[image]
[image]
[image]

By the way, I also made a logo with a resolution of 128*23, here is the display effect and here is the source code.. The display area is 29*66mm, and the best thing about e-ink displays is that they can keep displaying content even after power off, it's like magic, isn't it? However, its driver chip is SSD1680Z, and at the moment, I don't have the ability to make the driver and add it to the u8g2 project. To be honest, I couldn't even read your code. So I was wondering if you have considered adding support for the SSD1680Z chip? If it's possible, I would like to gift you two screens to support your development. Looking forward to your reply.

By the way, I also made a logo with a resolution of 128*23, here is the display effect and here is the source code.`void drawLogo12832(u8g2_t *u8g2)
{
    u8g2_SetFontMode(u8g2, 1); /*字体模式选择*/
    u8g2_SetFontDirection(u8g2, 0); /*字体方向选择*/
    u8g2_SetFont(u8g2, u8g2_font_inb19_mr); /*字库选择*/
    u8g2_DrawStr(u8g2, 0, 19, "U");

    u8g2_SetFontDirection(u8g2, 1);
    u8g2_SetFont(u8g2, u8g2_font_inb19_mn);
    u8g2_DrawStr(u8g2, 15, 6, "8");

    u8g2_SetFontDirection(u8g2, 0);
    u8g2_SetFont(u8g2, u8g2_font_inb19_mf);
    u8g2_DrawStr(u8g2, 36, 19, "g");
    u8g2_DrawStr(u8g2, 51, 19, "\xb2");

    u8g2_DrawHLine(u8g2, 0, 23, 36);
    u8g2_DrawHLine(u8g2, 1, 24, 36);
    u8g2_DrawVLine(u8g2, 32, 21, 8);
    u8g2_DrawVLine(u8g2, 33, 22, 8);

    u8g2_SetFont(u8g2, u8g2_font_lucasfont_alternate_tf);
    u8g2_DrawStr(u8g2, 63, 13, "  github.com");
    u8g2_DrawStr(u8g2, 55, 23, "olikraus/u8g2");
}`

olikraus, 2023-05-03:
U8g2 support for epaper is very limited. In general u8g2 will not support epaper any more. There are better libraries like https://github.com/ZinggJM/GxEPD2

piogrammer, 2023-05-08:
Okay bro, I understand. Thanks again. I  bought 2 screens, and when I have time, I may study it, but I can't say for sure, after all, my coding is amateur   T_T