olikraus/u8g2 #2170: LD7032 - I2C Problem

MightyBenster, 2023-05-07:
Hi Olikraus,
Thanks for all your work here so far - I've gotten a tiny 0.5 inch LD7032 OLED working via SPI.

However, my use requires I2C, which you note as 'not tested'.
It appears to make the display operate, but not correctly.

I'm not familiar with libraries, and would appreciate a push in the right direction - it may be an easy fix (incorrect pixel dimensions somewhere?). I also don't know how it found the I2C address.

If you could spare some time to check this out, or even help me understand how the library works a little deeper, it would be much appreciated.
I also wonder if I can donate for your support? I can't offer much, but like to play a part.

Thanks,
Ben

```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

//U8G2_LD7032_60X32_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 9, /* data=*/ 8, /* cs=*/ 11, /* dc=*/ 10, /* reset=*/ 12);	// SW SPI Nano Board
U8G2_LD7032_60X32_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 9, /* data=*/ 8, /* reset=*/ U8X8_PIN_NONE);	// NOT TESTED!

//U8G2_LD7032_60X32_ALT_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 11, /* data=*/ 12, /* cs=*/ 9, /* dc=*/ 10, /* reset=*/ 8);	// SW SPI Nano Board
//U8G2_LD7032_60X32_ALT_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 9, /* data=*/ 8, /* reset=*/ U8X8_PIN_NONE);	// NOT TESTED!

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(1000);  
}
```

[image]

olikraus, 2023-05-08:
Did you update the PCB to operatie in I2C mode? If I remember correctly, then there must be a jumper or solder pad, which must be changed.

MightyBenster, 2023-05-08:
Thanks for your fast response!
Yes you're right, there is a Jumper to change from SPI to I2C.

Before jumping, there was no response to I2C.
After, there is no response using SPI, while I2C shows as above.

Although not displaying correctly, the _HelloWorld.ino_ does appear to print in part, at the very top of the screen.
I will try to take a better photo tonight.

MightyBenster, 2023-06-08:
Sorry for the delay in response, I have been busy reading!
It appears my I2C problems are not a one-off, issue [#865](https://github.com/olikraus/u8g2/issues/865)

I have fresh boards to rule out hardware errors:
- With the resistor in original position (R4), SW SPI works reliably.
- Moving the resistor for I2C (R5) produces no display. It will not work in either SPI or I2C modes.

I have followed the issue above but can find no resolution.
Could you make any suggestion for me?

I have tried:
- confirm SPI stops after moving resistor
- confirm I2C pins with another device
- I cannot see Display I2C address, due to one-way level-shifter

MightyBenster, 2023-06-08:
I have also tried the workaround suggested in issue #865 to no avail.
```
p = arg_ptr;
while( arg_int > 64 )
      {
	u8x8->byte_cb(u8x8, U8X8_MSG_CAD_SEND_DATA, 64, p);
	arg_int-=64;
	p+=64;
```

MightyBenster, 2023-06-08:
Yes, these are the constructors I have been trying;
```
//U8G2_LD7032_60X32_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 9, /* data=*/ 8, /* cs=*/ 11, /* dc=*/ 10, /* reset=*/ 12);	// SW SPI Nano Board
//U8G2_LD7032_60X32_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 9, /* data=*/ 8, /* reset=*/ U8X8_PIN_NONE);	// NOT TESTED!

//U8G2_LD7032_60X32_ALT_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 11, /* data=*/ 12, /* cs=*/ 9, /* dc=*/ 10, /* reset=*/ 8);	// SW SPI Nano Board
//U8G2_LD7032_60X32_ALT_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 9, /* data=*/ 8, /* reset=*/ U8X8_PIN_NONE);	// NOT TESTED!
```

Both SPI appear to work.
Neither I2C works.

MightyBenster, 2023-06-08:
I think above was Hardware error, as you suggested - perhaps the resistor move was not correct.

I have got new boards/displays, moved the resistor, and now see nothing on the display - which i am more confident about.

But yes, I have used both the F and ALT_F constructors.
Neither gives response

olikraus, 2023-06-08:
So... All fine now?

olikraus, 2023-06-08:
I am not sure whether i2c ever was tested. Let me check.

olikraus, 2023-06-08:
CAD reference: https://github.com/olikraus/u8g2/blob/a143183f0d117d57572a3646567a78900b0de75d/tools/codebuild/codebuild.c#L777
u8x8_cad_ld7032_i2c

Code is here: https://github.com/olikraus/u8g2/blob/a143183f0d117d57572a3646567a78900b0de75d/csrc/u8x8_cad.c#L657

As far as  I can see in the code, it looks like, that I tested this successfully. 
Maybe try different I2C addresses (https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress):

u8g2.setI2CAddress(0x060);
u8g2.setI2CAddress(0x061);
u8g2.setI2CAddress(0x062);
u8g2.setI2CAddress(0x063);
u8g2.setI2CAddress(0x064);
u8g2.setI2CAddress(0x065);
u8g2.setI2CAddress(0x066);
u8g2.setI2CAddress(0x067);

closed 2023-06-13 by MightyBenster