olikraus/u8g2 #2172: 64x48 I2C SSD1317 OLED

fabrice1236, 2023-05-08:
Hi,

so I've bought a screen from aliexpress, but I can't get it to work with my Arduino Due, I'm using `U8G2_SSD1306_64X48_ER_1_HW_I2C u8g2(U8G2_R0, 8)`.. When I run a basic sketch that looks for I2C devices, I can only see the display for 5-10 secs after uploading the code/powering the arduino, then nothing.. I've tried connecting the RST pin of the display to pin 8, but it didn't change anything.. Do you have any idea of how I could fix this ? Also, I am using SDA1 and SCL1, which means I'm using Wire1, if that changes something. The display I am talking about is the same as in #2115.

olikraus, 2023-05-08:
Maybe a power supply issue?

fabrice1236, 2023-05-08:
Checked everything with a multimeter, and it all seemed alright on my side..

olikraus, 2023-05-08:
From my side also everything seems to be alright, at least according the information you provided.

fabrice1236, 2023-05-08:
I'm suspecting it's probably something with the screen itself, since other I2C devices work, but I find it really strange that the I2C scanner can only find for a few seconds after the arduino is powered up..

olikraus, 2023-05-08:
Again... it looks like a power issue. 
- Always solder
- Do not trust the power output of the Arduino

olikraus, 2023-05-08:
Voltage is one point. But the question is also how much current is drawn by the display. What is the current consumption of your display? Arduino Uno will not deliver more than 100mA

fabrice1236, 2023-05-15:
[image]
I don't have data concerning it's power consumption (expected for a 0.80$ aliexpress screen), but here is it's size compared to my arduino due, I don't think that power consumption could be an issue considering its so tiny..

olikraus, 2023-05-15:
Correct, current consumption should be an issue. So you are saying, that a 64x32 works with u8g2 but a 64x48 doesn't work, correct?

Can you test the 64x48 with a the 128x64 SSD1306 noname constructor? Maybe try the U8g2 logo. If you see something there, please post the result as a picture.

fabrice1236, 2023-05-16:
Yes, a 64x32 works but not the 64x48, exactly.

I've tried running the logo example code with the `U8G2_SSD1306_128X64_NONAME_F_HW_I2C` constructor, and while I writing these lines, it wasn't working, but now it does for some reason ! It's not incredibly stable, and it just stops working if something moves by a millimeter, but I wouldn't be surprised if it's because of my cheap dupont wires maybe ?
[image]

fabrice1236, 2023-05-17:
So, I suspect that the pin headers on the display are the reason for the instability, since using the exact same setup with my other (64x32) display is perfectly fine..

> So you did already see something? This could also indicate that there is a wiring issue...

My full sentence was 

> **When I run a basic sketch that looks for I2C devices**, I can only see the display for 5-10 secs after uploading the code/powering the arduino

I meant that I could see that it's connected to the arduino, since using a basic I2C scanner script outputs the display's address (though only for a short period of time)

fabrice1236, 2023-05-17:
No, I meant that I ran a I2C scanner sketch on my arduino, which outputed the display's address on the arduino's serial monitor.

olikraus, 2023-05-18:
Hmm... ok... I think it is a wiring issue. I you dont't want to solder your display, perhaps try to avoid the proto board. You could use a wire with two female connectors to connect the display directly to the Arduino board.