olikraus/u8g2 #2178: wrong prototype on getU8x8(void)

deHarro, 2023-05-19:
Hi Oli,
nice work!

You probably made a minor mistake in  the description of 

[C++ Prototype of getU8x8](https://github.com/olikraus/u8g2/wiki/u8x8reference#getu8x8)

Returning a pointer this should not read as "**void** U8X8::getU8x8(void)".

Cheers
Harald

olikraus, 2023-05-19:
Nice finding. Thanks!

olikraus, 2023-05-19:
fixed...

closed 2023-05-19 by olikraus

deHarro, 2023-05-19:
Fast! Super, Thanks!

Why did I stumble over this?
I made a four display temperature panel (don't ask... "he" wanted 4 displays... ;)
To handle the things for the different displays I always found myself setting up switch/case costructs containing 4 identical routines, only differing in the name of the display (something like display1.clearDisplay(), display2.clearDisplay(),...) You get it.

Now I searched for a feature to iterate through an array of u8x8 items in a for loop and thought, fine, getU8x8() will do the trick for me.
But I failed :(
Is there a way to handle the displays in the described manner?

Thanks!
Harald

olikraus, 2023-05-19:
You need the constructor four times. Once the objects (display1, ... display4) are created, you can put them into an array:
```
U8G2 *array[4];
...
array[0] = &display1;
...
```

olikraus, 2023-05-19:
also possible, depends, on which function you want to use... the C or C++ interface.

deHarro, 2023-05-19:
I 'think' I use the C++ interface (Arduino), but I'm not a professional coder 😅

Main thing: Your hint works as expected, I already changed most routines accordingly and saved a lot of valuable flash space.