olikraus/u8g2 #2179: Issue with converting XBM frame to .bin format

sakibyousuf, 2023-05-19:
Hello,

I am experiencing an issue while converting my XBM frame to .bin format for storage on an SD card. The conversion process is not working correctly when I try to read the resulting .bin file. However, it seems to work fine for the example provided in the documentation.

Here is the XBM frame I am trying to convert:
`
#define u8g2_logo_128x64_width 128
#define u8g2_logo_128x64_height 64
static const unsigned char u8g2_logo_128x64_bits[] U8X8_PROGMEM = { 
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x54,0x01,0x00,0x00,0x00,0x00,0x80,0xb6,0x02,0x00,0x00,0x00,
0x00,0x00,0x80,0xff,0x7f,0x00,0x00,0x00,0x00,0xf0,0xff,0x0f,0x00,0x00,0x00,
0x00,0x00,0xe0,0xff,0xff,0x00,0x00,0x00,0x00,0xfc,0xff,0x3f,0x00,0x00,0x00,
0x00,0x00,0xe0,0xff,0xff,0x01,0x00,0x00,0x00,0xfc,0xff,0x3f,0x00,0x00,0x00,
0x00,0x00,0xf0,0xff,0xff,0x03,0x00,0x00,0x00,0xfe,0xff,0x7f,0x00,0x00,0x00,
0x00,0x00,0xf0,0xff,0xff,0x03,0x00,0x00,0x00,0xff,0xff,0x7f,0x00,0x00,0x00,
0x00,0x00,0xf8,0xff,0xff,0x07,0x00,0x00,0x00,0xff,0xff,0xff,0x00,0x00,0x00,
0x00,0x00,0xf8,0xff,0xff,0x07,0x00,0x00,0x00,0xff,0xff,0xff,0x00,0x00,0x00,
0x00,0x00,0xf8,0xff,0xff,0x07,0x00,0x00,0x00,0xff,0xff,0xff,0x00,0x00,0x00,
0x00,0x00,0xf8,0xff,0xff,0x07,0x00,0x00,0x00,0xff,0xff,0xff,0x00,0x00,0x00,
0x00,0x00,0xfc,0xff,0xff,0x07,0x00,0x00,0x00,0xff,0xff,0xff,0x00,0x00,0x00,
0x00,0x00,0xf8,0xff,0xff,0x07,0x00,0x00,0x00,0xff,0xff,0xff,0x01,0x00,0x00,
0x00,0x00,0xfc,0xff,0xff,0x07,0x00,0x00,0x00,0xff,0xff,0xff,0x00,0x00,0x00,
0x00,0x00,0xf8,0xff,0xff,0x03,0x00,0x00,0x00,0xfe,0xff,0xff,0x01,0x00,0x00,
0x00,0x00,0xfc,0xff,0xff,0x07,0x00,0x00,0x00,0xff,0xff,0xff,0x00,0x00,0x00,
0x00,0x00,0xfc,0xff,0xff,0x03,0x00,0x00,0x00,0xff,0xff,0xff,0x01,0x00,0x00,
0x00,0x00,0xf8,0xff,0xff,0x07,0x00,0x00,0x00,0xfe,0xff,0xff,0x00,0x00,0x00,
0x00,0x00,0xfc,0xff,0xff,0x07,0x00,0x00,0x00,0xff,0xff,0xff,0x01,0x00,0x00,
0x00,0x00,0xfc,0xff,0xff,0x03,0x00,0x00,0x00,0xfe,0xff,0xff,0x00,0x00,0x00,
0x00,0x00,0xf8,0xff,0xff,0x07,0x00,0x00,0x00,0xff,0xff,0xff,0x01,0x00,0x00,
0x00,0x00,0xfc,0xff,0xff,0x07,0x00,0x00,0x00,0xff,0xff,0xff,0x00,0x00,0x00,
0x00,0x00,0xfc,0xff,0xff,0x03,0x00,0x00,0x00,0xfe,0xff,0xff,0x01,0x00,0x00,
0x00,0x00,0xf8,0xff,0xff,0x07,0x00,0x00,0x00,0xff,0xff,0xff,0x00,0x00,0x00,
0x00,0x00,0xfc,0xff,0xff,0x07,0x00,0x00,0x00,0xff,0xff,0xff,0x01,0x00,0x00,
0x00,0x00,0xf8,0xff,0xff,0x07,0x00,0x00,0x00,0xff,0xff,0xff,0x00,0x00,0x00,
0x00,0x00,0xf8,0xff,0xff,0x07,0x00,0x00,0x00,0xff,0xff,0xff,0x00,0x00,0x00,
0x00,0x00,0xf8,0xff,0xff,0x07,0x00,0x00,0x00,0xff,0xff,0xff,0x00,0x00,0x00,
0x00,0x00,0xf8,0xff,0xff,0x07,0x00,0x00,0x00,0xff,0xff,0xff,0x00,0x00,0x00,
0x00,0x00,0xf0,0xff,0xff,0x03,0x00,0x00,0x00,0xff,0xff,0x7f,0x00,0x00,0x00,
0x00,0x00,0xf0,0xff,0xff,0x03,0x00,0x00,0x00,0xfe,0xff,0x7f,0x00,0x00,0x00,
0x00,0x00,0xf0,0xff,0xff,0x03,0x00,0x00,0x00,0xfe,0xff,0x7f,0x00,0x00,0x00,
0x00,0x00,0xe0,0xff,0xff,0x01,0x00,0x00,0x00,0xfc,0xff,0x3f,0x00,0x00,0x00,
0x00,0x00,0xc0,0xff,0xff,0x00,0x00,0x00,0x00,0xf8,0xff,0x1f,0x00,0x00,0x00,
0x00,0x00,0x00,0xff,0x3f,0x00,0x00,0x00,0x00,0xf0,0xff,0x07,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x92,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00
};
`
I have followed the provided instructions for the conversion process, but the resulting .bin file does not seem to be correct. 
Could you please assist me in resolving this issue? It would be greatly appreciated.

Thank you for your attention to this matter.

olikraus, 2023-05-19:
Please send the complete code along with the error description

sakibyousuf, 2023-05-20:
thanks for your timely reply here is my code. 
`
#include <FS.h>
#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>
#include <SD.h>

#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

#define acc99_width 128
#define acc99_height 64
static const unsigned char acc99[] U8X8_PROGMEM = {
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x80,0x01,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0xc0,0x01,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x60,0xc0,0x00,0x03,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x60,0x80,0x01,0x03,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x40,0xc0,0x01,
0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x30,0xc0,0xc0,
0x01,0x03,0x06,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x70,0x40,
0xc0,0x00,0x01,0x07,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x60,
0x00,0x80,0x01,0x00,0x03,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0xe0,0x00,0xc0,0x00,0x80,0x03,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0xc0,0x01,0x00,0x00,0x80,0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0xc0,0x01,0x00,0x00,0xc0,0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x80,0x01,0x00,0x00,0xc0,0x00,0x02,0x00,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x60,0x80,0x01,0x00,0x00,0xc0,0x80,0x03,0x00,0x00,0x00,0x00,
0x00,0x00,0x00,0x00,0xc0,0x00,0x00,0x00,0x00,0x00,0x80,0x01,0x00,0x00,0x00,
0x00,0x00,0xfa,0xff,0x00,0x80,0x01,0x00,0x00,0x00,0x04,0xc0,0x00,0x00,0xa8,
0x0a,0x00,0x80,0xff,0xff,0x03,0x00,0x00,0x00,0x00,0x00,0x04,0x00,0x00,0x80,
0xff,0xff,0x01,0xe0,0xff,0xff,0x07,0x00,0x00,0x00,0x00,0x00,0x06,0x00,0x00,
0xe0,0xff,0xff,0x07,0xf0,0xff,0xff,0x0f,0x00,0x00,0x00,0x00,0x00,0x02,0x00,
0x00,0xe0,0xff,0xff,0x0f,0xf0,0xff,0xff,0x87,0x00,0x00,0x00,0x00,0x00,0x03,
0x00,0x80,0xe0,0xff,0xff,0x1f,0xf8,0xff,0xff,0xc7,0x03,0x00,0x00,0x00,0x00,
0x01,0x00,0xe0,0xe0,0xff,0xff,0x1f,0xf8,0xff,0xff,0x87,0x07,0x00,0x00,0x00,
0x80,0x01,0x00,0xf8,0xe0,0xff,0xff,0x3f,0xfc,0xff,0xff,0x07,0x3e,0x00,0x00,
0x00,0x80,0x00,0x00,0x3c,0xc0,0xff,0xff,0x3f,0xfc,0xff,0xff,0x03,0x3c,0x00,
0x00,0x00,0xc0,0x00,0x00,0x1f,0xc0,0xff,0xff,0x7f,0xfc,0xff,0xff,0x03,0x30,
0x00,0x00,0x00,0xc0,0x00,0x00,0x06,0xc0,0xff,0xff,0x7f,0xfe,0xff,0xff,0x01,
0x00,0x00,0x00,0x00,0x60,0x00,0x00,0x00,0x80,0xff,0xff,0x7f,0xfc,0xff,0xff,
0x01,0x00,0x00,0x00,0x00,0x60,0x00,0x00,0x00,0x80,0xff,0xff,0x7f,0xfe,0xff,
0xff,0x00,0x00,0x00,0x00,0x00,0x30,0x00,0x00,0x00,0x00,0xff,0xff,0xff,0xfe,
0xff,0xff,0x00,0x00,0x00,0x00,0x00,0x30,0x00,0x00,0x00,0x04,0xfe,0xff,0xff,
0xfe,0xff,0xff,0x78,0x00,0x00,0x00,0x00,0x18,0x00,0x00,0x80,0x0f,0xfe,0xff,
0xff,0xfe,0xff,0x7f,0xf0,0x00,0x00,0x00,0x00,0x18,0x00,0x00,0x80,0x05,0xfe,
0xff,0xff,0xfe,0xff,0x7f,0x00,0x00,0x00,0x00,0x00,0x0c,0x00,0x00,0x00,0x0e,
0xfe,0xff,0xff,0xfe,0xff,0x7f,0x00,0x00,0x00,0x00,0x00,0x0c,0x00,0x00,0x00,
0x0e,0xfc,0xff,0xff,0xfe,0xff,0x7f,0x00,0x00,0x00,0x00,0x00,0x06,0x00,0x00,
0x00,0x1c,0xfc,0xff,0xff,0xfe,0xff,0x7f,0x00,0x00,0x00,0x00,0x00,0x06,0x00,
0x00,0x00,0x1e,0xfc,0xff,0xff,0xfe,0xff,0x7f,0x00,0x00,0x00,0x00,0x00,0x07,
0x00,0x00,0x00,0x1c,0xfc,0xff,0xff,0xfe,0xff,0x7f,0x00,0x00,0x00,0x00,0x00,
0x03,0x00,0x00,0x00,0x1c,0xfc,0xff,0xff,0xfe,0xff,0x3f,0x00,0x00,0x00,0x00,
0x80,0x03,0x00,0x00,0x00,0x04,0xfc,0xff,0xff,0xfe,0xff,0x7f,0xfc,0x03,0x00,
0x00,0x80,0x01,0x00,0x00,0xf0,0x1f,0xfc,0xff,0xff,0xfe,0xff,0x7f,0xfe,0x07,
0x00,0x00,0xc0,0x01,0x00,0x00,0xe0,0x3f,0xfc,0xff,0xff,0xfe,0xff,0x7f,0x00,
0x00,0x00,0x00,0xe0,0x00,0x00,0x00,0x00,0x0c,0xfc,0xff,0xff,0xfe,0xff,0x7f,
0x00,0x00,0x00,0x00,0xe0,0x00,0x00,0x00,0x00,0x1c,0xfc,0xff,0xff,0xfe,0xff,
0x7f,0x00,0x00,0x00,0x00,0x60,0x00,0x00,0x00,0x00,0x1c,0xfc,0xff,0xff,0xfe,
0xff,0x7f,0x00,0x00,0x00,0x00,0x70,0x00,0x00,0x00,0x00,0x1e,0xfc,0xff,0xff,
0xfe,0xff,0x7f,0x00,0x00,0x00,0x00,0x20,0x00,0x00,0x00,0x00,0x0e,0xfc,0xff,
0xff,0xfe,0xff,0x7f,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x1e,0xfe,
0xff,0xff,0xfe,0xff,0xff,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x80,0x04,
0xfe,0xff,0xff,0xfe,0xff,0xff,0xe0,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x80,
0x0f,0xfe,0xff,0xff,0xfe,0xff,0xff,0xf0,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
[... 98 lines cut ...]
  }
  Serial.println("initialization done.");

  // only to ensure that there is some example data on the SD card...
  writeFile("/acc99.bin", acc99_width, acc99_height, acc99);

  u8g2.begin();

}

void loop() {
  u8g2.clearDisplay();
  u8g2.setFont(u8g2_font_helvR10_tr);
  u8g2.firstPage();
  do {
    // u8g2.setCursor(0, 12);
    // u8g2.print(F("Read from SD:"));
    drawFile(0, 0, "/acc99.bin");
    
  } while ( u8g2.nextPage() );
  delay(1000);
}
`
if i use the same xbm data with `u8g2.drawXBM( 0, 0, acc99_width, acc99_height, acc99);` the image is Shown properly
also if i use the xbm data frame thats in the example  [https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/LoadFromSD/LoadFromSD.ino](url) and save that in sd card that also works fine.

sakibyousuf, 2023-05-20:
**this is the result when i fetch .bin file from sd card **
[image]

olikraus, 2023-05-21:
Hmmm, the example 
https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/LoadFromSD/LoadFromSD.ino
works as expected, right?
And your code only differs by the XBM bitmap, correct?
Actually I do not see a difference between your code and the example. Strange...

sakibyousuf, 2023-05-21:
yes my code only differs by xbm bitmap. i almost checked all the image to xbm converters , gimp,etc  tried with png to xbm, jpg to xbm. also wrote a python script to covert image to xbm but nothing works at all. Can know which software or script you used to convert image to xbm.
as i am not too expert in the image processing i think the issue lies when i convert my xbm bitmap to .bin with the code that is given in the example. In my xbm bitmap there are 15 columns and in the example code xbm bitmap there are 12 columns.

 thanks

sakibyousuf, 2023-05-21:
**Also checked with the xbm bitmap of **  [https://github.com/olikraus/U8g2_Arduino/blob/master/examples/page_buffer/GraphicsTest/GraphicsTest.ino](url) but didn't worked 
following is the xbm bitmap from GraphicsTest.ino
`#define cross_width 24`
`#define cross_height 24`
`static const unsigned char cross_bits[] U8X8_PROGMEM  = {
  0x00, 0x18, 0x00, 0x00, 0x24, 0x00, 0x00, 0x24, 0x00, 0x00, 0x42, 0x00, 
  0x00, 0x42, 0x00, 0x00, 0x42, 0x00, 0x00, 0x81, 0x00, 0x00, 0x81, 0x00, 
  0xC0, 0x00, 0x03, 0x38, 0x3C, 0x1C, 0x06, 0x42, 0x60, 0x01, 0x42, 0x80, 
  0x01, 0x42, 0x80, 0x06, 0x42, 0x60, 0x38, 0x3C, 0x1C, 0xC0, 0x00, 0x03, 
  0x00, 0x81, 0x00, 0x00, 0x81, 0x00, 0x00, 0x42, 0x00, 0x00, 0x42, 0x00, 
  0x00, 0x42, 0x00, 0x00, 0x24, 0x00, 0x00, 0x24, 0x00, 0x00, 0x18, 0x00, };`

olikraus, 2023-05-21:
> yes my code only differs by xbm bitmap. i almost checked all the image to xbm converters , gimp,etc tried with png to xbm, jpg to xbm. also wrote a python script to covert image to xbm but nothing works at all. Can i know which software or script you used to convert image to xbm. as i am not too expert in the image processing i think the issue lies when i convert my xbm bitmap to .bin with the code that is given in the example. In my xbm bitmap there are 15 columns and in the example code xbm bitmap there are 12 columns.
> 
> thanks

I have used gimp to convert monochrome images to xbm. 
A tutorial is here: https://youtu.be/ac5CYQ3OJ7c

The binary file on the SD card is created by the example code.

sakibyousuf, 2023-05-21:
Thanks for your reply but i have already done that procedure and didn't get any success.
Can you check with the xbm bitmap that's provided in the graphics test.ino that also doesn't work at all

olikraus, 2023-05-21:
> Can you check with the xbm bitmap that's provided in the graphics test.ino that also doesn't work at all

What exactly does not work? Does graphics_test.ino work?

sakibyousuf, 2023-05-22:
Yes graphics_test.ino works fine but if i convert any one of the xbm bitmap into .bin those used in graphics_test.ino as mentioned above that doesn't work at all

olikraus, 2023-05-23:
How did you do the conversation to .bin?

sakibyousuf, 2023-05-23:
Using the code in the example that is provided
[https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/LoadFromSD/LoadFromSD.ino]()  only replaced the the xbm bitmap

olikraus, 2023-05-28:
Strange... that should work...

sakibyousuf, 2023-05-28:
You can test it. Doesn't work at all.

olikraus, 2023-05-28:
Maybe the xbm file itself is wrong.

sakibyousuf, 2023-05-28:
I tried almost everything. Can you please help me out

olikraus, 2023-05-28:
I am afraid, I don't have any setup to test this. I mean I don't have an SD card reader installed.

sakibyousuf, 2023-05-28:
Oh so how can we figure it out. Or do you have any other xbm bitmap so that I can test that. Not the xbm bitmap that's in the example

olikraus, 2023-05-28:
Maybe post your complete code here.

sakibyousuf, 2023-05-28:
Posted already  [](https://github.com/olikraus/u8g2/issues/2179#issuecomment-1555560000)

sakibyousuf, 2023-05-28:
Posted already [https://github.com/olikraus/u8g2/issues/2179#issuecomment-1555560000]()

olikraus, 2023-05-28:
Where?

olikraus, 2023-05-28:
> Posted already https://github.com/olikraus/u8g2/issues/2179#issuecomment-1555560000

This is just your xbm file. Please post the complete .ino file including your xbm code, which is used to write the .bin file and which, as you wrote, doesn't work.

olikraus, 2023-05-28:
> > Posted already [#2179 (comment)](https://github.com/olikraus/u8g2/issues/2179#issuecomment-1555560000)
> 
> This is just your xbm file. Please post the complete .ino file including your xbm code, which is used to write the .bin file and which, as you wrote, doesn't work.

ah, excause, some issue with my browser... let me recheck your code

olikraus, 2023-05-28:
ok, i think the code is correct as far as I can see. What is your Arduino Board by the way?

olikraus, 2023-05-28:
Ideas for further testing.
You could forcefully assign the dimensions, just to see whether the image is correct:
``` C++
  if (myFile) {
    w = myFile.read();        // read the dimension of the bitmap
    h = myFile.read();
    while( h > 0 ) {          // handle all lines of the bitmap
```
by
``` C++
  if (myFile) {
    w = myFile.read();        // read the dimension of the bitmap
    h = myFile.read();
    w = 128;
    h = 64;
    while( h > 0 ) {          // handle all lines of the bitmap
```
How will it look like now?

sakibyousuf, 2023-05-28:
I am using esp32

sakibyousuf, 2023-05-28:
> 

tried to forcefully assign the dimensions but didn't worked

sakibyousuf, 2023-05-28:
@olikraus  i found the issue the conversion to bin code only converts a xbm bimpa that has dimensions of width 97 and height 51.
i resized my xbm bitmap to 97x51 that worked fine..  what i think is issue lies with this line  w = (w+7)/8;

olikraus, 2023-05-28:
Strange, maybe try 117x64 instead of 128x64

sakibyousuf, 2023-05-28:
yes 117x64 works perfectly. what is the issue with 128x64 or other size

olikraus, 2023-05-29:
hmmm....

I think the loop is not 100% correct:

``` C++
      while(len > 0) {        // draw all pixel of one line
        if ( b & mask ) {     // check one pixel
          u8g2.setDrawColor(1);
          u8g2.drawPixel(xpos,y);
        } else {
          u8g2.setDrawColor(0);
          u8g2.drawPixel(xpos,y);
        }
        xpos++;               // calculate next x pos of the pixel
        mask <<= 1;           // update the mask
        if ( mask == 0 )
        {
          mask = 1;           // revert mask and ...
          b = myFile.read();  // ... load the next 8 pixel values from the file
        }
        len--;                // decrease the horizontal width (remaining pixel)
      }
```
instead it should be:

``` C++
      for(;;) {        // draw all pixel of one line
        if ( b & mask ) {     // check one pixel
          u8g2.setDrawColor(1);
          u8g2.drawPixel(xpos,y);
        } else {
          u8g2.setDrawColor(0);
          u8g2.drawPixel(xpos,y);
        }
        xpos++;               // calculate next x pos of the pixel
        mask <<= 1;           // update the mask
        len--;                // decrease the horizontal width (remaining pixel)
        if ( len == 0 )    // check if row is finished
            break;
        if ( mask == 0 )
        {
          mask = 1;           // revert mask and ...
          b = myFile.read();  // ... load the next 8 pixel values from the file
        }
      }
```

sakibyousuf, 2023-05-30:
@olikraus i tested the new code it is working perfectly. i tried it with multiple size xbm bitmaps all is working fine you can add this code to the example.
thanks

olikraus, 2023-06-03:
Great, thanks for the feedback!

sakibyousuf, 2023-06-03:
Can we decrease the rendering time.
My gif is about 145 xbm bitmap frames  but it takes a lot of time from sdcard to rendering.
Is there any way to make it fast

olikraus, 2023-06-03:
> Is there any way to make it fast

Sure:
1. Measure your code, identify bottlenecks.
2. Optimize the code.

If you still think, that u8g2 is the problem, you might use the direct buffer API:
https://github.com/olikraus/u8g2/wiki/u8g2reference#direct-access-buffer-api

Store the xbm in native format of the direct buffer memory and write the data directly into the memory.

closed 2023-06-12 by olikraus