olikraus/u8g2 #2180: [PROBLEM] Unable to get equal glyph width with generated monospace font

zalexzperez, 2023-05-20:
I want to display a number from 0 to 255 at the horizontal center of my OLED display. However, using getStrWidth doesn't return the same amount of pixels for each number glyph, even though the font is monospaced at a width of 43px for each glyph.  

Using [this font](https://www.fontsquirrel.com/fonts/M-1m), for instance.

Used FontForge to convert it to .bdf format and then bdfconv to get the .c file.

This is the pixel width I get for numbers 0-10 with getStrWidth

0: 43 pixels 
1: 35 pixels 
2: 42 pixels 
3: 41 pixels 
4: 42 pixels 
5: 42 pixels 
6: 42 pixels 
7: 45 pixels 
8: 42 pixels 
9: 41 pixels 

I'm attaching here 
[generatedFontfiles.zip](https://github.com/olikraus/u8g2/files/11522376/generatedFontfiles.zip)
the generated .bdf and .c files.

Thanks in advance.

olikraus, 2023-05-20:
Did you use the -b 2 option for bdfconv?

>  printf("-b <n>      Font build mode, 0: proportional, 1: common height, 2: monospace, 3: multiple of 8, 4: 5x7 mode\n");

zalexzperez, 2023-05-20:
I swear I tried looking for a list of bdfconv arguments but couldn't find it anywhere.

I used u8g2 unifont helper website and got this command:

bdfconv -v -f 1 -m "0-127" mplus-1m-bold-86.bdf -o mplus_1m_bold_86.c -n mplus_1m_bold_86 -d mplus-1m-bold-86.bdf

I will try with your suggested argument and report back. Thanks

zalexzperez, 2023-05-20:
@olikraus That worked! You saved me a lot of frustrations. 

Just one more unrelated question. How do you precisely get your fonts to be 64px in height with FontForge without trial and error? 
Setting the size in pixels to 86 on the Bitmap strikes available window made the font 65px tall and I got to that value by trying different values...

olikraus, 2023-05-20:
> I swear I tried looking for a list of bdfconv arguments but couldn't find it anywhere.

You need to call bdfconf with "-h" argument

> trial and error?

I used trial and error ;-)

I have used otf2bdf instead of fontforge.

closed 2023-05-21 by zalexzperez