olikraus/u8g2 #2182: I2C communication stops controller

Heideachim, 2023-05-25:
Hi, I have a ST7567S 128 x 64 LCD connected via I2C. After a random amount of time (seconds to minutes), the processor gets stuck in the sendBuffer() routine. Then, if unplug the module, the controller resumes like nothing had happend. It seems to wait for some response from the display that never comes. 
I tested this with an Arduino Nano and an Adafruit Feather M0. Both behaved the same.
This happens both for sendBuffer() as well using a frame buffer.
My constructor right now: U8G2_ST7567_JLX12864_F_HW_I2C u8g2(U8G2_R0);
I thinks it's the displays fault (two almost identical displays show the same problem), however, I2C communication problems should not freeze the entire board.
Probably something like issue #1218, but I cannot change to SPI.
Is this known behaviour and can I do anything about it?

Heideachim, 2023-05-25:
State of the I2C lines while it waits indefinitely: SDA is low (0.4V), SCL ist high (3.2V). That must be the acknowledge phase, since the 0.4V are generated by the display. The processor gets much closer to 0.0V. Is it true that the display does not release the SDA any more?

olikraus, 2023-05-28:
Will the problem also appear with "software emulated" I2C? It is indeed true, that the Arduino I2C implementation can block the board, because there is no timeout included.

Heideachim, 2023-05-29:
Switching to Software-I2C circumvents the problem, the display ran for more than an hour. Of course, it also changes I2C speed to about 1/10th of the hardware version. But this points into an interesting direction.
Then I tried the easy way, using Wire.setTimeout (with the hardware I2C, of course):
`
Wire.setTimeout(3000);
u8g2.sendBuffer();
`
Unfortunately, the problem was back again: the microcontroller / communication stalled after several seconds. 

Further steps on my side: 
1. Check the hardware, especially those really strange 100 ohms series resistors in the I2C-lines
2. Try different I2C speeds

Somebody got some more ideas, tell me! Thanks!

olikraus, 2023-05-29:
I2C requires pull up resistors on both lines. Which values did you use for those pull up resistors? 4K7 should be best.

Heideachim, 2023-05-29:
I even went down to 1 kOhm, since with 4.7k HIGH level is only at 2.4V (supply voltage is 3.3V). With 1k in parallel (yielding 820 Ohms) it went up to 2.8V, but then the LOW level (only when the display acknowledges transmission) goes up to 1.4V. In the end, that did not solve the problem, it remained exactly the same.
I did remove those 100 Ohms series resistors. They only made voltage levels even worse.

olikraus, 2023-05-30:
I am confused. There must be only one resistor involved, which is the pullup resistor for each line.

olikraus, 2023-05-30:
With how i2c try to reduce bus clock (with the corresponding u8g2 command). maybe this helps.

Heideachim, 2023-05-30:
I already tried 400000, 340000, 100000. It was always the same: After some time the communication stopped. But now I finally managed to record the very moment the bad stuff happens. See next post.

Heideachim, 2023-05-30:
Okay, here is a scope of the moment, when communication stalls. See the attached LibreOffice Calc file. The top channel (yellow) is SDA at 1V per division, the lower, blue one, is SCL at 2V per division.
There are four bytes in the transmission: First the display's address (0x7E), then the command (its data bytes that follow) and then two complete bytes transmissions and a third, incomplete one. Clearly we see the start bit (green) and the acknowledgements from the display, marked in red. They are easily recognizable since the low level from the display is about 0.4V instead of 0.0V. With the last bit, the ACK already is issued with the 7th bit. Then, the master stops the clock.
[Faulty communication ST7567A.ods](https://github.com/olikraus/u8g2/files/11604964/Faulty.communication.ST7567A.ods)
So, is my interpretation of this I2C is more or less correct??

Heideachim, 2023-06-01:
Since the root cause for this problem seems to be the less-then-optimal signals of the I2C, I moved the discussion over to the [electronics forum of stackexchange](https://electronics.stackexchange.com/questions/668769/i2c-display-rarely-acknowledges).
But even if the hardware is faulty: Is there absolutely no way NOT to block sendBuffer()? The Wire-library of Arduino contains a setTimout(). function. But putting this in setup does not help.

Even if a peripheral device is in a stalled state, it should never block the main processor. How can we achieve this?

olikraus, 2023-06-01:
I personally think, that the  voltage levels, caused by the additional resistors are the root cause for the problem. I don't see any need for 470 Ohm resistors. Hmm I see in your post in stackexchange, that you did this already.

> The Wire-library of Arduino contains a setTimout(). function.

When I started to write u8g2 such a function did not exist. Now it looks like, that every so called Arduino Core Library has implemented its own flavor:

Classic Arduino Code:
https://github.com/arduino/ArduinoCore-avr/blob/eabd762a1edcf076877b7bec28b7f99099141473/libraries/Wire/src/Wire.h#L58-L60

ESP32:
https://github.com/espressif/arduino-esp32/blob/72c41d09538663ebef80d29eb986cd5bc3395c2d/libraries/Wire/src/Wire.h#L106-L107

> Even if a peripheral device is in a stalled state, it should never block the main processor. How can we achieve this?

I agree, however...
Arduino Environment is a great simplification along with a lot of assumptions. Assuming that I2C never fails is one of those assumptions.

U8g2 is advertised as an Arduino lib. Although amazing projects can be created with Arduino,  i think it is not difficult to do any professional development with Arduino. Like you pointed out, the error handling is way to bad (at least in history). U8g2 itself has a core C library which does not include the I2C interface, instead I2C has to be added by the developer including all the error detection and recovery mechanisms: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-i2c-communication

For U8g2 Arduino lib I added the common subset of I2C handling...

Heideachim, 2023-06-02:
Okay, thanks for the clarification! 
Admittedly, it is easy for me as a user to stumble upon somthing in your open-source library and then state, that this should not have happend. This might be true, but it utterly and completely underestimates the complexity of this library. Running on so many platforms, dealing with hundreds of different displays driven by at least four different busses creates a three-dimensional solution room. And that is almost impossible to handle. Cudos to you for getting this far!!!

My system is an Adafruit Feather M0, so I have to investigate how thy implemented the Wire library.

One thing that helped a little bit is this part of U8x8lib.cpp: 
`/* software i2c,
ignores ACK response (which is anyway not provided by some displays)`

That at least explaines why software I2C is working perfectly on my display.

Heideachim, 2023-06-04:
Okay, I finally found a way how to evade this problem.
The original problem: Rarely, my I2C display acknowledges a transmission from the microcontroller already after the 8th bit, instead of after the 9th bit. After that, communication from the microcontroller stalled. Sometimes, I even found the µC stayed forever in the sendBuffer() routine and never came back.
The first step was to find a way to force to problem to occurr as fast as possible. With a 1.2kOhm pull-up on the SDA line this was the case (instead of the usual 3k...4.7kOhm resistor). 
Then it seems that the processor did only stall with a call to the **next** sendBuffer(), when in the preceeding loop the display acknowledged wrongly. 
Also I noticed that disconnecting and reconnecting the SDA line in the error case made the display and the µC run again. 
So I wrote a routine that does something like a reset of the display if it detects a stalled situation. 

First, I put the code for initializing the display in a separate routine:

```c++
bool display_initialized = false;

void init_display() {
  Wire.end();   // just in case this is not the first time that this function is called
  u8g2.setI2CAddress (display_address); 
  u8g2.setBusClock(400000);
  u8g2.begin();
  u8g2.setContrast (245); 
  u8g2.setFlipMode(true);
  Wire.setTimeout(3000);    // acts differently on different platforms, seems no to work on Adafruit SAMD21-Arduino
  display_initialized = true;
}
``` 
Then, I inserted a part right at the beginning of loop() that checks, if there is an error condition. That would be true if the I2C data line was low. If this is the case, reset the communication lines and init the display again:
```c++
void loop() {

  if (!display_initialized)
    init_display();

  while (digitalRead(SDA) == 0) {
    pinMode(SDA, OUTPUT); 
    digitalWrite(SDA, 0);   // pull down SDA line
    pinMode(SCL, OUTPUT);
    digitalWrite(SCL, 0);   // pull down SCL line
    delay(10);
    digitalWrite(SDA, 1);   // pull up SDA and SCL
    digitalWrite(SCL, 1);
    delay(10);
    init_display();   // force reset of display
  }

``` 
That helped once and forever. So every now and then the display goes blank because it is initializing but that is insignificant compared to a stalled processor.

Again, thanks to Oli for all his help and his great library!

closed 2023-06-04 by Heideachim

olikraus, 2023-06-04:
Wow, thanks for the detailed analysis.