olikraus/u8g2 #2183: LCD only works after board reset(ST7920

Saeedafshari11235, 2023-05-26:
Hey!
I'm having problem with ST7920 glcd.
It works well after I upload the program but whenever I turn off the power and reset the board it shows some unknown characters. it only works well after I push the reset button or reupload the program
```
#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>
#include "SPI.h"
#define CLK 25   //E on lcd
#define DATA 26  //R/W on lcd
#define CS 27    //RS on lcd
#define RST 33   //RST on lcd
U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R2, CLK, DATA, CS, RST);
void setup() {
  u8g2.begin();
  Wire.begin();
  u8g2.setContrast(255);
  u8g2.setBusClock(950000); 
}
void loop() {
  lcd1();
}
void lcd1() {
  u8g2.clearBuffer();
  u8g2.setDrawColor(1);
  u8g2.drawBox(0, 0, 128, 9);
  u8g2.setFont(u8g2_font_4x6_mf);
  u8g2.setDrawColor(0);
  u8g2.drawStr(25, 7, "0000/00/00 24:24:24");
  u8g2.setDrawColor(1);
  u8g2.drawStr(5, 17, "0000/00/00 00:00:00");
  u8g2.drawStr(5, 24, "0000/00/00 00:00:00");
  u8g2.drawStr(5, 31, "136 seconds");
  u8g2.drawStr(5, 38, ("RECORD COUNT: " + String(15)).c_str());
  u8g2.setFont(u8g2_font_unifont_t_arabic);
  int i = 81, j = 64;
  u8g2.setDrawColor(1);
  u8g2.drawBox(0, 53, 128, 15);
  u8g2.setDrawColor(0);
  u8g2.drawGlyph(i, j, 0xfebb);
  u8g2.drawGlyph(i - 6, j, 0xfed3);
  u8g2.drawGlyph(i - 11, j, 0xfea3);
  u8g2.drawGlyph(i - 16, j, 0xfeea);
  u8g2.drawGlyph(i - 20, j, 0x0020);
  u8g2.drawGlyph(i - 25, j, 0xfe8d);
  u8g2.drawGlyph(i - 30, j, 0xfebb);
  u8g2.drawGlyph(i - 35, j, 0xfee0);
  u8g2.drawGlyph(i - 40, j, 0xfef0);
  u8g2.drawGlyph(0, j - 1, 0x003c);
  u8g2.drawGlyph(120, j - 1, 0x003e);
  u8g2.sendBuffer();
}
```
[image]

olikraus, 2023-05-28:
Difficult to say. You code looks good.

Saeedafshari11235, 2023-06-13:
oh, I forgot something my board is esp32s.
this code worked well on arduino but on esp32...

olikraus, 2023-06-14:
The ST7920 is a 5V device, but the ESP32 is a 3.3V controller. Maybe the signal voltage level is not high enough.

Saeedafshari11235, 2023-06-15:
I found a way,
```
#include <EEPROM.h>
#define EEPROM_SIZE 1
int eprom = HIGH;
void setup() {
  Serial.begin(115200);
  EEPROM.begin(EEPROM_SIZE);
  eprom = EEPROM.read(0);
  delay(1000);
  if(!eprom){
    EEPROM.write(0,1);
    EEPROM.commit();
    Serial.println("Restarting...");
    delay(3000);
    ESP.restart();
  }
}

void loop() {
  EEPROM.write(0,0);
  EEPROM.commit();
  delay(1000);
}
```
this will resets the esp once and then it works as well as clock

closed 2023-06-15 by Saeedafshari11235