olikraus/u8g2 #2186: ST75256/ST7586S 320160G COG LCD Screen Support

AtlasTheProto, 2023-05-31:
Hello, I recently purchased [this display](https://www.aliexpress.us/item/3256804758550505.html) from aliexpress, and apparently neglected to notice this specific configuration wasn't yet supported by a constructor in u8g2 before it shipped.

Resolution: 320x160
Drive: 1/160 Duty, 1/12 Bias
Interface: 4-wire SPI, I2C, Parallel

Is there any chance that support for this display could be trivial? Or could be accomplished through a different constructor? It seems the IC shows up in the constructors list, just not the 320x160 resolution (shown below)

[ST75256 JLX256160](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st75256-jlx256160)
[ST75256 JLX256160M](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st75256-jlx256160m)
[ST75256 JLX256160_ALT](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st75256-jlx256160_alt)
[ST75256 JLX256160](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st75256-jlx256160)
[ST75256 JLX256160M](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st75256-jlx256160m)
[ST75256 JLX256160_ALT](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st75256-jlx256160_alt)
[ST75256 JLX240160](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st75256-jlx240160)
[ST75256 JLX240160](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st75256-jlx240160)
[ST7586S JLX384160](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7586s-jlx384160)
[ST7586S ERC240160](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7586s-erc240160)
[ST7586S YMC240160](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7586s-ymc240160)

AtlasTheProto, 2023-06-13:
2 weeks later, the screen has arrived! Ready to test whenever.

olikraus, 2023-06-18:
Actually I am a little bit confused by the description on aliexpress. The ST75256 driver doesn't support the 320 columns.
So I assume the chip is a ST75320 device. 
Can you test your display with the `U8G2_ST75320_JLX320240` constructor? You might need to adjust the contrast with `u8g2.setContrast()`. How will it look like?

AtlasTheProto, 2023-07-01:
currently unsure about the SPI pin configuration, image attached is what is shown on the store page, but the names on the list dont always line up with the module, should i follow 1:1? RS is listed on different pins and usually d0 is sck and d1 is data, just confused
[image]

olikraus, 2023-07-01:
The SPI pin connection seems to be ok and fits to the ST7586S datasheet: https://www.topwaydisplay.com/sites/default/files/2020-04/ST7586S.pdf page 16
So indeed the RS / A0 line moves to pin 17 in SPI mode.

Meanwhile the same display arrived here, but I was not able yet to work on this. 
Are you able to try one of the existing ST7586 constructors?

AtlasTheProto, 2023-07-01:
> The SPI pin connection seems to be ok and fits to the ST7586S datasheet: https://www.topwaydisplay.com/sites/default/files/2020-04/ST7586S.pdf page 16 So indeed the RS / A0 line moves to pin 17 in SPI mode.

Thank you for this clarification, will re-test soon

olikraus, 2023-07-03:
https://raw.githubusercontent.com/wiki/olikraus/u8g2/img/st7586s_jlx320160.jpg

No u8x8, no flipmode, but works as expected. I will create a beta release soon.

olikraus, 2023-07-03:
I have created beta release 2.35.2, it includes the new constructor U8G2_ST7586S_JLX320160...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

mspudi, 2023-07-29:
> [image]
> 
> No u8x8, no flipmode, but works as expected. I will create a beta release soon.

Is the ESP32 driver and wiring diagram provided? I tested all 384160 of the latest library but failed,

olikraus, 2023-07-29:
The wiring is the same as for almost all other SPI graphics devices, see also here: https://github.com/olikraus/u8g2/wiki/gallery#4-jul-2023-st7586s-jlx320160

Connect serial data, serial clock, DC, A0 and Reset lines to some GPIO ports and tell the u8g2 constructor the pin numbers.

Additionally check the datasheet of your display to see how to connect other pins.

mspudi, 2023-07-29:
> 接线与几乎所有其他SPI图形设备相同，另请参阅此处：https://github.com/olikraus/u8g2/wiki/gallery#4-jul-2023-st7586s-jlx320160
> 
> 将串行数据、串行时钟、DC、A0 和复位线连接到某些 GPIO 端口，并告知 u8g2 构造函数引脚号。
> 
> 此外，请查看显示器的数据表，了解如何连接其他引脚。

384160JLXPCB >>RS-8=SCL  RST-10=reset, CS-10=CS , D1-17=AO,  D0-18=SDA
U8G2_ST7586S_JLX384160_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 13, /* dc=*/ 2, /* reset=*/ 23); 
I don't know how to lose   scl, data 
use
U8G2_ST7586S_JLX384160_2_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 15, /* data=*/ 14, /* cs=*/ 13, /* dc=*/ 2, /* reset=*/ 23);
Can not display
Can share code  JLX384160 test u8g2logo ?  
Thank you

mspudi, 2023-07-29:
> The wiring is the same as for almost all other SPI graphics devices, see also here: https://github.com/olikraus/u8g2/wiki/gallery#4-jul-2023-st7586s-jlx320160
>
> Connect serial data, serial clock, DC, [...]

use  U8G2_ST7586S_JLX384160_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 13, /* dc=*/ 2, /* reset=*/ 23);     How to inform   u8g2  data, serial clock,

mspudi, 2023-07-29:
Can share code  JLX384160 test u8g2logo ?  
Thank you

olikraus, 2023-07-29:
Examples are part of the Arduino IDE

mspudi, 2023-07-29:
> Examples are part of the Arduino IDE

384160JLXPCB >>RS-8= clock  RST-10=reset, CS-10=CS , D1-17=AO,  D0-18=SDA
U8G2_ST7586S_JLX384160_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 13, /* dc=*/ 2, /* reset=*/ 23); 
I don't  Add  clock , data

mspudi, 2023-07-29:
384160JLXPCB >>RS-8= clock  RST-9=reset,  CS-10=CS ,  D1-17=AO,  D0-18=SDA
U8G2_ST7586S_JLX384160_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 13, /* dc=*/ 2, /* reset=*/ 23); 
clock , data   How to add

olikraus, 2023-07-29:
See here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#communication

U8G2_ST7586S_JLX384160_1_4W_HW_SPI --> Data & clock are defined by your Arduino Board.

closed 2024-08-24 by olikraus