olikraus/u8g2 #2189: OLED SPI SSD1309 128x64 2,4''

fatima00ruma, 2023-06-02:
Hi, I'm ussing an ESP32 wroom 32 from az delivery with OLED 2,4'' SPI SSD1309 128x64. With the library U8g2 I finally see some letters in the screen, but only 8 pixels. Anyone knows the reason? 
This is the code I made for Visual Studio Code: 

#include <Arduino.h>
#include <SPI.h>
#include <U8g2lib.h>

U8G2_SSD1306_128X64_NONAME_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 5, /* dc=*/ 32, /* reset=*/ 4);
void setup() {
  u8g2.begin();
}

void loop() {
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_ncenB14_tr);
  u8g2.setCursor(0, 14);
  u8g2.print("Hello World!");
  //u8g2.setFont(u8g2_font_ncenB08_tr);
  //u8g2.drawStr(0,8,"Hello World!");
  u8g2.sendBuffer();
  delay(1000);
}

[image]

fatima00ruma, 2023-06-02:
who solves the problem I will add him in my end-of-degree project, thanks

fatima00ruma, 2023-06-02:
@olikraus help me please!!!

olikraus, 2023-06-03:
Use
```
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=/ 5, / dc=/ 32, / reset=*/ 4);
```

fatima00ruma, 2023-06-03:
It doesnt't work, NONAME with SSD1306 is only for I2C, and I'm ussing SPI. This are the options for SSD1306: 
//U8G2_SSD1306_128X32_UNIVISION_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ 21, /* data=*/ 20, /* reset=*/ U8X8_PIN_NONE);   // Adafruit Feather M0 Basic Proto + FeatherWing OLED
//U8G2_SSD1306_128X32_UNIVISION_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);   // Adafruit Feather ESP8266/32u4 Boards + FeatherWing OLED
//U8G2_SSD1306_128X32_UNIVISION_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);   // Adafruit ESP8266/32u4/ARM Boards + FeatherWing OLED
//U8G2_SSD1306_128X32_UNIVISION_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, /* clock=*/ SCL, /* data=*/ SDA);   // pin remapping with ESP8266 HW I2C
//U8G2_SSD1306_128X32_WINSTAR_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, /* clock=*/ SCL, /* data=*/ SDA);   // pin remapping with ESP8266 HW I2C
//U8G2_SSD1306_64X48_ER_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);   // EastRising 0.66" OLED breakout board, Uno: A4=SDA, A5=SCL, 5V powered
//U8G2_SSD1306_48X64_WINSTAR_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);   
//U8G2_SSD1306_64X32_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SSD1306_64X32_1F_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SSD1306_96X16_ER_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);   // EastRising 0.69" OLED 
//U8G2_SSD1306_72X40_ER_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);   // EastRising 0.42" OLED 
//U8G2_SSD1306_96X40_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SSD1306_96X39_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

Also, my screen is SSD1309

fatima00ruma, 2023-06-03:
The code you gave me, doesn't exist in the library. So it makes errors. I hope we find the solution :)

olikraus, 2023-06-03:
Your code in the initial post is wrong (whether it is 1309 or 1306).
You must use the "F" (full buffer) constructor instead of the "1" constructor.

It is always worth to read the manuals to find quick solutions: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#constructor-name

fatima00ruma, 2023-06-03:
IT WORKS!!!!! YOU ARE A GENIOUS!!!! thankyou so much