olikraus/u8g2 #2191: Additional functionality

xygax, 2023-06-03:
A lot of the GFX displays have an 8 bit bus and 3/4 control lines. It would be useful if with a small amount of logic the port definition could contain I2C for T6963 for instance and then you can use these older LCD's in the same way as the newer OLEDs utilizing the same library 

So an idea for the interface is in the attached
[SCHEMATIC1 _ PAGE1.pdf](https://github.com/olikraus/u8g2/files/11641668/SCHEMATIC1._.PAGE1.pdf)

olikraus, 2023-06-03:
Nice idea :-)

VCC = 5V, correct? What is VBL?

olikraus, 2023-06-03:
One general point: I do not think, that there will be a generic constructor for this solution, instead I think this could be implemented via the C low level interface.

xygax, 2023-06-03:
VBL is a separate backlight supply otherwise 5V

olikraus, 2023-06-24:
https://raw.githubusercontent.com/wiki/olikraus/u8g2/img/i2c_to_parallel_lcd_board_1024.png

Purpose of this issue is to add support for the above board.

Testing could be done with the T6969 board, which should fit into J4:
https://github.com/olikraus/u8g2/wiki/gallery#16-jul-2016-t6963-240x128-lcd

Other displays in my lib might require an adapter:

https://github.com/olikraus/u8g2/wiki/gallery#31-dec-2016-lc7981-160x80

https://github.com/olikraus/u8g2/wiki/gallery#11-aug-2017-sed1520-122x32-lcd

olikraus, 2023-06-25:
The setup for the I2C GFX board differs from standard setup:
``` C++
U8G2 u8g2;              // create a plain U8g2 object
void setup(void) {
  u8g2_Setup_t6963_240x128_1(u8g2.getU8g2(), U8G2_R0, u8x8_byte_i2c_lcd_board, u8x8_gpio_and_delay_i2c_lcd_board);
  u8g2.begin();  
}
```

The setup requires the C API for the displays from here: https://github.com/olikraus/u8g2/wiki/u8g2setupc

Additionally two special callback function are required, they are listed here:
https://github.com/olikraus/u8g2/blob/c8014793981ec12eda8bda0461627c209fab4f2f/sys/arduino/u8g2_page_buffer/I2CLCDBoard/I2CLCDBoard.ino#L99
and
https://github.com/olikraus/u8g2/blob/c8014793981ec12eda8bda0461627c209fab4f2f/sys/arduino/u8g2_page_buffer/I2CLCDBoard/I2CLCDBoard.ino#L119

olikraus, 2023-07-06:
ToDo:

- [ ] SED1520 Example
- [ ] SED1520 Test Board
- [ ] T6963 128x128 constructor / setup procedure

olikraus, 2023-07-09:
The setup procedure is: u8g2_Setup_t6963_128x128_1
Not yet tested...

@xygax : Code is in repo only at the moment. Let me know, whether you need a beta release.

xygax, 2023-07-09:
I will look at it tomorrow.   Thanks... So you wsnt me to send one of those sed1520s... i have a few here...SSent from Outlook for Android

olikraus, 2023-07-09:
I have a sed1520 here, but it has a different pinout. No problem to get it connected, but I need some more time.

olikraus, 2023-07-16:
[image]

Added support for the SED1520. In the above picture, backlight is not enabled and also the protection foil is still attached to the display, so the contrast is little bit poor.

Support is described in I2CLCDBoard.ino example.

https://github.com/olikraus/u8g2/blob/fdbd47ee37ac0cb1f9eca84aa1f9609525eb1972/sys/arduino/u8g2_page_buffer/I2CLCDBoard/I2CLCDBoard.ino#L268-L269

olikraus, 2023-07-23:
ToDo:

- [ ] Support for ST7920: Adapter required, new byte procedure required
- [ ] Support for KS0108: Adapter required, new byte procedure required
- [ ] u8g2_Setup_t6963_128x128_2 doesn't compile, why? Code required
- [ ] u8g2_Setup_lc7981_240x128_2 compiles, but doesn't work. Problem here is, that I don't know whether the LC7981 driver works.

olikraus, 2023-08-14:
The T6963 128x128 works as expected.
The ST7920 is now added to the example code https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/I2CLCDBoard/I2CLCDBoard.ino

See also images below.

The ST7920 display must be connected to J3.
Backlight needs to be added additionally. The 10 Ohm Resistor will limited the current, otherwise the Uno may not start up as expected (so it might be required to use another power source for the ST7920 backlight).

|    Pin    |     ST7920   |       J3  | Note |
|---|---|---|---|
|    1      |     GND      |       VSS (GND) | |
 |   2      |     VCC      |       VCC | |
 |   3      |     V0       |       V0 | |
 |   4      |     RS        |      C/D | |
 |   5     |      R/W      |       R/W      |       0: write mode |
  |  6     |      E          |     E | |
 |   7     |      DB0       |      DB0 | |
 |   8      |     DB1  |  | |
 |   9       |    DB2 |  | |
 |   10      |    DB3 |  | |
 |   11     |     DB4 |  | |
  |  12     |     DB5 |  | |
 |   13     |     DB6 |  | |
|    14     |     DB7   |          DB7  | |
 |   15    |      PSB      |       CS       |       PSB=1 --> parallel bus mode |
  |  16     |     NC       |       /Reset | |
 |   17     |     RST       |      VBL |  |
 |   18    |      VOUT      |     LED |  |
  |  19   |       BL A |  | |
  |  20  |        BL K | | |

[image]

[image]

Due to light conditions, this white on blue display appears as blue on blue.

olikraus, 2023-08-28:
Here is another display: LC7981 with the following pinout:
|    Pin    |     ST7920   |       J3  | Note |
|---|---|---|---|
|    1      |     GND      |       VSS (GND) | |
 |   2      |     VCC      |       VCC | |
 |   3      |     V0       |       V0 | |
 |   4      |     RS        |      C/D | |
 |   5     |      R/W      |       R/W      |       0: write mode |
  |  6     |      E          |     E | |
 |   7     |      DB0       |      DB0 | |
 |   8      |     DB1  |  | |
 |   9       |    DB2 |  | |
 |   10      |    DB3 |  | |
 |   11     |     DB4 |  | |
  |  12     |     DB5 |  | |
 |   13     |     DB6 |  | |
|    14     |     DB7   |          DB7  | |
 |   15    |      /CS      |       CS       |      |
  |  16     |    /RST       |       Reset | |
 |   17     |     LED+       |      VBL |  |
 |   18    |      LED-     |     LED |  |
 
This display can be added to J3 of the I2C GFX converter.

olikraus, 2023-08-28:
Support has been added to u8g2 for this LC7981 240x128 display, however the power consumption is huge and also the Arduino Uno might be too slow.
For the pictures below the backlight LED is disabled and the I2C GFX board is modified to generate a more negative contrast voltage (1K replaced by 820 ohm)

[image]

[image]

olikraus, 2023-08-29:
[GFXSCHEMATIC_V1.02_ PAGE1.pdf](https://github.com/olikraus/u8g2/files/12468347/GFXSCHEMATIC_V1.02_.PAGE1.pdf)

olikraus, 2023-11-08:
Add support for a ST7920 128 x 32 px display

xygax, 2024-03-02:
Oli.. 
Did the KS0108 get added to the I2C module i have a 192 x 64 first 16 pins are identical there is a couple of extra CS lines and a switched positive for the back light so the CS 2,3 are connected to 0V and the backlight to the positive supply atm 

[image]

olikraus, 2024-03-03:
Hi Steve
The KS0108 is kind of a nightmare. Not sure, whether there is support for
the KS0108 in 192x64 resolution.
I think you sent another mail...

Oliver

olikraus, 2024-03-03:
Adding support for the 192x64 KS0108 should be doable, but precondition would be a way to control CS2 and CS3

xygax, 2024-03-03:
Just looked up the data sheet it needs 3 lines …

 

Ok so I could use E2 and /RD as these two serve no function atm and are available on J5 pin 6 (E2 >> CS2) J4 pin 6 (/RD >> CS3)

olikraus, 2024-03-03:
ok, so for the KS0108, the mapping would be:
https://github.com/olikraus/u8g2/blob/ec2559e8466dead8b13dd92cc17ac10e63dab6ac/sys/arduino/u8g2_page_buffer/I2CLCDBoard/I2CLCDBoard.ino#L52-L60

| Expander | KS0108 |
|---|---|
|       Bit 0: C/D  |   KS0108 RS|
|       Bit 1: R/W  | KS0108 R/W|
|       Bit 2: E      |   KS0108 E|
|       Bit 3: E2    |   KS0108 CS2|
|       Bit 4: /CS |    KS0108 CS1|
|       Bit 5: /RD  |   KS0108 CS3|
|       Bit 6: /Reset |  KS0108 /Reset|
|       Bit 7: LED |  |

?

xygax, 2024-03-03:
Yep that looks like its a soulition.. if i revise the module it needs another connector on it....ThanksSteveSent from Outlook for Android

xygax, 2024-03-03:
I have added J9 to the pcb to deal with the extra CS lines and the Vo which is about 8v and generated on the module. It lights up and the contrast can be adjusted ... thats all i have tested so far..Sent from my Galaxy

https://github.com/olikraus/u8g2/blob/ec2559e8466dead8b13dd92cc17ac10e63dab6ac/sys/arduino/u8g2_page_buffer/I2CLCDBoard/I2CLCDBoard.ino#L52-L60

olikraus, 2024-03-13:
I have updated the example .ino for the KS0108: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/I2CLCDBoard/I2CLCDBoard.ino

You can just use that file or get the latest lib from here:

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

xygax, 2024-03-15:
[image]
LCM19264A-1 pinout supported on J7 of the adaptor v1.01

xygax, 2024-03-15:
KS0108 (LCM19264A-1) hardware connection added J7 on V1.01 of the adaptor
[image]

xygax, 2024-03-15:
V1.01 i2c converter module ( manufacturing screenshot). The best mounting method is to fit the connector to the rear of the adaptor board and it then sits behind the dispaly where its not seen 
[image]

olikraus, 2024-08-24:
I will close this ticket to finish the current milestone. Feel free to add new comments here.

closed 2024-08-24 by olikraus