olikraus/u8g2 #2193: dht22 with arduino and iic12864 from ali

Nemo79, 2023-06-03:
hello, I have a problem with arduino nano and i2c display from ali.
I connected and programmed the temperature and humidity meter according to the following scheme and it does not work. Everything is ok for 15 minutes, but then it starts mirroring the text on the LCD. any advice what's wrong?

code: https://pastebin.com/xTHv5tVG

[image]

olikraus, 2023-06-03:
Your display a 3.3V display, so the I2C bus also has to have a 3.3V high level which is not the case according to your schematic. The DHT22 might also work with 3.3V, so maybe changing the +5V power rail for the attached devices to 3.3V might work already.

Nemo79, 2023-06-03:
maybe there is an issue, but on the back side of LCD is written 3-5V. I will add a voltage divider for the first approximation.

olikraus, 2023-06-03:
> I will add a voltage divider for the first approximation.

The voltage divider may not provide the required current. You could just use the 3.3V output of the Arduino Nano instead.

Nemo79, 2023-06-03:
hi Oliver, the issue was caused by death or crashed i2c bus on first Arduino. I change arduino nano board to a new unboxed Arduino nano and it works for two hours without a problem

Nemo79, 2023-06-03:
now i am thinking how to add sd card and logging temperature and humidity

olikraus, 2023-06-03:
> now i am thinking how to add sd card and logging temperature and humidity

In the same I2C bus? Be aware, that Arduino Wire doesn't has a concept to allow multiple I2C devices with different configuration requirements. U8g2 is already kind of smart here, but other libs may not.

mikcip, 2023-08-05:
Keresek egy kódot, ami kiírja a GM12864-59N LCD kijelzőre a HELO szöveget.