olikraus/u8g2 #2194: DIYMORE 2.42" SSD1309 not working on I2C

alextrical, 2023-06-03:
Ive currently got a couple of SSD1309 displays from DIYMORE found[ here ](https://www.diymore.cc/products/2-42-inch-12864-oled-display-module-iic-i2c-spi-serial-for-arduino-c51-stm32-green-white-blue-yellow)

I need them to be in I2C mode due to limited pins available on the board I'm trying to use (a Big tree tech EBB42), 
After a while of scratching around to get them to be configured to work in I2C mode, they are now showing up in I2C scanner as `0x3C`

However I have tried multiple configurations, both HW and Software config, with no luck getting anything to be displayed
```
U8G2_SSD1309_128X64_NONAME0_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
U8G2_SSD1309_128X64_NONAME2_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
U8G2_SSD1309_128X64_NONAME0_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCLPin, /* data=*/ SDAPin, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display
U8G2_SSD1309_128X64_NONAME2_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCLPin, /* data=*/ SDAPin, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display
```

Setup has been tested with a known working SSD1306. I'm not sure what to try next after this

olikraus, 2023-06-03:
Just to be on the safe side, assign the I2C address manually via 
``` C++
u8g2.setI2CAddress(2*0x3c);
u8g2.begin();
```
see also https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress 

You could also try the SSD1306 constructor, because the SSD1306 is compatible with the SSD1309.

If this doesn't help, check the power consumption: SSD1309 require LOT more current than a SSD1306.

alextrical, 2023-06-03:
Well, after spending 3.5 hours trying to sort this display before sending a help message, i have it fixed in under 15 minutes. 
It seems that this particular display needs the reset line pulled low momentarily after power up to display anything, i was hoping it could be done by SW, but that may still be a bug (don't know if the SSD1309 works anything like the SSD1306 behind the scenes)

[image]
For this particular board, to convert it to I2C you need to do the following:

Move the resistor from R4 to R3
Short R5 & R7
Short DC to ground (you can short DC to RS on the pin header) to set ID to 0x3C (Pull up to vcc for 0x3D iirc)
Add a 10K pull-up resistor between RES and VCC

Pull RES to GND after power up to reset the display and allow the display to work. If you don't do this, the module will show up on the I2C scanner, but not display anything

alextrical, 2023-06-03:
[image]

olikraus, 2023-06-03:
> Is there a way to reset this screen over I2C without having to toggle the RST pin on the board?

The datasheet does not mention any software reset command, so the answer is "no".

alextrical, 2023-06-03:
Sorry, I really should have checked that myself. 

In that case everything is working as expected, thank you so much. :) 
Hopefully the notes I made about the connections will be of use to the next person who buys one of these boards and uses it with I2C

closed 2023-06-03 by alextrical